Koenkk/zigbee2mqtt #26089: [Feature request]: HT402: Missing Hilo challenge mode attribute

lit-af, 2025-01-31:
### Is your feature request related to a problem? Please describe

While the HT402 is supported, it's missing the Hilo challenge mode (icon 4) attribute in the converter.

https://www.hiloenergie.com/en-ca/products/heating/thermostat-intelligent-hilo-thermostat/

[image]

### Describe the solution you'd like

I tried sniffing the Zigbee bus, but I can't find the attribute at the moment. I suspect that it might be enabled once then canceled by the thermostat itself 4 hours later.

I'd like to be able to set pictogram 4 on for four hours so that I know when there's a peak demand event occuring at the moment.

### Describe alternatives you've considered

Live without it.

### Additional context

I'll try asking Stelpro nicely by email for the engineering info.

closed 2025-04-09 by github-actions[bot]

spookyneo, 2025-04-13:
@lit-af, did you ever reach out to Stelpro about this and found out the attribute to use ? I too am looking for this.

lit-af, 2025-04-13:
I didn't contact them. I was able to sniff my ZigBee network while there was a peak event. I'll post them here eventually.

spookyneo, 2025-04-13:
Awesome ! Please do post them when you get some free time.

lit-af, 2025-04-25:
Since I don't know how to share the pcapng files without reveiling my network private key, here's some screenshots.

At 4AM when the peak event starts this is written to the thermostat:

[image]

At 11AM when the peak event ends this is written to the thermostat:

[image]

I wonder if this is enough information to add this feature to the zigbee-herdsman-converters for HT402

lit-af, 2025-12-19:
I figured it out! Using zha_toolbox but adding it to z2m might not be too hard, I'll look into it eventually.

To turn it on:
```yaml
action: zha_toolkit.attr_write
data:
  ieee: climate.stello_ht402
  endpoint_id: 25
  cluster: 0x0201
  attribute: 0x4105
  attr_val: 25200
  value_type: uint16
```

To turn it off:
```yaml
action: zha_toolkit.attr_write
data:
  ieee: climate.stello_ht402
  endpoint_id: 25
  cluster: 0x0201
  attribute: 0x4105
  attr_val: 0
  value_type: uint16
  ```

lit-af, 2026-02-02:
@fersingb I added the peak demand event icon converter in the linked PR, if you'd like to test it. 
I wasn't able to add a converter for the local_time sync it keeps telling me that the attribute isn't writable for some reason.

fersingb, 2026-02-02:
@lit-af I never ran non prod Z2m code. I'll check how to test it.

The local time attribute is indeed not writable. The thermostat listens to the report attributes and updates the value on its own from there. That's why I didn't find a way to implement this properly.

The hilo gateway sends a report attributes as broadcast and the thermostats act based on that. That's why I used raw commands to replicate that behavior.