Koenkk/zigbee2mqtt #26095: "Aqara" "WS-EUK03" "Decoupled mode" does not update "Action" state

TheRealVira, 2025-02-01:
### What happened?

Action trigger is not updated.
### Specs 
- Device: https://www.zigbee2mqtt.io/devices/WS-EUK03.html#aqara-ws-euk03
- Action (enum): https://www.zigbee2mqtt.io/devices/WS-EUK03.html#action-enum
### Exposed
[image]
### Settings (specific)
[image]
### OTA
Seems to be stuck at `11-10-2021` without any firmware version linked. Strange since OTA is mentioned as supported:
https://www.zigbee2mqtt.io/devices/WS-EUK03.html#ota-updates
[image]
[image]

### What did you expect to happen?

I expected the action enum to change state or trigger an event upon single or double clicking the single rocker.

### How to reproduce it (minimal and precise)

**Pair a WS-EUK03**
_Exposes:_
- State: ON
- Flip indicator light: OFF
- Power outage memory: True
- LED disabled night: True
- Operation mode: decoupled
- State action: True

_Config_
- State action: True

### Zigbee2MQTT version

2.0.0 commit: unknown

### Adapter firmware version

0x26780700

### Adapter

ConBee II

### Setup

Add-On; OS Version: Home Assistant OS 14.2; Home Assistant Core: 2025.1.4; VM on Proxmox

TheRealVira, 2025-02-06:
„Fixed“ it by ditching my old conbee II for sonoff 3.0 P

closed 2025-02-06 by TheRealVira

JakubMazur, 2025-02-22:
Thanks for a fix. This also fixed my issue, I had conbee III after migrating to Sonoff ZBDongle-E with firmware 7.4.4 everything is fine

omBratteng, 2025-03-21:
I'm having the same issue, but I'm using a SLZB-06M. I also have a ZBDongle-E that I used before the SLZB-06M, and there it works as expected.