Koenkk/zigbee2mqtt #26098: Moes ZHT-002 [GA] needs z2m support, (is not be recognized)

ftz19, 2025-02-01:
### Link

https://de.aliexpress.com/item/1005008266328948.html?spm=a2g0o.order_list.order_list_main.31.2a245c5fNNLjLr&gatewayAdapt=glo2deu

### Database entry

{"id":51,"type":"Router","ieeeAddr":"0xa4c138460a3dd895","nwkAddr":43957,"manufId":4417,"manufName":"_TZE204_xalsoe3m","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�\u000f1/f�\u000f1/\u0012�\u000f1/e�\u000f1/\u0012","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","appVersion":74,"manufacturerName":"_TZE204_xalsoe3m","zclVersion":3,"modelId":"TS0601","powerSource":1}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00212effff0e567e","endpointID":1},{"cluster":10,"type":"endpoint","deviceIeeeAddress":"0x00212effff0e567e","endpointID":1},{"cluster":4,"type":"endpoint","deviceIeeeAddress":"0x00212effff0e567e","endpointID":1},{"cluster":25,"type":"endpoint","deviceIeeeAddress":"0x00212effff0e567e","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1738427079793}

### Zigbee2MQTT version

2.0.0-2

### Comments

This is the successor of Moes BHT-002-GCLZB in Zigbee2MQTT.
And so it is not recognized, becuase the new name is now ZHT-002.
Please could you add it to the zigbee2MQTT database?

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_xalsoe3m',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Device will not show up. No temperature, no regulation possible.

mrmaximas, 2025-02-01:
https://github.com/Koenkk/zigbee2mqtt/issues/25372

ftz19, 2025-02-01:
Thanks So I close this.

closed 2025-02-01 by ftz19