Koenkk/zigbee2mqtt #26099: Problem with Aqara WXKG11LM button

Gobolin80, 2025-02-01:
### What happened?

I've recently migrated to version 2. In the previous version, I had created a sensor to detect single and double clicks from an Aqara WXKG11LM switch, which I used for various triggers. The configuration was as follows:

switch_bagno_event:
  value_template: >-
    {{ states('sensor.smart_switch_bagno_action') }}

After the migration, I redefined the sensor like this:

mqtt:
  sensor:
    - state_topic: "zigbee2mqtt/smart_switch_bagno/action"
      name: "switch_bagno_event"

**The Issue:**
Now, after a single click, the sensor remains stuck in the "single" state, and it doesn’t register a subsequent click. The same behavior occurs with double clicks—it stays in the "double" state without detecting new events.

This is problematic because, when using the "toggle" function, for example, the first click turns on a light, but since the second click isn’t detected, I can’t turn it off.

Could you help identify the issue or suggest a workaround?

Thanks in advance for your support!

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

 7.4.2 [GA]

### Adapter

ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Setup

Home Assistant

yuriyb80, 2025-02-02:
HI , same issue with the Aqara WXKG11LM switch , after migration to new version  homeassistant doesn't see action commands. 

the mqtt explorer detect command in the topic 

[image]

   
but it automation shows an error   

[image]

Gobolin80, 2025-02-03:
Hi @AlexzeyK ,
"_action" sensore were removed in zigbee2mqtt 2 as indicated in the official notes: [](https://github.com/Koenkk/zigbee2mqtt/discussions/24198)

AlexzeyK, 2025-02-04:
Thank you @Gobolin80  for the tip.
Now I know how to repair my scene switches.

PS: the outdated "_action" function was much more convenient))

closed 2025-04-13 by github-actions[bot]