Koenkk/zigbee2mqtt #26107: Ikea E1744 no longer exposing "Action" to HA correctly

Merijeek, 2025-02-01:
### What happened?

Since update to 2.0, unable to use Ikea Symfonisk Dial (E1744) correctly in HA. 

It appears that the required exposed entity "Action" is no longer an entity in Home Assistant. I don't know for a fact that this used to be in there, but AFAIK it is necessary for the commonly used blueprint (https://github.com/EPMatt/awesome-ha-blueprints/blob/main/blueprints/controllers/ikea_e1744/ikea_e1744.yaml) to work correctly. 

Without the entity existing, I don't think there's a way to make it function in HA. 

### What did you expect to happen?

Kind of hoped it would keep working after the update? 

### How to reproduce it (minimal and precise)

E1744 is paired and visible in Z2M in HA. Hit the button, I see the "Action" under "Exposed" show the last action. 

However, I see nothing inside HA regarding any state changes here. 

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240710

### Adapter

Sonoff ZBdongle-P

### Setup

Add on to Home Assistant OS running on ODRIOD N2+

bayview252, 2025-02-02:
Check out [https://github.com/yarafie/awesome-ha-blueprints/tree/main](url) for a fork of the original blueprints set up to work with MQTT Device Triggers.

Merijeek, 2025-02-02:
Thanks a zillion. This guy was about to land in the trash and get chalked up to a casualty of the march of versions. 

Only did a quick test, but so far so good.

papperone, 2025-02-22:
> This is one of the Zigbee2MQTT 2.0.0 breaking changes, from [#24198](https://github.com/Koenkk/zigbee2mqtt/discussions/24198)
> 
> > * All action sensors are now disabled by default (`sensor.*_action` entities). It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead. In case you really need the action sensors, add the following to your `configuration.yaml`.
> >   homeassistant:
> >     legacy_action_sensor: true

how can I make "legacy_action_sensor" works? even if I add to Z2M configurations those 2 lines the "*_action" entities are not created :( what am I doing wrong?