Koenkk/zigbee2mqtt #26108: Seemingly broken CO2 values

theorlangur, 2025-02-01:
### What happened?

 I've just updated to 2.1 and my 2 CO2 sensors unexpectedly "jumped" up. My guess is: because of [this PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8644).
One of the sensors in question is [HEIMAN HS3AQ](https://www.zigbee2mqtt.io/devices/HS3AQ.html)
Another one is my own (DIY) SCD40.

### What did you expect to happen?

I don't think that calculating a measured value boils down to a 1:800=0.00125 (the example in PR), really. 
Since ppm is parts per million and you ought to get 800:1000'000. 
In other words the previous implementation seemed to be ok for CO2.

### How to reproduce it (minimal and precise)

in my case looking at a display of HEIMAN HS3AQ and comparing it to what zigbee2mqtt is showing was enough see the sudden discrepancy.

### Zigbee2MQTT version

2.1.0-1

### Adapter firmware version

7.4.4 [GA]

### Adapter

ZBDongle-E, EmberZNet driver

### Setup

Add-on on HomeAssistant

devbis, 2025-02-02:
https://github.com/Koenkk/zigbee-herdsman-converters/issues/8711

Eyenseo, 2025-02-02:
I find the standard quite vague (no units and no examples what is send), which can be seen in the open points of my PR. I guess now we know that some devices do use the old way.

I came to the interpretation of `1/X` based the quoted section and the table of supported substances (4.13.1.3 Cluster Identifiers). For the substances with `ppY` the old implementation can make sense - just divide by the multiplier in the unit, similarly with other units eg. "Dissolved Oxygen" with probably mg/L to kg/m³ or g/cm³? The caveat being that the client needs to know the correct unit to multiply by the right magnitude. And then there is the question of ... why? Why would it make sense to change the magnitude of the measurement? Floats precision scales with the size of the value anyway. I then noticed "Turbidity" with a range of 0-10 and came to the conclusion that they wanted to scale via `1/X`.
Looking more into "Turbidity"; This is probably the worst part about the standard:
* it escapes `0<=value<=1` - breaking the old implementation (unless I have not found the right units that provide some magnitude), BUT 
* it occupies `0<=value<=1` - breaking the new implementation, which I didn't notice when I created the PR - sorry about that!

Either way - I guess my PR should be reverted, but I expect more trouble based on section 4.13.

Koenkk, 2025-02-02:
Thanks, reverted!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-02-02 by Koenkk

devbis, 2025-02-03:
@Eyenseo I agree that standard is vague and lacks examples. It describes fractional concentration quite good but unclear when a measurement value has units, like mass in a volume: mg/m^3.

But it is NOT restricted to have values  <= 1.

E.g. for Turbidity measured in FTU/NTU you may pass 10 NTUs as 10 fractions of 1 (10.0f)
The same approach I apply to mg/m^3. 200 mg/m^3 should be represented as 200.0f in the attr value

bisquit2003, 2025-02-04:
> Thanks, reverted!
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Hello, 
is it possible to release a fix for this particular problem a little earlier? I've seen a lot of discussions about this issue, and switching to dev and then back doesn't look like the best solution.
Thx

bisquit2003, 2025-02-04:
@Koenkk I tried the dev version, but the problem remained(

Koenkk, 2025-02-05:
@bisquit2003 are you on zigbee-herdsman-converters 21.29.0? (frontend -> settings -> about) If not, try updating again.

bisquit2003, 2025-02-06:
> [@bisquit2003](https://github.com/bisquit2003) are you on zigbee-herdsman-converters 21.29.0? (frontend -> settings -> about) If not, try updating again.

After pr of @kirovilya its work good