Koenkk/zigbee2mqtt #26112: zigbee2mqtt 2.0 & higher will not run

elimjohnson, 2025-02-02:
### What happened?

I am currently on 1.42.0 and have not been able to get zigbee2mqtt onto the last three updates since.  I noticed in the log that it was claiming an error as shown:

Refusing to start because configuration is not valid, found the following errors:
- Device icon of 'Smart Outlet: Tablet (Mom's)' should start with 'device_icons/'

This is an Aqara outlet.  I deleted this device from my configuration and figured I would just repair it after the update.  The next time, another Aqara device came up as the reason zigbee2mqtt would not start.  I deleted that one, restored the backup and tried the update again.  Yet another Aqara device came up.  I have about 18 more Aqara devices and I do not want to remove them all if I can get this fixed another way.

Any help would be appreciated.  I am not technical  when it comes to HA and its integrations, so please be patient.

Thanks

HA OS 14.2 (Also happened on 14.1)
HA Core 2025.1.4
HA Supervisor 2024.12.3
RPi 5 8GB
Sonoff Dongle Plus
NVME Pi Hat

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20250109.2

### Adapter

Sonoff Dongle Plus TI CC2652P + CP2102N

### Setup

Add-on, RPi 5 8GB

### Debug log

[20:22:58] INFO: Preparing to start...
[20:22:58] INFO: Socat not enabled
[20:22:58] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
            READ THIS CAREFULLY
Refusing to start because configuration is not valid, found the following errors:
- Device icon of 'Smart Outlet: Tablet (Mom's)' should start with 'device_icons/', got 'data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAgAAAAIACAYAAAD0eNT6AAABHGlDQ1B.................................

GigiPompieru, 2025-02-02:
is that a custom icon? can't you just remove the device from configuration.yaml and then re-pair it?

elimjohnson, 2025-02-02:
Custom?  All the Aqara icons came through when I paired the devices.  I didn't add them if that is what you mean.  My configuration.yaml file has almost nothing it.

GigiPompieru, 2025-02-02:
in ```zigbee2mqtt\configuration.yaml``` you have your devices like:
```
devices:
  '0xa4c138c9ddcbc1da':
    friendly_name: Device1
  '0xa4c138a03a3d6ead':
    friendly_name: Device2
  '0xa4c138a97122bb0e':
    friendly_name: Device2
  '0x60a423fffe11799b':
    friendly_name: Device2
```
Remove the entry for the trouble device and try restarting Z2M.

closed 2025-04-11 by github-actions[bot]