Koenkk/zigbee2mqtt #26113: JavaScript heap out of memory during "pnpm run build"

michalw2-alt, 2025-02-02:
### What happened?

During update using script ./update.sh "JavaScript heap out of memory"  error is reported on Raspberry Pi Zero 2 which has only 0.5 GB RAM.

Issue can be solved by adding replacing: 
"pnpm run build" line 
with: 
"sudo NODE_OPTIONS="--max-old-space-size=1536" pnpm run build"

Afterwards operation gets completed without any errors.

sudo NODE_OPTIONS="--max-old-space-size=1536" pnpm run build

> zigbee2mqtt@2.1.0 build /opt/zigbee2mqtt
> tsc && node index.js writehash

I have not tested lower than 1536 values.

Is there a chance for permanent fix in ./update.sh so that it's not required for manual update after each release?

### Zigbee2MQTT version

2.1

### Adapter firmware version

20240710

### Adapter

zStack3x0

### Setup

Plain, HomebridgeOS on Raspberry Pi Zero 2

### Debug log

> zigbee2mqtt@2.1.0 build /opt/zigbee2mqtt
> tsc && node index.js writehash

<--- Last few GCs --->

[3314:0x5e06be8]    68244 ms: Mark-Compact 123.1 (129.5) -> 122.4 (129.5) MB, 539.72 / 0.00 ms  (average mu = 0.176, current mu = 0.041) allocation failure; scavenge might not succeed
[3314:0x5e06be8]    68795 ms: Mark-Compact 123.1 (129.5) -> 122.5 (129.5) MB, 529.19 / 0.00 ms  (average mu = 0.116, current mu = 0.040) allocation failure; scavenge might not succeed

<--- JS stacktrace --->

FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory
----- Native stack trace -----

Aborted
 ELIFECYCLE  Command failed with exit code 134.

LaurentChardin, 2025-02-02:
I would rather recommend to setup the NODE_OPTIONS variable globally on your pi zero: it would not be specific to z2m.

In the long run, it could be possible that the build process requires more memory during the build process (unless we switch to more optimized buildtool and runtime like deno or bun), than for the runtime process.

Or eventually think about having precompiled packages for tiny devices.

But i don't believe lowering down the memory limit during the build phase is good idea.

michalw2-alt, 2025-02-02:
> I would rather recommend to setup the NODE_OPTIONS variable globally on your pi zero: it would not be specific to z2m.
>
> In the long run, it could be possible that the build process requires more [...]

Thank you for this suggestion.

closed 2025-04-11 by github-actions[bot]