Koenkk/zigbee2mqtt #26115: [Feature request]: _TZ3000_xkap8wtb paired as TS0001_power

giandena, 2025-02-02:
### Is your feature request related to a problem? Please describe

This device does not show the electrical values in Z2M. 
I am probably circling around the solution but my limited skills do not allow me to solve it.

### Describe the solution you'd like

Obtain and monitor electrical values

### Describe alternatives you've considered

I have verified that the device shows the electrical values when it is associated with a Tuya gateway. 
The values are also shown in the device console when reading haElectricalMeasurement with the attributes msvoltage, mscurrent, and activepower

[image]

### Additional context

I also tried to apply the solution proposed in the issue #13835 but was unsuccessful.
I created an ext_converter.js file that I put in the /homeassistant/zigbee2mqtt/data/external_converters folder I inserted in configuration.yaml the reference to the ext_converter.js file but the problem was not solved.

[image]

[image]

[image]

HomeAssistantBarbados, 2025-02-03:
Same problem:
Zigbee model: TS0001
Zigbee manufacturer: _TZ3000_xkap8wtb
Firmware version: 500+TZSKT31Z3101
Model: TS0001_power.

z2m: Failed to configure 'k4_pol', attempt 1 (Error: ZCL command 0x70b3d52b60126759/1 genBasic.read(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:349:28) at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:954:26) at processTicksAndRejections (node:internal/process/task_queues:105:5) at Endpoint.read (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:445:29) at Object.configure (/app/data/extension/ext_converter.js:21:9) at Configure.configure (/app/lib/extension/configure.ts:124:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))

z2m: Failed to configure 'k4_pol', attempt 1 (Error: ZCL command 0x70b3d52b60126759/1 genBasic.read(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:349:28) at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:954:26) at processTicksAndRejections (node:internal/process/task_queues:105:5) at Endpoint.read (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:445:29) at Object.configure (/app/data/extension/ext_converter.js:21:9) at Configure.configure (/app/lib/extension/configure.ts:124:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))

giandena, 2025-02-03:
I also tried associating the device using ZHA on Homeassistant and the electrical values are displayed.
The only problem is that the electrical current value must be divided by 1000. It is shown 82A when it should be 82mA.

HomeAssistantBarbados, 2025-02-03:
I tried to connect via SLS Smart Home.
Converters are not detected automatically.
Works with the BW-SHP13 converter, manufacturer: BlitzWolf, description: Tuya Blitzwolf Smart Socket EU Plug 16A Adapter.
But I need to work with Zigbee2MQTT.
There is no solution yet.

HomeAssistantBarbados, 2025-02-04:
> Я также попробовал связать устройство с помощью ZHA на Homeassistant, и электрические значения отображаются. Единственная проблема в том, что значение электрического тока должно быть разделено на 1000. Оно отображается как 82A, когда должно быть 82mA.

Can you show your full ext_converter.
It doesn't show anything for me.
It only works as a switch. And there are many errors in Zigbee2MQTT.

giandena, 2025-02-04:
In zigbee2mqtt it only works as a switch. I removed zigbee2mqtt from Home Assistant and installed ZHA, only then does the smart switch show all entities. This shows that the problem is zigbee2mqtt not recognising this device correctly.

HomeAssistantBarbados, 2025-02-04:
> В zigbee2mqtt он работает только как коммутатор. Я удалил zigbee2mqtt из Home Assistant и установил ZHA, только тогда смарт-коммутатор показывает все сущности. Это показывает, что проблема в том, что zigbee2mqtt не распознает это устройство правильно.

I have the same story.
the manufacturer incorrectly indicated the device type, so someone wrote on one of the forums.
I don't know how to update it.
it doesn't work https://github.com/Koenkk/zigbee2mqtt/issues/13929

closed 2025-02-07 by giandena