Koenkk/zigbee2mqtt #26120: Divisors need updating after SP 242 firmware update

AlexanderLeporiday, 2025-02-02:
### What happened?

Firmware build 20250127 (1.9.28) for the Innr SP 242 has been released but it's changed the divisors.

@GerryInnr reported that:
> Before this firmware version, the device did not provide a multiplier or divisor for CurrentSummationDelivered, it only provided the CurrentSummationFormatting (because multiplier and divisor are optional in Zigbee and CurrentSummationFormatting is mandatory). The new version provides all three, with multiplier = 1 and divisor = 1000. If the converter just sets/overrides the divisor to 100 (which was the right value for the previous firmwares) without checking whether the device already provided a value, then yes, the converter would need fixing for the new firmware version to set it to 1000.

Energy should read 64.39 kWh but is off by a factor of 10 due to the hard coded divisors.
[image]

### What did you expect to happen?

Read the multiplier and divisor from the device (Remove the hard coded override?)

### How to reproduce it (minimal and precise)

Update firmware, see that the values now report differently because of the hard coded values and the change to the defaults.

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

20240710

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Setup

Home Assistant OS running Z2M as an addon

AlexanderLeporiday, 2025-02-02:
I am convinced the issue belongs to https://github.com/Koenkk/zigbee-herdsman-converters/ however I thought it best to follow the recommended issue reporting route and raise it here first.
I'll create a duplicate issue over on the other tracker so this can be closed as migrated or however those situations are handled.

mundschenk-at, 2025-02-02:
Draft PR: https://github.com/Koenkk/zigbee-herdsman-converters/pull/8715

jodewee, 2025-02-03:
same issue over here, with the SP240, firmware 1.9.28
Power looks OK, but energy is off by factor 10( reported 5.5kwh instead of 0.55kwh)

Balooforever, 2025-02-05:
Same for me with SP240 and the last firmware.

Thank for your work :)

mundschenk-at, 2025-02-05:
It's already fixed in [`dev`](https://github.com/Koenkk/zigbee2mqtt/tree/dev).

AlexanderLeporiday, 2025-02-06:
This has been pushed out as a release [2.1.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.1.1) and the issue is fixed.
You'll need to be on firmware build 20250127 version 1.9.28.

closed 2025-02-06 by AlexanderLeporiday

hmmbob, 2025-02-06:
For those updating to 2.1.1 with the new firmware and still seeing wrong values:

You need to hit the "reconfigure" button of the device in the zigbee2mqtt frontend (it's the yellow arrows button) before the changes come into effect.