Koenkk/zigbee2mqtt #26121: [Wrong device]: Tuya Zigbee USB Power Switch shown as TS0002

13robin37, 2025-02-02:
### Link

https://de.aliexpress.com/item/1005007444117700.html

### Model

XMSJ

### Description

Zigbee USB plug 2 Channel

### Vendor

Tuya

### Picture (link)

https://ae-pic-a1.aliexpress-media.com/kf/Sde5925c3691b40eaa48d5c17203cc046a.jpg

### Database entry

{"id":87,"type":"Router","ieeeAddr":"0xa4c138fad8542dde","nwkAddr":54964,"manufId":4417,"manufName":"_TZ3000_h1ipgkwn","powerSource":"Mains (single phase)","modelId":"TS0002","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\b\u0000\u0000\u0000\u0012\u0005\u0000\u0000\u0000\u0012\u000f\u0000\u0000\u0000\u0016","65506":56,"65508":1,"65534":0,"modelId":"TS0002","manufacturerName":"_TZ3000_h1ipgkwn","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}},"manuSpecificTuya_4":{"attributes":{"inching":"AAAAAgAA"}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002393375e","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002393375e","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1738495163362}

### Notes

Basically same issue as #24039 except this one has two ports instead of one USB port.

ShadowFistOkami, 2025-02-15:
Sorry to piggyback, but figured I might as well add the 3 usb version here too and have everything supported in 1 go.

The 3 USB version is recognised as a TS0003:

`{"id":17,"type":"Router","ieeeAddr":"0xa4c1387bf9d90f7e","nwkAddr":6466,"manufId":4417,"manufName":"_TZ3000_mw1pqqqt","powerSource":"Mains (single phase)","modelId":"TS0003","epList":[1,2,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAAAgAABAAA"}},"genBasic":{"attributes":{"65506":56,"65508":0,"65534":0,"modelId":"TS0003","manufacturerName":"_TZ3000_mw1pqqqt","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"switchType":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b4966f","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b4966f","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b4966f","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1739640737262}`

Feel free to let me know whether you prefer to have a separate issue.

mrmaximas, 2025-02-24:
@ShadowFistOkami i have same device 3-port USB adapter _TZ3000_mw1pqq
@Koenkk It's strange that the converter has no power-on behaviour. The classic TUYA / SL app has this option.

[image]

```
{
  "id": 123,
  "type": "Router",
  "ieeeAddr": "0xa4c138c8421d41d4",
  "nwkAddr": 20228,
  "manufId": 4417,
  "manufName": "_TZ3000_mw1pqqqt",
  "powerSource": "Mains (single phase)",
  "modelId": "TS0003",
  "epList": [
    1,
    2,
    3,
    242
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 256,
      "inClusterList": [
        3,
        4,
        5,
        6,
        1794,
        2820,
        57344,
        57345,
        0
      ],
      "outClusterList": [
        25,
        10
      ],
      "clusters": {
        "57344": {
          "attributes": {
            "53251": "AAAAAgAABAAA"
          }
        },
        "genBasic": {
          "attributes": {
            "65503": "U\u0012O/\u0012",
            "65506": 56,
            "65508": 0,
            "65534": 0,
            "modelId": "TS0003",
            "manufacturerName": "_TZ3000_mw1pqqqt",
            "powerSource": 1,
            "zclVersion": 3,
            "appVersion": 74,
            "stackVersion": 0,
            "hwVersion": 1,
            "dateCode": ""
          }
        },
        "genOnOff": {
          "attributes": {
            "onOff": 1,
            "onTime": 0,
            "offWaitTime": 0,
            "moesStartUpOnOff": 2
          }
        },
        "manuSpecificTuya_3": {
          "attributes": {
            "53248": 0,
            "53249": 0,
            "53250": 0,
            "53251": 0,
            "53252": 0,
            "53253": 0,
            "switchType": 0
          }
        }
      },
      "binds": [
        {
          "cluster": 6,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b002a1aa3c4",
          "endpointID": 1
        }
      ],
      "configuredReportings": [
        {
          "cluster": 6,
          "attrId": 0,
          "minRepIntval": 0,
          "maxRepIntval": 65000,
          "repChange": 1
        }
      ],
      "meta": {}
    },
    "2": {
      "profId": 260,
      "epId": 2,
      "devId": 256,
      "inClusterList": [
        4,
        5,
        6
      ],
      "outClusterList": [],
      "clusters": {
        "genOnOff": {
          "attributes": {
            "onOff": 1,
            "onTime": 0,
            "offWaitTime": 0
          }
        }
      },
      "binds": [
        {
          "cluster": 6,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b002a1aa3c4",
          "endpointID": 1
        }
      ],
      "configuredReportings": [
        {
          "cluster": 6,
          "attrId": 0,
          "minRepIntval": 0,
          "maxRepIntval": 65000,
          "repChange": 1
        }
      ],
      "meta": {}
    },
    "3": {
      "profId": 260,
      "epId": 3,
      "devId": 256,
      "inClusterList": [
        4,
        5,
        6
      ],
      "outClusterList": [],
      "clusters": {
        "genOnOff": {
          "attributes": {
            "onOff": 0,
            "onTime": 0,
            "offWaitTime": 0
          }
        }
      },
      "binds": [
        {
          "cluster": 6,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b002a1aa3c4",
          "endpointID": 1
        }
      ],
      "configuredReportings": [
        {
          "cluster": 6,
          "attrId": 0,
          "minRepIntval": 0,
          "maxRepIntval": 65000,
          "repChange": 1
        }
      ],
      "meta": {}
    },
    "242": {
      "profId": 41440,
      "epId": 242,
      "devId": 97,
      "inClusterList": [],
      "outClusterList": [
        33
      ],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 74,
  "stackVersion": 0,
  "hwVersion": 1,
  "dateCode": "",
  "zclVersion": 3,
  "interviewCompleted": true,
  "meta": {
    "configured": 332242049
  },
  "lastSeen": 1740395306651
}
```

mrmaximas, 2025-02-24:
> ### Link
>
> https://de.aliexpress.com/item/1005007444117700.html
>
> ### Model
>
> XMSJ
>
> ### Description
>
> Zigbee USB plug 2 Channel
>
> ### Vendor
>
> Tuya
>
> ### Picture (link) [...]

13robin37, 2025-04-26:
Issue still exists

ha-bnst, 2025-06-08:
Issue still exists

tabearunzheimer, 2025-07-18:
I've successfully added the 3 port usb and then tried to pair the 2 port version. Only the 3 port usb switch is listed in the device section in z2m, so there seems to be no support for the 1 or 2 port version. I used the dev console to extract the converter of the 3 port version and adapted it:
```mjs
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['TS0002'],
    model: 'TS0002-USB2',
    vendor: '_TZ3000_h1ipgkwn', 
    description: 'Custom Tuya 2-port USB switch',
    extend: [
        m.deviceEndpoints({endpoints: {'1': 1, '2': 2}}),
        m.onOff({powerOnBehavior: true, endpointNames: ['1', '2']})
    ],
    meta: {multiEndpoint: true},
};
```  
Add this as an `mjs` file to the `external_converters` folders and link `external_converters` in the z2m `configuration.yml`. Remove the 2 port usb device from z2m, then restart z2m and pair them again.

This works to control the usb ports, however I gues this overwrites the original `TS0002`. I can't test it because I dont have that type of switch. I also don't know how to get the actual values for zigbeeModel, model and vendor. Also, the vendor shown in e.g. Homeassistant will be `_TZ3000_h1ipgkwn`.
Maybe someone more proficient with the topic can help.

tabearunzheimer, 2025-08-01:
@13robin37 Check the [latest release](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.6.0). It's part of the supported devices now. If it works, close the issue please.