Koenkk/zigbee2mqtt #26123: Configuration error after updating to 2.1.0

jbjardine, 2025-02-02:
### What happened?

Just updated to version 2.1.0 today and encountered a crash.

Here is the output:
'[13:30:19] INFO: Preparing to start...
[13:30:19] INFO: Socat not enabled
[13:30:20] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
            READ THIS CAREFULLY
Refusing to start because configuration is not valid, found the following errors:
- advanced/log_level must be equal to one of the allowed values
If you don't know how to solve this, read https://www.zigbee2mqtt.io/guide/configuration
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!'

I have not modified the configuration, which was originally set as follows:
'advanced:
  log_level: warn'

After switching to 'info', everything worked fine.

Is this behavior expected or is it a bug introduced in the 2.1.0 update? It is easy to debug for advanced users, but not very user-friendly for basic users.

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

7.4.4 [GA]

### Adapter

EmberZNet

### Setup

Add-on on Home Assistant (Yellow)

chris-1243, 2025-02-02:
`warn` has been removed. See #25634.
It is part of the release note of 2.1.0.

You may use:
 - debug
 - info
 - warning
 - error

jbjardine, 2025-02-02:
@chris-1243 I totally agree with the removal in 2.1.0—I understand the reasoning. However, could we perhaps include an automatic fix for this obsolete parameter? That way, non-expert users wouldn’t have to manually edit Home Assistant files, keeping the update truly plug-and-play. 😄👍

Looking forward to your thoughts!

chris-1243, 2025-02-02:
@jbjardine 
I am not an expert user at all and I understand it might be frustrating when an error occurs.
This a minor correction to do in your `configuration.yaml` and by far easiest than the migration from 1.x.x to 2.x.x.
It was a word to suppress in a file for the dev-team which is doing a hard job to keep everything as clean as possible. We, as users, should also try to learn a bit how things work...
By the way, the error given by Z2M is well detailed with a link for a solution. That is fair enough for me (only😉)

closed 2025-04-11 by github-actions[bot]