Koenkk/zigbee2mqtt #26124: Tuya RB-SRAIN01 no longer updates in 2.1.0

JSVHolmes, 2025-02-02:
### What happened?

Tuya RB-SRAIN01 was previously updating values as normal up to v2.0.0, however after updating to v2.1.0 the device still updates last seen every minute, but none of the other values update anymore. Reverting back to v2.0.0 restores all values back to updating as normal.

I am using Zigbee2MQTT via a Contianer and using Home Assistant 2025.1.4 (also in Docker) to record the values.

It is possible that the "Reportings" have been wiped, although this seems to be the same on the older version, under "manuSpecificTuya" there are no attributes listed for this device.

### What did you expect to happen?

The same behaviour as before, Illuminance and rain sensors should update.

### How to reproduce it (minimal and precise)

Have the Tuya RB-SRAIN01 sensor under version 2.0.0 and update to 2.1.0, the values on the device no longer update

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

20230507

### Adapter

SONOFF Universal Zigbee 3.0 USB

### Setup

Docker Container

Malte42, 2025-02-02:
Same Probleme here, but not in docker. Except the rain sensor works for me  but iliminance, rain intensity and battery not updating only showing null.

Sergey7up, 2025-02-02:
Confirmed. After upgrading to Zigbee2MQTT 2.1.0, all reported values from the TS0207 [RB-SRAIN01] (_TZ3210_tgvtvdoc) device remain frozen.

Troubleshooting Steps Taken:
Attempted re-pairing the device
Reconfigured device
Observed that only the "link quality" and "rain true/false" updates, even when pressing "Interview".

Home assistant OS 
Core 2025.1.4
Supervisor 2024.12.3
Operating System 14.2

SargonofAssyria, 2025-02-02:
Same here, going back to 2.0.0 everything works with this device.

cedarearthy, 2025-02-02:
Same problem here

keidischoi, 2025-02-03:
Same problem too

Murianwind, 2025-02-03:
Same problem too

pantherale0, 2025-02-03:
Same issue here

SargonofAssyria, 2025-02-03:
I went back to 2.0.0-dev, it works fine initially but after half a day the device does not update anymore.
Will try to include it again and see what happens.

andrazek, 2025-02-03:
I've also noticed that the device stops reporting after a few days or even sooner. However, I didn't experience these issues between November and mid-January. I've now moved it closer to the coordinator to see if distance was a factor, though the LQI has consistently remained above 100.

pantherale0, 2025-02-03:
I don't think the distance is a problem, mine is much closer to my coordinator and multiple routers than some other devices on my ZB network that are working fine.

Generally I don't experience any problems with this. I've restored back and no problems so far.

Marbie2024, 2025-02-03:
I got the same Problems. I've seen the following errors for the device:

Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "42613".

Sergi-CP, 2025-02-03:
Same problem here. After back to version 2.0
O all started to works fine.

jpbede, 2025-02-03:
Ok, I found the problem with this and probably some other Tuya devices. This line https://github.com/Koenkk/zigbee-herdsman-converters/commit/d7f9a3c1aeccd1cd3e912ebe07fa8b337fbcda3d prevents Zigbee packets for Tuya devices from being reprocessed if the sequence ID is the same, but the rain sensor always sends its data with the same sequence ID:

```
[2025-02-03 19:50:39] debug: 	zh:controller: Received payload: clusterID=61184, address=52189, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=116, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":153,"commandIdentifier":6},"payload":{"seq":256,"dpValues":[{"dp":105,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,241]}}]},"command":{"ID":6,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"activeStatusReport"}}

[2025-02-03 19:51:39] debug: 	zh:controller: Received payload: clusterID=61184, address=52189, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=116, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":156,"commandIdentifier":6},"payload":{"seq":256,"dpValues":[{"dp":105,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,246]}}]},"command":{"ID":6,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"activeStatusReport"}}
```

crguez, 2025-02-03:
Same issue here. Any solution?

reylinux, 2025-02-04:
Also notice this behavior after the update. Tested and working fine on the SmartLife app, but not in HA.

francouai, 2025-02-05:
Same problem here.
waiting for a new update.

jpbede, 2025-02-05:
Should be fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/8723

Is currently available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) or in the next release which is every 1st of the month.

adamlc, 2025-02-05:
> Let me know if it works in the dev branch, I'll create a new z2m release after that.
> 
> > Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

It's working great for me using the dev branch 🙌

Koenkk, 2025-02-05:
Thanks, I will release z2m 2.1.1 with a fix soon!

closed 2025-02-05 by Koenkk

josemaX, 2025-02-09:
After update z2m to 2.1.1 this device worked fine a day or so, then stopped updating.

I've restarted the Home-Assistant machine (HAOS) and nothing (new) happened.

If you want me to make some tests or get some info, i'm available to check.

Thanks!

Sergi-CP, 2025-02-09:
Hello, here also, but after Home Assistant Core the problem appears. Or doesn't update or values have strange behaviour.

Malte42, 2025-02-09:
Same problem here working for a few days then this morning it stopt working with the same problems as bevor

Sergi-CP, 2025-02-13:
Hi, any news? Device is freezed. Can you reopen the problem please?

SargonofAssyria, 2025-02-13:
With 2.1.1-dev commit: 16cecd91 The device works without any problems for more then 6 days now.

Sergi-CP, 2025-02-13:
Thanks a lot for your quick responses and support, I've installed Edge add-on and my device started again to report.

10der, 2025-03-31:
Rain
Indicates whether the device detected rainfall
False (always)

Cleaning reminder
Cleaning reminder
True (always)

Zigbee2MQTT version
[2.1.3](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.1.3)

joaopedros2, 2025-04-02:
i having the same problem, the only sensor updating is 'rain_intensity'

lynara, 2025-04-19:
Hi, for me it randomly starts and stops updating :/

luismipr, 2025-04-25:
Same Problem here. its randomly offline:

[image]

LQI is about 100lqi but ramdomly it appears offline.

josemaX, 2025-04-26:
Hello

I Had this problem and if I left the device outside home, it disconnected in a few hours, but if I left it near the coordinator, It worked without interruption.

Now, I've replaced my Conbee II coordinator with SLZB-06 (installed in the same position the Conbee was) and it worked more than a week just until yesterday that I needed to reboot SLZB because nothing worked after a Home Assistant update (2025.4.4) and since then, the SRAIN01 is not updating, so I'll try to repair later today.

jonathh21, 2025-09-15:
I've had one of these for about 6 months.

I can't say nothing else has changed, but the location this device resides at hasn't and the position of devices they existed when it was added hasnt either.

ATM is flaky as anything, repairing gets it "seen" but no updates.

I can't work out why.