Koenkk/zigbee2mqtt #26128: TS0601_MIR-HE200-TY presence state does not work properly after updating to 2.1.0

pbartek20, 2025-02-02:
### What happened?

After updating to 2.1.0 the TS0601_MIR-HE200-TY Human Presence Sensor has issues with proper showing/updating the presence status.

Once updated when I entered to the bathroom it changed status to True (as it should) but when I left the room it stayed in this state and not updated after again. It is showing infinite "True". Tried restartin Z2M, restarting HA, deleting the device and pair it again, cutting power to the sensor but nothing mentioned helps.
Below the exact function I referee.

[image]

Only when I reverted the VM backup with 2.0.0 version which I had befor upgrade it works perfectly fine. The status of presence is updating properly after.
Then tried once again update to 2.1.0 and issue happened again which should indicate problem with this version.

### What did you expect to happen?

I expected that above will be fixed.

### How to reproduce it (minimal and precise)

Update to 2.1.0, enter the room where the sensor is and left the room.

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Setup

Add-on on Home Assisnant OS on Proxmox VM

brot123, 2025-02-02:
I have also the presence Error with a _TZE200_lu01t0zl
For me also Stuck the illuminance(every time on 350lx) presence, and movement direction. Occupancy works but very slow detection.

cdemi, 2025-02-03:
Having the same problem with `MIR-HE200-TY`

mescalum, 2025-02-03:
Similar Issue with TS0601 where Presence stays unknown but Occupancy works. Reverting back to 2.0.0 fixes the issue.

Koenkk, 2025-02-03:
Could you provide the debug of presence not detected 2.1.0 (but it should) and when presence is detected on 2.0.0?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Koenkk, 2025-02-05:
It should be working now, can you try?

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

brot123, 2025-02-07:
Thanks for upload the fix, now it works! Thanks at all

closed 2025-02-07 by Koenkk