Koenkk/zigbee2mqtt #26129: Error while starting zigbee-herdsman

ashkii7, 2025-02-02:
### What happened?

My configuration:
RPI4b Debian 11 bullseye
Adapter: Raspbee 2 deconz with firmware deCONZ_RaspBeeII_0x26780700.bin.GCF

pi@RPI-PORTAIL:~ $ GCFFlasher -l
Path              | Serial      | Type
------------------+-------------+---------------
/dev/ttyS0        |             | RaspBee

Zigbee2MQTT version 2.1.0 (commit #5e9d53f)

my configuration file:
 homeassistant:
  enabled: false
permit_join:
  enabled: true
frontend:
  enabled: true
mqtt:
  base_topic: z2m-portail
  server: '!secret.yaml server'
  ca: /etc/ssl/certs_intel-nuc/maeva56_ca.crt
  key: /etc/ssl/certs_intel-nuc/portail_client.key
  cert: /etc/ssl/certs_intel-nuc/portail_client.crt
  user: '!secret.yaml user'
  password: '!secret.yaml password'
  reject_unauthorized: false
serial:
  port: /dev/ttyS0
  adapter: deconz
  baudrate: 38400
  adapter_delay: 200
advanced:
  channel: 20
  network_key:
    - X
    - Y
    - etc.
  pan_id: 21467
  ext_pan_id:
    - 124
    - 13
    - 73
    - 41
    - 36
    - 171
    - 192
    - 219
  log_level: debug
  log_file: log_%TIMESTAMP%.log
  log_rotation: true
  log_output:
    - console
    - file
zigbee_herdsman_debug: true

### What did you expect to happen?

Starting Z2M and just working fine
I specify that in the frontend no parameters are displayed

[image]

Sorry for my english, please help me, i am lost ...

### How to reproduce it (minimal and precise)

I install z2m like https://www.zigbee2mqtt.io/guide/installation/01_linux.html#installing
When I launch the 'pnpm start' command, Z2M does not start and no file 'database.db' is created in '/opt/zigbee2mqtt/data/'

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

26780700

### Adapter

Deconz Raspbee 2

### Setup

On RPI4b with Debian 11 bullseye

### Debug log

pi@RPI-PORTAIL:/opt/zigbee2mqtt $ pnpm start

> zigbee2mqtt@2.1.0 start /opt/zigbee2mqtt
> node index.js

Starting Zigbee2MQTT without watchdog.
[2025-02-02 15:18:21] info:     z2m: Logging to console, file (filename: log_2025-02-02.15-18-21.log)
[2025-02-02 15:18:21] debug:    z2m: Loaded state from file /opt/zigbee2mqtt/data/state.json
[2025-02-02 15:18:21] info:     z2m: Starting Zigbee2MQTT version 2.1.0 (commit #5e9d53f)
[2025-02-02 15:18:21] debug:    z2m: sd-notify loaded
[2025-02-02 15:18:21] info:     z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-02 15:18:21] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":21467,\"extendedPanID\":[124,13,73,41,36,171,192,219],\"channelList\":[20],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/opt/zigbee2mqtt/data/database.db\",\"databaseBackupPath\":\"/opt/zigbee2mqtt/data/database.db.backup\",\"backupPath\":\"/opt/zigbee2mqtt/data/coordinator_backup.json\",\"serialPort\":{\"baudRate\":38400,\"path\":\"/dev/ttyS0\",\"adapter\":\"deconz\"},\"adapter\":{\"disableLED\":false}}"'
[2025-02-02 15:18:22] debug:    zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyAMA0"},{"path":"/dev/ttyS0"}]
[2025-02-02 15:18:22] debug:    zh:adapter:discovery: Unable to match USB adapter: deconz | /dev/ttyS0
[2025-02-02 15:18:22] debug:    zh:deconz:driver: Set delay to 0
[2025-02-02 15:18:22] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":21467,"extendedPanID":[124,13,73,41,36,171,192,219],"channelList":[20]},"serialPort":{"baudRate":38400,"path":"/dev/ttyS0","adapter":"deconz"},"adapter":{"disableLED":false},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json"}'
[2025-02-02 15:18:22] debug:    zh:deconz:driver: Opening with /dev/ttyS0
[2025-02-02 15:18:22] debug:    zh:deconz:driver: Serialport opened
[2025-02-02 15:18:22] debug:    zh:deconz:driver: send read parameter request from queue. seqNr: 1 paramId: 5
[2025-02-02 15:18:32] debug:    zh:deconz:driver: Timeout for request - CMD: 0xa seqNr: 1
[2025-02-02 15:18:32] error:    z2m: Error while starting zigbee-herdsman
[2025-02-02 15:18:32] error:    z2m: Failed to start zigbee-herdsman
[2025-02-02 15:18:32] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-02-02 15:18:32] error:    z2m: Exiting...
[2025-02-02 15:18:32] error:    z2m: Error: TIMEOUT
    at Driver.processBusyQueue (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:508:28)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:160:46)
    at listOnTimeout (node:internal/timers:581:17)
    at processTimers (node:internal/timers:519:7)
 ELIFECYCLE  Command failed with exit code 1.

chris-1243, 2025-02-02:
> [2025-02-02 16:36:18] debug: zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyAMA0"},{"path":"/dev/ttyS0"}]
[2025-02-02 16:36:18] debug: zh:adapter:discovery: Unable to match USB adapter: deconz | /dev/ttyS0

I would check carefully the port of your adapter. It seems, according to your log, that Z2M can not communicate with your coordinator.
Try to use the `/dev/serial/by-id/your_adapter_serial` instead

ashkii7, 2025-02-03:
Thank you for your answer.
However the "/dev/serial/by-ID" repertoire does not exist on my RPI4.
For proof:
`pi@RPI-PORTAIL:~ $ ls -l /dev/serial/by-id
ls: impossible d'accéder à '/dev/serial/by-id': Aucun fichier ou dossier de ce type
pi@RPI-PORTAIL:~ $ ls -l /dev/serial
ls: impossible d'accéder à '/dev/serial': Aucun fichier ou dossier de ce type`
I only have this:
`pi@RPI-PORTAIL:/opt/zigbee2mqtt $ ls -l /dev/s*
brw-rw---- 1 root disk  8, 0 25 août  21:05 /dev/sda
brw-rw---- 1 root disk  8, 1 25 août  21:05 /dev/sda1
brw-rw---- 1 root disk  8, 2 25 août  21:05 /dev/sda2
lrwxrwxrwx 1 root root     5 25 août  21:05 /dev/serial0 -> ttyS0
lrwxrwxrwx 1 root root     7 25 août  21:05 /dev/serial1 -> ttyAMA0
crw-rw---- 1 root disk 21, 0 25 août  21:05 /dev/sg0
lrwxrwxrwx 1 root root    15 25 août  21:05 /dev/stderr -> /proc/self/fd/2
lrwxrwxrwx 1 root root    15 25 août  21:05 /dev/stdin -> /proc/self/fd/0
lrwxrwxrwx 1 root root    15 25 août  21:05 /dev/stdout -> /proc/self/fd/1`
I also tried with it:
`/dev/serial0`
But nothing done, it's the same!

chris-1243, 2025-02-03:
Are you sure your raspbee is connected to `/dev/ttyS0`?

Have you tried to use `/dev/ttyAMA0`? 

Also and according to a really fast Google research, try to disconnect everything from your raspberryPI. Users seem to report conflicts when an usb device is connected at the same time...

ashkii7, 2025-02-03:
Are you sure your raspbee is connected to /dev/ttyS0? -->  I don't know, how to know? (i am a very basic user!)

Same result with `/dev/ttyAMA0`

On my RPI are connected in:
-USB3: my RPI start-up SSD --> Unable to disconnect it otherwise the RPI would no longer work!
-USB2: A Bluetooth key --> Associated with a Blea antenna from the 'Blea advertisement' plugin on Jeedom who is installed on another RPI in my house.
For your info, the set is installed in a waterproof box near my entrance gate located 20 m from my house.

With my Google research I also saw that there were possible conflicts but I cannot disconnect the other USB devices. Otherwise I lose the associated features.

chris-1243, 2025-02-03:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log

You have to take care of interference between UBS3.0 and zigbee protocol. SSD are well known to cause lots of trouble and your raspbee is just above your usb ports...

In your scenario, I would go for an usb dongle with a 1m usb2.0 cable or a lan coordinatoor. You would solve many problems. Even if the `deconz` driver is supported, you do not have all the feature like the coordinator backup... see https://www.zigbee2mqtt.io/guide/adapters/deconz.html

I know it is not the answer you are looking for. I would definitely check in `dresden` github for any advices. Definitely a connection issue between your coordinator and Z2M.

ashkii7, 2025-02-03:
Thank you for your answer and the first link.
In my configuration to avoid these interference, I will look for now:
- either deporting in another waterproof box a USB Deconz Combee 2 key, but that does not enchant me,
- Or use a ZigBee SMLIGHT coordinator - Ethernet USB adapter EFR32MG21. This would have the advantage of being connected to the LAN and of being integrated into my current shelf box, so I should no longer be generous by interference. What do you think of this solution?

chris-1243, 2025-02-03:
> Or use a ZigBee SMLIGHT coordinator - Ethernet USB adapter EFR32MG21. This would have the advantage of being connected to the LAN and of being integrated into my current shelf box, so I should no longer be generous by interference

Accordimg to the small description of your setup, I would go for a LAN solution. Read carefully this [link](https://www.zigbee2mqtt.io/advanced/zigbee/02_improve_network_range_and_stability.html) as well. Wifi and Zigbee are both in 2.4GHz bands... set carefully your zigbee channel!

EFR32MG21/24 or CC2652P7 chips are well supported in Z2M and are a good choice. I am using a CC2652P2 chip (USB) for quite a long time without any particular issues.

closed 2025-02-05 by ashkii7