Koenkk/zigbee2mqtt #26132: Tuya TS0042 no longer works after update to 2.1.0

mlepannetier, 2025-02-02:
### What happened?

After update to 2.1.0 :

 2025-02-02 18:12:25z2m: Failed to configure '0xa4c1383b88516d8d', attempt 1 (Error: ZCL command 0xa4c1383b88516d8d/1 genBasic.read(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:545:23) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:642:24) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:642:24) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:642:24) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:642:24) at /app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:443:20 at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:441:16) at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20) at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:324:20))

Two devices Tuya TS0042 no longer works

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

20220219

### Adapter

zStack3x0

### Setup

docker

### Debug log

 2025-02-02 18:12:25z2m: Failed to configure '0xa4c1383b88516d8d', attempt 1 (Error: ZCL command 0xa4c1383b88516d8d/1 genBasic.read(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:545:23) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:642:24) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:642:24) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:642:24) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:642:24) at /app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:443:20 at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:441:16) at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20) at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:324:20))

eluijer, 2025-02-04:
Same issue with the Tuya TS0044 (4 button switch)

unixtall, 2025-02-12:
In my case, the device ts0044 is discovered but actions have disappeared

Tjilie, 2025-02-16:
Same issue here for Tuya and Aqara (Xiaomi) switches

The update from Zigbee2MQTT 1.42.0 to 2.1.1 went well. Everything works perfectly in Domoticz, except for my push-button battery switches. In the 'Last Seen' section of the devices, I can see new times for the battery level, but not for the push-buttons themselves; these remain at the times before the update. How is it that my push-button switches are not working after the update to 2.1.1?

I roll back to 1.42.0 all oke

Tjilie, 2025-02-22:
I tested it and it works with domoticz beta version