Koenkk/zigbee2mqtt #26133: Smartwings WM25/L-Z Roller shade open/close commands reversed in 2.1.0

brettmiller, 2025-02-02:
### What happened?

After the 2.1.0 my Smartwings WM25/L-Z Roller shades no longer respond to open and close commands as expected. Sending close opens the shades and open closes the shades. They still report the correct open/close state and percentages. Changing the "Invert cover" setting from false to true doesn't change the behavior. 

### What did you expect to happen?

Sending "open" opens the shades and sending "close" closes the shades. 

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Setup

Add-on on Home Assistant OS on Proxmox VM

ThrashinVictim, 2025-02-03:
I saw this as well. 
I have another problem. 
I  have 4 WM25/L-Z Roller shades

Before this change, 1 blind worked fine but for the other 3, I had to do the invert for Alexa and set it as inverted in Z2M in the other settings tab. 
More info here https://github.com/Koenkk/zigbee2mqtt/issues/22847#issuecomment-2335290877
But in HA it would say the blinds are open but the slider shows as closed. 

Now with this change, the 3 that didn't work do, and the 1 that did work now doesn't show the correct state. 

I think it is due to me buying the blinds at different times and they have different firmware. The first one I bought was in March and the rest 3 was in May. But I am speculating.

brettmiller, 2025-02-03:
For now I took the file from the [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8681) and corrected it to have `coverInverted: false` in the "meta" section like what's in the external converter in   https://github.com/Koenkk/zigbee2mqtt/issues/22847#issuecomment-2335290877. I also flipped the motor direction back to what the factor default, but now the buttons on the remote are backwards and they can't be fixed without factory reseting the shades since the buttons on the remote can't be flipped after the shade limits have been set. I happed to have factory reset one of my shades while debugging and the remote buttons for that shade are correct.

denmoses1, 2025-02-04:
Same issue here.   Though I don’t mind flipping the app direction back to the factory setting…the mismatch between the state and open/close is a more significant problem.   
Has anyone figured out a combination of app flip + mqtt settings that can resolve the issue?

amaisano, 2025-02-04:
With the current build, set so up on remote is up on the shade (using the remote programming). Set invert in z2m to TRUE.

That's what got it working for me which is fine... for my remaining shades.

denmoses1, 2025-02-04:
> With the current build, set so up on remote is up on the shade (using the remote programming). Set invert in z2m to TRUE.
> 
> That's what got it working for me which is fine... for my remaining shades.

Thanks
I might be missing something here.   When I set it up that way, the open and close work perfectly.   But the state doesn’t.  Meaning if I open it to 5% it will physically open the window 95%.

henryjarend, 2025-02-04:
I'm having the same problem where my open and closed are reversed in the Z2M UI but in Home Assistant things seem to understand what's going on. However I can't perform any actions using the "open" or "close" buttons and instead have to set the percent open using the slider or setting "100%" in an automation. It appears that Z2M understands the percentage but has some issues understanding "open" and "close"
From the logs after setting position to 100%, Note that the state says open
`[2025-02-04 14:13:52] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/middle_loft_blind', payload '{"battery":27,"linkquality":104,"position":100,"state":"OPEN"}'` 
after I hit the "open" button in Z2M
`[2025-02-04 14:14:18] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/middle_loft_blind', payload '{"battery":27,"linkquality":104,"position":0,"state":"CLOSE"}'`

brettmiller, 2025-02-04:
Because of an inconsistence introduced in the code there are no settings changes or shade configuration changes that will make things work correctly. Until the code is fixed or the change is reverted the only solution is to use an external converter. I created and external converter based on the [file](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8681/files) from the PR that caused this and modified line 30 to have `coverInverted: false` instead of `coverInverted: true`. 
```
import fz from '../converters/fromZigbee';
import tz from '../converters/toZigbee';
import * as exposes from '../lib/exposes';
import * as reporting from '../lib/reporting';
import {DefinitionWithExtend, Tz, Zh} from '../lib/types';
import {assertString, getFromLookup, getOptions} from '../lib/utils';

const tzLocal = {
    backwards_cover_state: {
        key: ['state'],
        convertSet: async (entity: Zh.Endpoint, key: string, value: number | string, meta: Tz.Meta) => {
            const lookup = {open: 'downClose', close: 'upOpen', stop: 'stop', on: 'downClose', off: 'upOpen'};
            assertString(value, key);
            value = value.toLowerCase();
            await entity.command('closuresWindowCovering', getFromLookup(value, lookup), {}, getOptions(meta.mapped, entity));
        },
    },
};

const e = exposes.presets;

const definitions: DefinitionWithExtend[] = [
    {
        zigbeeModel: ['WM25/L-Z'],
        model: 'WM25L-Z',
        vendor: 'Smartwings',
        description: 'Roller shade',
        fromZigbee: [fz.cover_position_tilt, fz.battery],
        toZigbee: [tzLocal.backwards_cover_state, tz.cover_position_tilt],
        meta: {battery: {dontDividePercentage: true}, coverInverted: false},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'closuresWindowCovering']);
            device.powerSource = 'Mains (single phase)';
            device.save();
        },
        exposes: [e.cover_position(), e.battery()],
    },
];
```

To do this in Home Assistant (at least on HAOS) create the directory `/homeassistant/zigbee2mqtt/external_converters` if it doesn't exist. Then in that directory create a `smartwings.js` file with the code above and restart zigbee2mqtt. If you had originally reversed the motor direction of your shade you will probably need to reverse it back. It's possible that you may be able to instead set the invert toggle to true in z2m under the devices "Settings (specific)" tab, but I have not tested that.

amaisano, 2025-02-04:
IMO, because it was working with document programming instructions ([see the Z2M notes I actually contributed a while back](https://github.com/Koenkk/zigbee2mqtt.io/commit/da2bd5404435fadddd661a7a392a37d5ce30adce)), I vote this gets reverted, at least until we have a solid fix with no inconsistencies.

ThrashinVictim, 2025-02-04:
I think this is dependent on what firmware your blinds have. I have 3 that didn't work and 1 that did work correctly and now with the change I have 3 that work and 1 that doesn't.
I purchased the first set of blinds 4 months before the other 3 so I am sure they are different firmware. (Is there a way to find out?)

denmoses1, 2025-02-04:
> IMO, because it was working with documented programming instructions ([see the Z2M notes I actually contributed a while back](https://github.com/Koenkk/zigbee2mqtt.io/commit/da2bd5404435fadddd661a7a392a37d5ce30adce)), I vote this gets reverted, at least until we have a solid fix with no inconsistencies.

+1
Totally agree.

brettmiller, 2025-02-04:
The change has reverted, https://github.com/Koenkk/zigbee-herdsman-converters/pull/8681#issuecomment-2634935362, but won't be a release until the normal z2m release cycle the first of next month. So, it get it fixed you'll either need to switch to the dev branch (link in that comment) or use the external converter above.

closed 2025-02-05 by Koenkk

brettmiller, 2025-02-06:
For those following along. The fix for this, reverting the change, is in z2m 2.1.1 that was just release.

amaisano, 2025-02-06:
Heh, time to go around and re-hack the shades and flip the zigbee settings 🙃

ThrashinVictim, 2025-02-06:
This fixed the old shades I had and broke the 3 that the other revision fixed. So I have a choice to stick with the 2.1.0 and have three that work correctly with Alexa and HA (showing correct state) and have one that doesn't Or go with 2.1.1 and have 3 that show incorrect state and 1 that does.

Is there a way to use a converter for just one set of blinds or is it an all or nothing?

amaisano, 2025-02-06:
@ThrashinVictim yes - see https://github.com/Koenkk/zigbee2mqtt/issues/26133#issuecomment-2634520945 above

ThrashinVictim, 2025-02-06:
@amaisano Those instructions would affect all 4 sets of shades which would still put me in the situation I am in. I just want to point the converter to 1 set of shades while leaving the others with the default settings.

amaisano, 2025-02-07:
Apologies - I misread the question 😬 - I'm personally not sure how to do that, unless you have two zigbee/z2m instances.

dogtreatfairy, 2025-04-30:
This should probably be reopened. I just bought a SmartWings Zigbee shade and no amount of reversing the motor, converter scripts, or combination of available settings has made my Open/Closed commands in Home Assistant agree with the Open/Closed buttons on my 5 Channel remote. Multiple resets, repairing, resetting upper and lower limits, etc has all been tried.