Koenkk/zigbee2mqtt #26138: Failed migration to new maachine

pflavio, 2025-02-02:
### What happened?

I moved my Zigbee2MQTT from one machine to another (docker to LXC container), keeping the same Sonoff stick and following [this short paragraph](https://www.zigbee2mqtt.io/guide/faq/#how-do-i-migrate-from-one-adapter-to-another) copying everything in ```/data``` to the new installs ```/data``` directory (replacing everything inside the folder). I then went into the ```configuration.yaml``` file (my old config copied to the new location) and changed the mqtt and serial sections. After starting up Zigbee2MQTT, the logs indicated that it wouldn't start unless I deleted ```coordinator_backup.json```, which I did and then it started fine. Unfortunately however, none of my devices are visible.

### What did you expect to happen?

I thought all my paired devices would appear in the web UI exactly as they were.

### How to reproduce it (minimal and precise)

For server migration
Copy old ```/data``` directory
Delete ```/data``` directory of new install
Paste old ```/data``` directory into new location

For startup
Start Zigbee2MQTT -> won't start
delete ```coordinator_backup.json```
Start Zigbee2MQTT -> starts fine (but no devices)

### Zigbee2MQTT version

2.1.0 

### Adapter firmware version

20230507

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

LXC container on Proxmox on x86-64

### Debug log

[log.log](https://github.com/user-attachments/files/18634398/log.log)

pflavio, 2025-02-03:
So I tried it again with a fresh install of Zigbee2MQTT in an LXC container, again deleting the entire ```data``` directory and replacing it with the one from my previous machine. Now I made the following changes to the ```configuration.yaml``` file, saved and started zigbee2mqtt:

```mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.1.144:1883
  user: redacted
  password: redacted
  keepalive: 60
  reject_unauthorized: true
  version: 4
serial:
  adapter: zstack
  port: >-
    /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_fc88b22c6fd8ed119da16f6162c613ac-if00-port0```

I'm getting this output again, which is before making further changes to the config so it runs. This time I won't delete ```coordinator_backup.json```, instead it be great if someone could tell me what changes to make to preserve my devices and set-up. As written above all devices and the adapter are exactly the same, the only thing that changed is zigbee2mqtt moved from a docker container on machine 1 to an LXC container on machien 2 (and the adapter moved with it, but it looks like it gets detected fine).

```
> zigbee2mqtt@2.1.0 start
> node index.js

Starting Zigbee2MQTT without watchdog.
[2025-02-03 13:36:20] info: 	z2m: Logging to console, file (filename: log.log)
[2025-02-03 13:36:20] info: 	z2m: Starting Zigbee2MQTT version 2.1.0 (commit #unknown)
[2025-02-03 13:36:20] info: 	z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-03 13:36:20] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_fc88b22c6fd8ed119da16f6162c613ac-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-02-03 13:36:20] info: 	zh:zstack:znp: Serialport opened
[2025-02-03 13:36:20] error: 	zh:adapter:zstack:manager: Configuration is not consistent with adapter state/backup!
[2025-02-03 13:36:20] error: 	zh:adapter:zstack:manager: - PAN ID: configured=26718, adapter=31740
[2025-02-03 13:36:20] error: 	zh:adapter:zstack:manager: - Extended PAN ID: configured=6bdfe8f9f65d9e4a, adapter=0102030405060708
[2025-02-03 13:36:20] error: 	zh:adapter:zstack:manager: - Network Key: configured=f6169005ef28f3ed19b84dc431e70ff2, adapter=87cd2db7c63239066c8430e21d92275b
[2025-02-03 13:36:20] error: 	zh:adapter:zstack:manager: - Channel List: configured=11, adapter=11
[2025-02-03 13:36:20] error: 	zh:adapter:zstack:manager: Please update configuration to prevent further issues.
[2025-02-03 13:36:20] error: 	zh:adapter:zstack:manager: If you wish to re-commission your network, please remove coordinator backup at /opt/zigbee2mqtt/data/coordinator_backup.json.
[2025-02-03 13:36:20] error: 	zh:adapter:zstack:manager: Re-commissioning your network will require re-pairing of all devices!
[2025-02-03 13:36:20] error: 	z2m: Error while starting zigbee-herdsman
[2025-02-03 13:36:20] error: 	z2m: Failed to start zigbee-herdsman
[2025-02-03 13:36:20] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-02-03 13:36:20] error: 	z2m: Exiting...
[2025-02-03 13:36:20] error: 	z2m: Error: startup failed - configuration-adapter mismatch - see logs above for more information
    at ZnpAdapterManager.determineStrategy (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:252:35)
    at ZnpAdapterManager.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:62:26)
    at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:158:16)
    at Controller.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:69:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:142:13)
    at start (/opt/zigbee2mqtt/index.js:161:5)```

pflavio, 2025-02-03:
I figured it out! I'll leave this post up in case anyone ever googles a similar issue. In total four files were needed from my old machine: ```configuration.yaml```, ```coordinator_backup.json```, ```database.db``` and ```state.json```. Only the first two needed to be modifed though. I think tipping this all out helped me realize what the issue was.

pflavio, 2025-02-03:
I closed the issue prematurely. Even though all devices are visible in Zigbee2MQTT, I cannot interact with them. I'm receiving ```z2m: Failed to execute LQI for 'inser random device'``` errors for all my devices and in Home Assistant, their respective switches are greyed out. Any pointers?

pflavio, 2025-02-03:
I  also needed to include the ```"devices":``` section from ```coordinator_backup.json```. Now Zigbee2MQTT works as expected and I can control my devices again.

closed 2025-02-03 by pflavio

peterhow11, 2025-02-26:
I am using the same ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus and was getting the same error. I solved it by removing the entire "serial:" section in the configuration.yaml file (see below)

serial:
  adaptor: zstack
  port: >-
    /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_004e3cc63239ec11a592e0680aac08d5-if00-port0

I assume that this let the device be auto-discovered and it works now.