Koenkk/zigbee2mqtt #26139: [New device support]: TS0601 _TZE200_jokk8yje Tuya 8-gang switch

rickzed, 2025-02-03:
### Link

https://item.taobao.com/item.htm?spm=a230r.1.14.23.48a65805j9HsKZ&id=680517215901&ns=1&abbucket=17#detail

### Database entry

{"id":6,"type":"Router","ieeeAddr":"0xa4c1381bc97d53b9","nwkAddr":59970,"manufId":4417,"manufName":"_TZE200_jokk8yje","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"9\u0017}*i","65506":53,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_jokk8yje","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b00257924db","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1659512439776,"defaultSendRequestWhen":"immediate"}

### Zigbee2MQTT version

2.10

### Comments

2022-08-03 11:22:23zigbee-herdsman-converters:tuya_data_point_dump: Received Tuya DataPoint https://github.com/Koenkk/zigbee2mqtt/issues/24 from 0xa4c1381bc97d53b9 with raw data '{"dp":24,"datatype":1,"data":{"type":"Buffer","data":[0]}}': type='commandDataReport', datatype='bool', value='false', known DP# usage: ["moesLocalTemp","tvLocalTemp","connecteLocalTemp"]

2022-08-03 11:22:42zigbee-herdsman-converters:tuya_data_point_dump: Received Tuya DataPoint https://github.com/Koenkk/zigbee2mqtt/issues/25 from 0xa4c1381bc97d53b9 with raw data '{"dp":25,"datatype":1,"data":{"type":"Buffer","data":[0]}}': type='commandDataReport', datatype='bool', value='false', known DP# usage: []

2022-08-03 11:20:51zigbee-herdsman-converters:tuya_data_point_dump: Received Tuya DataPoint https://github.com/Koenkk/zigbee2mqtt/issues/26 from 0xa4c1381bc97d53b9 with raw data '{"dp":26,"datatype":1,"data":{"type":"Buffer","data":[1]}}': type='commandDataReport', datatype='bool', value='true', known DP# usage: ["hochFaultCode"]

2022-08-03 11:21:25zigbee-herdsman-converters:tuya_data_point_dump: Received Tuya DataPoint https://github.com/Koenkk/zigbee2mqtt/issues/27 from 0xa4c1381bc97d53b9 with raw data '{"dp":27,"datatype":1,"data":{"type":"Buffer","data":[1]}}': type='commandDataReport', datatype='bool', value='true', known DP# usage: ["moesTempCalibration","saswellTempCalibration","tvTempCalibration","hochRelayStatus"]

2022-08-03 11:18:55zigbee-herdsman-converters:tuya_data_point_dump: Received Tuya DataPoint https://github.com/Koenkk/zigbee2mqtt/issues/5 from 0xa4c1381bc97d53b9 with raw data '{"dp":5,"datatype":4,"data":{"type":"Buffer","data":[0]}}': type='commandDataReport', datatype='enum', value='0', known DP# usage: ["fanMode","dimmerMaxLevel","motorDirection","config","moesSboostHeatingCountdown","neoAOVolume","silvercrestSetColor","lidlTimer","haozeeBoostHeatingCountdown","evanellLocalTemp","AM02Direction"]

 2022-08-03 11:19:30zigbee-herdsman-converters:tuya_data_point_dump: Received Tuya DataPoint https://github.com/Koenkk/zigbee2mqtt/issues/6 from 0xa4c1381bc97d53b9 with raw data '{"dp":6,"datatype":4,"data":{"type":"Buffer","data":[0]}}': type='commandDataReport', datatype='enum', value='0', known DP# usage: ["silvercrestSetEffect","haozeeRunningState","evanellBattery","tshpsSelfTest"]

2022-08-03 11:20:00zigbee-herdsman-converters:tuya_data_point_dump: Received Tuya DataPoint https://github.com/Koenkk/zigbee2mqtt/issues/7 from 0xa4c1381bc97d53b9 with raw data '{"dp":7,"datatype":4,"data":{"type":"Buffer","data":[0]}}': type='commandDataReport', datatype='enum', value='0', known DP# usage: ["childLock","coverChange","moesSreset","moesCoverBacklight","neoAODuration","moes105DimmerState2","haozeeWindowState","thitIlluminanceLux","AM02WorkState"]

2022-08-03 11:20:20zigbee-herdsman-converters:tuya_data_point_dump: Received Tuya DataPoint https://github.com/Koenkk/zigbee2mqtt/issues/8 from 0xa4c1381bc97d53b9 with raw data '{"dp":8,"datatype":4,"data":{"type":"Buffer","data":[0]}}': type='commandDataReport', datatype='enum', value='0', known DP# usage: ["moesSwindowDetectionFunktion_A2","moesCoverMotorReversal","saswellWindowDetection","wooxSmokeTest","moes105DimmerLevel2","tvWindowDetection","haozeeWindowDetection","thitTemperature","evanellChildLock","alectoSelfChecking"]

### External definition

```shell
No response
```

### What does/doesn't work with the external definition?

No response

Koenkk, 2025-02-03:
Could you try to:

- Stop Z2M
- Open `data/database.db` with a text editor and change all occurrences of `_TZE200_jokk8yje` to `_TZE200_vmcgja59`
- Start Z2M and see if the device now works

rickzed, 2025-02-04:
> Could you try to:
> 
> * Stop Z2M
> * Open `data/database.db` with a text editor and change all occurrences of `_TZE200_jokk8yje` to `_TZE200_vmcgja59`
> * Start Z2M and see if the device now works

sorry,can't work ,I described wrong, it's not an 8  gang switch，it's 4 gang switch with 4 scene control

uncle-wong, 2025-02-16:
> > Could you try to:
> > 
> > * Stop Z2M
> > * Open `data/database.db` with a text editor and change all occurrences of `_TZE200_jokk8yje` to `_TZE200_vmcgja59`
> > * Start Z2M and see if the device now works
> 
> sorry,can't work ,I described wrong, it's not an 8 gang switch，it's 4 gang switch with 4 scene control

我是一个4键的面板，双击后就切换为场景，我去涂鸦上找了dp点的定义，但是还是不太会写

closed 2025-04-26 by github-actions[bot]