Koenkk/zigbee2mqtt #26142: Boot Loop after Update

Ruko2010, 2025-02-03:
### What happened?

Hi, I user Zigbee2MQTT as a Docker Container in Unraid. Today I updated and now it is not working any more. Looking in the logs, I see this:

```
[2025-02-03 10:24:04] info:     z2m: Logging to console, file (filename: log.log)
[2025-02-03 10:24:04] info:     z2m: Starting Zigbee2MQTT version 2.1.0 (commit #unknown)
[2025-02-03 10:24:04] info:     z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-03 10:24:04] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
Starting Zigbee2MQTT without watchdog. 
```

This is constantly repeating after some secounds.
I can't reach the frontend of Z2M and also Home Assistant can't discover/control any devices.

This is my Configuration:

```
homeassistant:
  enabled: true
frontend:
  enabled: true
  port: 9442
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.10.245:1883
  keepalive: 60
  reject_unauthorized: false
  version: 4
serial:
  port: /dev/ttyACM0
  adapter: zstack
advanced:
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 514
    protocol: udp4
    type: '5424'
  last_seen: ISO_8601_local
device_options: {}
devices:
  '0x00124b002a533fab':
    friendly_name: LQM 2
  '0x00124b002a539fac':
    friendly_name: LQM 1
  '0x00124b002a53061d':
    friendly_name: LQM 3
  '0x00124b002931de01':
    friendly_name: Garagentor
  '0xf0d1b8be24091d8e':
    friendly_name: Couchlicht
  '0xa4c1383866dcf4f0':
    friendly_name: BadTV
  '0xa4c138b0b095c1c6':
    friendly_name: KücheTV
  '0xa4c13829b1c80f3f':
    friendly_name: GaragenRouter
version: 4
```

I also tryed it with the "ember" configuration, but it is also not working. Hope somebody can help me.

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

Since I can't reach the frontend, I have no idea

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2 

### Setup

Docker Container on Unraid

Ruko2010, 2025-02-03:
Small Update:
I changed the configuration back to

```
serial:
  port: /dev/ttyACM0
  adapter: ezsp
```

And Downgraded the version to 2.0.0, now it is working again. But can somebody help, why it is not working after the upgrade to 2.1.0?

Collectathon, 2025-02-04:
This has also happened to me when upgrading from 2.0.0 to 2.1.0. I have the same looping message. Downgrading back to 2.0.0 seems to have fixed it for now.

Ruko2010, 2025-02-25:
I tried it with Update 2.1.1, but no change. Still the boot loop. This is my Logfile, maybe it helps identifying the problem:

```
Starting Zigbee2MQTT without watchdog.
[2025-02-25 09:29:44] debug:    z2m: Removing old log directory '/config/log/2025-02-25.09-29-21'
[2025-02-25 09:29:44] info:     z2m: Logging to console, file (filename: log.log)
[2025-02-25 09:29:44] debug:    z2m: Loaded state from file /config/state.json
[2025-02-25 09:29:44] info:     z2m: Starting Zigbee2MQTT version 2.1.1 (commit #unknown)
[2025-02-25 09:29:44] debug:    z2m: sd-notify loaded
[2025-02-25 09:29:44] info:     z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-25 09:29:44] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/database.db\",\"databaseBackupPath\":\"/config/database.db.backup\",\"backupPath\":\"/config/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyACM0\",\"adapter\":\"ezsp\"},\"adapter\":{\"disableLED\":false}}"'
[2025-02-25 09:29:44] debug:    zh:adapter:discovery: Error while trying to match USB adapter (spawn udevadm ENOENT).
[2025-02-25 09:29:44] debug:    zh:ezsp: Adapter concurrent: 8
[2025-02-25 09:29:44] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0","adapter":"ezsp"},"adapter":{"disableLED":false},"databasePath":"/config/database.db","databaseBackupPath":"/config/database.db.backup","backupPath":"/config/coordinator_backup.json"}'
[2025-02-25 09:29:44] warning:  zh:ezsp: 'ezsp' driver is deprecated and will only remain to provide support for older firmware (pre 7.4.x). Migration to 'ember' is recommended. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462
[2025-02-25 09:29:44] debug:    zh:ezsp:uart: RTS/CTS config is off, enabling software flow control.
[2025-02-25 09:29:44] debug:    zh:ezsp:uart: Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false,"parity":"none","stopBits":1,"xon":true,"xoff":true}
```

Koenkk, 2025-02-25:
@Ruko2010 You are using the old `ezsp` driver, it's recommended to use the new `ember` drivers ([info](https://github.com/Koenkk/zigbee2mqtt/discussions/21462)), could you try that and see if the issue is fixed?

Ruko2010, 2025-02-26:
I already tried this, but is was not working. But, I think maybe it is because of an old firmware on my Stick:
Coordinator-Version 6.10.3.0 build 297

Could this be the reason?

Koenkk, 2025-02-27:
yes that could be

constancyGmbH, 2025-02-27:
Just another test. I changed the Adpater Type to `ember` in my z2m 2.0.0. Version, and it is working.
So it must be something comming with the 2.1.0 Update.

Ruko2010, 2025-02-27:
Sorry, I was logged in with the wrong account. Both comments above are from me too...

Ruko2010, 2025-03-17:
@Koenkk any idea how I can debug this?

Koenkk, 2025-03-18:
On what architecture are you running? amd64/armv?

Ruko2010, 2025-03-18:
I use an "normal mini PC", with Intel® Core™ i3-8145U CPU

pakar1, 2025-03-19:
I too see this issue on the latest version. After reverting back to 2.0.0 it all started working again.

Running with logging debug i get a Segmentation fault:
[Opening serial port wit](zh:ember:uart:ash: Opening serial port with ............
Segmentation fault

Running it on a OpenWRT router with arch aarch64

Koenkk, 2025-03-20:
@Ruko2010 after the container exits, can you execute a `docker inspect CONTAINER_ID` and post the output here?

constancyGmbH, 2025-03-20:
@Koenkk the container does not exits (unexpected). The Docker is running, just the z2m "server" is stuck in the bootloop

Koenkk, 2025-03-21:
with what command do you run the container?

Ruko2010, 2025-03-21:
I'm running it inside of unraid:

[image]

Koenkk, 2025-03-21:
Try matching the command from https://www.zigbee2mqtt.io/guide/installation/02_docker.html#running-the-container (remove things like PUID)

Ruko2010, 2025-03-24:
I tried it with the original image with this dockercommand on unraid, and now the newest version is running. So I think maybe the unraid image (https://github.com/imagegenius/docker-zigbee2mqtt) is missing some dependency or something:

```
docker run \
   --name zigbee2mqtt_2 \
   --net='bridge'  \
   --restart=unless-stopped \
   --device='/dev/ttyACM0' \
  -e HOST_OS="Unraid"  \
  -e HOST_HOSTNAME="Tower"  \
  -e HOST_CONTAINERNAME="zigbee2mqtt"  \
  -e 'PUID'='99'  \
  -e 'PGID'='100'  \
  -e 'UMASK'='022'  \
   -p 9442:9442 \
  -l net.unraid.docker.managed=dockerman  \
  -l net.unraid.docker.icon='https://raw.githubusercontent.com/imagegenius/templates/main/unraid/img/zigbee2mqtt.png'  \
   -v /mnt/user/appdata/zigbee2mqtt:/app/data \
   -v /run/udev:/run/udev:ro \
   -e TZ=Europe/Amsterdam \
   ghcr.io/koenkk/zigbee2mqtt
```

Ruko2010, 2025-03-24:
So, as far as I can see, this is no issue on this project, so I close the issue. Thanks for your help @Koenkk

closed 2025-03-24 by Ruko2010