Koenkk/zigbee2mqtt #26145: Tuya TS0601_dimmer_knob not reporting back

sbolten, 2025-02-03:
### Discussed in https://github.com/Koenkk/zigbee2mqtt/discussions/26143

<div type='discussions-op-text'>

<sup>Originally posted by **sbolten** February  3, 2025</sup>
Hi,

since a while I have these dimmers (https://www.zigbee2mqtt.io/devices/TS0601_dimmer_knob.html) up and running.

They are recognized in Z2M as follow:
Zigbee-Modell: TS0601
Zigbee-Hersteller: _TZE200_p0gzbqct

I can change status from Z2M GUI without any problems (light on/off, dimming). 

But: changing the status on the physical device does not report back to Z2M (after re-pairing only the very first small change is reported back but after that changes through the physical devices are not reported back to Z2M).

Now - I am not 100% sure - but I think this worked a couple of days ago. Maybe I am wrong and it never worked (I only recently introduced a dashboard showing the status of the dimmer...).

What can I do about this? Is it maybe related to upgrade to version 2.1?

Best regards,

Stephan

</div>

wimb0, 2025-02-03:
I am having the same issue with the same Moes / Tuya dimmers after updating to 2.1.0. 
2.0 and earlier were working fine.

Koenkk, 2025-02-03:
Could you provide the debug log on 2.0.0 where it works and 2.1.0 where it doesn't work?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

wimb0, 2025-02-03:
Logfile from 2.1.0: https://gist.githubusercontent.com/wimb0/6dc7a2fd4f83ac9acf9bcb70f03bf21f/raw/b76d0723a0782765f373462b4d522d6bc061f959/log.log

This was the state according to Z2M when the logfile was created:
[image]

This is wrong, because at that time, moes_dimmer_eettafel, moes_dimmer_keuken and moes_dimmer_woonkamer were on.

jpbede, 2025-02-03:
Based on the log I tend to say it is the same issue as in https://github.com/Koenkk/zigbee2mqtt/issues/26124.

On 20:49:08 your device proactively reported its state with `seq=1`. On 20:49:57 you turned it on, the device again proactively reported its state but again with `seq=1`, due to a change this packet is now discarded as there was already a packet with `seq=1`. I already proposed a fix for tha.

```
[2025-02-03 20:49:08] debug: 	zh:controller: Received payload: clusterID=61184, address=32262, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=40, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":71,"commandIdentifier":2},"payload":{"seq":1,"dpValues":[{"dp":1,"datatype":1,"data":{"type":"Buffer","data":[1]}}]},"command":{"ID":2,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataReport"}}

[2025-02-03 20:49:57] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/moes_dimmer_eettafel/set' with data '{"state":"OFF"}'
[2025-02-03 20:49:57] debug: 	z2m: Publishing 'set' 'state' to 'moes_dimmer_eettafel'
[2025-02-03 20:49:57] debug: 	zh:controller:endpoint: ZCL command 0xdc8e95fffe3ae965/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":1,"datatype":1,"data":[0]}]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-02-03 20:49:57] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xdc8e95fffe3ae965:32262/1 (0,0,1)
[2025-02-03 20:49:57] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":32262,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":30,"options":0,"radius":30,"len":10,"data":{"type":"Buffer","data":[17,7,0,1,0,1,1,0,1,0]}}

[2025-02-03 20:49:57] debug: 	zh:controller: Received payload: clusterID=61184, address=32262, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=40, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":72,"commandIdentifier":2},"payload":{"seq":1,"dpValues":[{"dp":1,"datatype":1,"data":{"type":"Buffer","data":[0]}}]},"command":{"ID":2,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataReport"}}
```

WarNeedle, 2025-02-03:
Exactly the same problem here with these dimmers.
https://www.zigbee2mqtt.io/devices/TS0601_dimmer_knob.html

Koenkk, 2025-02-04:
Should be fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/8723

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

tgroszkowski, 2025-02-05:
Same issue with:
[TS0601_dimmer_1_gang_1](https://www.zigbee2mqtt.io/devices/TS0601_dimmer_1_gang_1.html#tuya-ts0601-dimmer-1-gang-1)

sbolten, 2025-02-07:
Can confirm that the update / hot fix  from last night solved the problem for me.

Thx, S.

wimb0, 2025-02-07:
Updating to 2.1.1 also solved the problem here. Thanks!

closed 2025-02-07 by Koenkk