Koenkk/zigbee2mqtt #26146: [New device support]: Lixee Zlinky _TIC - LTARF not updating since 2.1.0-1

CokACola44, 2025-02-03:
### Link

https://www.zigbee2mqtt.io/devices/ZLinky_TIC.html

### Database entry

{"id":4,"type":"Router","ieeeAddr":"0x00158d0005d2663d","nwkAddr":41223,"manufId":4151,"manufName":"LiXee","powerSource":"Mains (single phase)","modelId":"ZLinky_TIC\u0000","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":83,"inClusterList":[0,1,3,1794,2820,2817,65382],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"ZLinky_TIC\u0000","manufacturerName":"LiXee","powerSource":1,"zclVersion":3,"appVersion":12,"stackVersion":2,"hwVersion":1,"dateCode":"20240422","swBuildId":"4000-0014"}},"liXeePrivate":{"attributes":{"linkyMode":1,"currentTarif":"H PLEINE/CREUSE\u0000","warnDPS":0,"statusRegister":{"type":"Buffer","data":[48,48,51,65,67,52,48,49]},"activeEnerfyOutD01":2128531,"currentDate":"H250203104934","currentIndexTarif":2,"currentPrice":" HEURE  CREUSE\u0000 ","drawnVAMaxN1":4790,"activeEnergyOutD01":18763027,"activeEnergyOutD02":8443049,"activeEnergyOutD03":2058335,"activeEnergyOutD04":4696590,"daysNumberCurrentCalendar":0,"daysNumberNextCalendar":0,"daysProfileCurrentCalendar":"\u0000","daysProfileNextCalendar":"\u0000","message1":"PAS DE          MESSAGE\u0000        ","message2":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","relais":0,"startMobilePoint1":0,"startMobilePoint2":0,"startMobilePoint3":0,"stopMobilePoint1":0,"stopMobilePoint2":0,"stopMobilePoint3":0,"motDEtat":"000000","scheduleHPHC":0}},"haElectricalMeasurement":{"attributes":{"apparentPower":350,"rmsCurrent":1,"rmsCurrentMax":90,"activePower":132,"activePowerPhB":128,"averageRmsVoltageMeasPeriod":234,"rmsVoltage":237,"activePowerMax":4760}},"seMetering":{"attributes":{"currentSummDelivered":36036609,"activeRegisterTierDelivered":"HP..","meterSerialNumber":{"type":"Buffer","data":[56,49,49,57,55,53,49,53,51,48,49,48]},"currentTier1SummDelivered":17759588,"currentTier2SummDelivered":18277021,"siteId":{"type":"Buffer","data":[49,52,50,57,51,52,56,55,54,52,55,57,52,52]},"currentTier3SummDelivered":[0,0],"currentTier4SummDelivered":[0,0],"currentTier5SummDelivered":[0,0],"currentTier6SummDelivered":[0,0],"currentTier7SummDelivered":[0,0],"currentTier8SummDelivered":[0,0],"currentTier9SummDelivered":[0,0],"currentTier10SummDelivered":[0,0]}},"haMeterIdentification":{"attributes":{"availablePower":9,"powerThreshold":9,"softwareRevision":2}}},"binds":[{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00212effff09b89f","endpointID":1},{"cluster":2817,"type":"endpoint","deviceIeeeAddress":"0x00212effff09b89f","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00212effff09b89f","endpointID":1},{"cluster":65382,"type":"endpoint","deviceIeeeAddress":"0x00212effff09b89f","endpointID":1}],"configuredReportings":[{"cluster":1794,"attrId":0,"minRepIntval":60,"maxRepIntval":900,"repChange":100,"manufacturerCode":null},{"cluster":1794,"attrId":256,"minRepIntval":60,"maxRepIntval":900,"repChange":100,"manufacturerCode":null},{"cluster":1794,"attrId":258,"minRepIntval":60,"maxRepIntval":900,"repChange":100,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":60,"maxRepIntval":900,"repChange":1,"manufacturerCode":null},{"cluster":2820,"attrId":2315,"minRepIntval":60,"maxRepIntval":900,"repChange":1,"manufacturerCode":null},{"cluster":2820,"attrId":1297,"minRepIntval":60,"maxRepIntval":900,"repChange":1,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":60,"maxRepIntval":900,"repChange":1,"manufacturerCode":null},{"cluster":2820,"attrId":1285,"minRepIntval":60,"maxRepIntval":900,"repChange":1,"manufacturerCode":null},{"cluster":65382,"attrId":515,"minRepIntval":60,"maxRepIntval":900,"repChange":100,"manufacturerCode":null},{"cluster":65382,"attrId":516,"minRepIntval":60,"maxRepIntval":900,"repChange":100,"manufacturerCode":null},{"cluster":2820,"attrId":1293,"minRepIntval":60,"maxRepIntval":900,"repChange":1,"manufacturerCode":null},{"cluster":2820,"attrId":1295,"minRepIntval":60,"maxRepIntval":900,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":12,"stackVersion":2,"hwVersion":1,"dateCode":"20240422","swBuildId":"4000-0014","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1858900957},"lastSeen":1738576750702}

### Zigbee2MQTT version

2.1.0-1

### Comments

Hello,

Since the latest Zigbee2MQTT update (2.1.0-1), my Lixee Zlinky _TIC device is no longer updating the LTARF attribute (current_price in the database entry), which should return either "HEURE CREUSE" or "HEURE PLEINE". Before this update, it was working correctly.
LTARF should update to "HEURE CREUSE" or "HEURE PLEINE" according to my electricity tariff status.
The device is still reporting general electricity meter data (voltage, power, etc.). There are no error messages in the logs related to this device.

Let me know if I need to provide any additional details.

### External definition

```shell
const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['ZLinky_TIC'],
    model: 'ZLinky_TIC',
    vendor: 'LiXee',
    description: 'Automatically generated definition',
    extend: [m.battery(), m.electricityMeter()],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

What works:
The device is still reporting general electricity meter data (voltage, power, etc.).

What doesn't work:
The LTARF attribute (expected to return "HEURE CREUSE" or "HEURE PLEINE") is now static and does not update anymore.

AURELB87, 2025-02-05:
Hello,

Same here. But for me, It stops to send information since the last OTA update that I made yesterday.

cperrier, 2025-02-05:
I reported the same issue in https://github.com/fairecasoimeme/Zlinky_TIC/issues/319 and I can confirm this is *not* related to an OTA update of the device, as I didn't do one.....but the problem seems clearly related to the Z2M update to 2.1.0-1

git-greatguy, 2025-02-05:
Facing same problem since i updated z2m to 2.1. reverting to 2.00.2 until this is fixed.

Koenkk, 2025-02-05:
Should be fixed now, let me know if it works!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-02-05 by Koenkk

AURELB87, 2025-02-05:
I just installed the last version of Home assistant core (version 2025.2.0) and it seems to work now ...