Koenkk/zigbee2mqtt #26147: ASH_NCP_FATAL_ERROR with Sonoff Zigbee 3.0 USB Dongle Plus V2

Tribalwarrior, 2025-02-03:
### What happened?

I have two Sonoff Zigbee 3.0 USB Dongle Plus V2, both with firmware 7.4.4. One of them works in EMBER properly the second does not.
Externally they are not different but they have internal differences.

**Problem one:**
> DEVLINKS: >-
>   /dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_e02c0ea18d38ef119efb3e7af3d9b1e5-if00-port0
>   /dev/serial/by-path/platform-xhci-hcd.1.auto-usb-0:2:1.0-port0
> DEVNAME: /dev/ttyUSB0
> DEVPATH: >-
>   /devices/platform/soc/d0078080.usb/c9000000.usb/xhci-hcd.1.auto/usb1/1-2/1-2:1.0/ttyUSB0/tty/ttyUSB0
> ID_BUS: usb
> ID_MODEL: Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2
> ID_MODEL_ENC: Sonoff\x20Zigbee\x203.0\x20USB\x20Dongle\x20Plus\x20V2
> ID_MODEL_FROM_DATABASE: CP210x UART Bridge
> ID_MODEL_ID: ea60
> ID_PATH: platform-xhci-hcd.1.auto-usb-0:2:1.0
> ID_PATH_TAG: platform-xhci-hcd_1_auto-usb-0_2_1_0
> ID_REVISION: '0100'
> ID_SERIAL: Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_e02c0ea18d38ef119efb3e7af3d9b1e5
> ID_SERIAL_SHORT: e02c0ea18d38ef119efb3e7af3d9b1e5
> ID_TYPE: generic
> ID_USB_DRIVER: cp210x
> ID_USB_INTERFACES: ':ff0000:'
> ID_USB_INTERFACE_NUM: '00'
> ID_USB_MODEL: Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2
> ID_USB_MODEL_ENC: Sonoff\x20Zigbee\x203.0\x20USB\x20Dongle\x20Plus\x20V2
> ID_USB_MODEL_ID: ea60
> ID_USB_REVISION: '0100'
> ID_USB_SERIAL: Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_e02c0ea18d38ef119efb3e7af3d9b1e5
> ID_USB_SERIAL_SHORT: e02c0ea18d38ef119efb3e7af3d9b1e5
> ID_USB_TYPE: generic
> ID_USB_VENDOR: Itead
> ID_USB_VENDOR_ENC: Itead
> ID_USB_VENDOR_ID: 10c4
> ID_VENDOR: Itead
> ID_VENDOR_ENC: Itead
> ID_VENDOR_FROM_DATABASE: Silicon Labs
> ID_VENDOR_ID: 10c4
> MAJOR: '188'
> MINOR: '0'
> SUBSYSTEM: tty
> TAGS: ':systemd:'
> USEC_INITIALIZED: '9567701'

**Good one:** 

> DEVLINKS: >-
>   /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20231215154001-if00
>   /dev/serial/by-path/platform-5311400.usb-usb-0:1:1.0
> DEVNAME: /dev/ttyACM0
> DEVPATH: /devices/platform/soc/5311400.usb/usb7/7-1/7-1:1.0/tty/ttyACM0
> ID_BUS: usb
> ID_MODEL: SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2
> ID_MODEL_ENC: SONOFF\x20Zigbee\x203.0\x20USB\x20Dongle\x20Plus\x20V2
> ID_MODEL_ID: 55d4
> ID_PATH: platform-5311400.usb-usb-0:1:1.0
> ID_PATH_TAG: platform-5311400_usb-usb-0_1_1_0
> ID_REVISION: '0442'
> ID_SERIAL: ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20231215154001
> ID_SERIAL_SHORT: '20231215154001'
> ID_TYPE: generic
> ID_USB_CLASS_FROM_DATABASE: Communications
> ID_USB_DRIVER: cdc_acm
> ID_USB_INTERFACES: ':020201:0a0000:'
> ID_USB_INTERFACE_NUM: '00'
> ID_USB_MODEL: SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2
> ID_USB_MODEL_ENC: SONOFF\x20Zigbee\x203.0\x20USB\x20Dongle\x20Plus\x20V2
> ID_USB_MODEL_ID: 55d4
> ID_USB_REVISION: '0442'
> ID_USB_SERIAL: ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20231215154001
> ID_USB_SERIAL_SHORT: '20231215154001'
> ID_USB_TYPE: generic
> ID_USB_VENDOR: ITEAD
> ID_USB_VENDOR_ENC: ITEAD
> ID_USB_VENDOR_ID: 1a86
> ID_VENDOR: ITEAD
> ID_VENDOR_ENC: ITEAD
> ID_VENDOR_FROM_DATABASE: QinHeng Electronics
> ID_VENDOR_ID: 1a86
> MAJOR: '166'
> MINOR: '0'
> SUBSYSTEM: tty
> TAGS: ':systemd:'
> USEC_INITIALIZED: '11274165'

The problem is that when trying to OTA update firmware of Zigbee devices the dongle falls off.

### What did you expect to happen?

normal work

### How to reproduce it (minimal and precise)

use dongle with ID_USB_DRIVER: cp210x

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

7.4.4 [GA]

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus V2

### Setup

HomeAssistant Add-on

### Debug log

[log.log](https://github.com/user-attachments/files/18641390/log.log)

Tribalwarrior, 2025-02-03:
I also tried the firmware [zbdonglee_zigbee_ncp_8.0.2.0_sw_flow_115200.gbl](https://github.com/darkxst/silabs-firmware-builder/releases/download/20250203/zbdonglee_zigbee_ncp_8.0.2.0_sw_flow_115200.gbl) the result is the same

Tribalwarrior, 2025-02-05:
[Same issue](https://github.com/Koenkk/zigbee2mqtt/issues/26190) with similar problem

palmitoto, 2025-02-16:
I'm encountering the same issue, but with the SkyConnect. Any progress on this major problem?

Foguet, 2025-02-16:
I'm having the same problem, I was on firmware 7.4.1 and I've updated to 7.4.5 and the same thing keeps happening to me

Tribalwarrior, 2025-02-16:
I rolled back to version 6 and changed the driver to ezsp. But I think this is a temporary solution. I would still like to use modern technologies. By the way, what are the risks of using the ezsp driver?

AlexKorsakov, 2025-03-07:
> I rolled back to version 6 and changed the driver to ezsp. But I think this is a temporary solution. I would still like to use modern technologies. By the way, what are the risks of using the ezsp driver?

And it works properly after rollback?

I have the same issue in topic and I'm stuck with it.

Tribalwarrior, 2025-03-07:
> And it works properly after rollback?
> 
> I have the same issue in topic and I'm stuck with it.
Yep

AlexKorsakov, 2025-03-07:
> > And it works properly after rollback?
> > I have the same issue in topic and I'm stuck with it.
> 
> Yep

Can you send a link to the firmware?

Tribalwarrior, 2025-06-19:
> Can you send a link to the firmware? Which version exactly?

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP

Tribalwarrior, 2025-08-19:
Up!

closed 2025-09-22 by Tribalwarrior