Koenkk/zigbee2mqtt #26148: No update of the temperature for Tuya _TZE200_hue3yfsn

PJuergen, 2025-02-03:
### What happened?

Since the update from version 2.0.0 to 2.1.0, the “Local temperature” field is no longer updated.

Zigbee model: TS0601
Zigbee manufacturer: _TZE200_hue3yfsn
Description: Thermostat radiator valve

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

20240710

### Adapter

ZigBee

### Setup

Docker

oandry72, 2025-02-03:
The same with my devices.

Zigbee Model: TS0601
Zigbee Manufacturer: _TZE200_hue3yfsn
Model: TV02-Zigbee

Synochy, 2025-02-03:
Same issue with my device
Zigbee Model: TS0601
Zigbee Manufacturer:: TZE204_mpbki2zm
Model: [TYBAC-006](https://www.zigbee2mqtt.io/devices/TYBAC-006.html#tuya-tybac-006)

Sander544, 2025-02-04:
The same with my devices.
Zigbee Model - TS0601
Zigbee Manufacturer - _TZE200_hue3yfsn
Model TV02-Zigbee

Robert904, 2025-02-04:
Same problem. Working again after downgrade to 2.0.0
Zigbee Model - TS0601
Zigbee Manufacturer - _TZE200_hue3yfsn
Model TV02-Zigbee

Zayeen, 2025-02-04:
Same problem

Zigbee Model - TS0601
Zigbee Manufacturer - _TZE200_hue3yfsn
Model TV02-Zigbee

closed 2025-04-13 by github-actions[bot]