Koenkk/zigbee2mqtt #26150: BTicino LN4570CWI wireless switch not associable

BertoGT, 2025-02-03:
### What happened?

Hi,

I see that the BTicino LN4570CWI wireless switch (064873) seems to be supported by Zigbee2MQTT -> https://www.zigbee2mqtt.io/devices/064873.html

I'm not able to add it to the zigbee network. The others biticino devices (example https://www.zigbee2mqtt.io/devices/4411C_L4411C_N4411C_NT4411C.html) can be added correctly.

I'm using version 2.1.0 Z2MQTT
Zigbee stick: Sonoff ZigBee 3.0 USB Dongle Plus ZBDongle-E (v2) with firmware version 8.0.2.0 (ember)

### What did you expect to happen?

The switch is added to zigbee network

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

8.0.2 [GA]

### Adapter

EmberZNet

### Setup

Docker container on RPi 4

dinukro, 2025-02-14:
Hello, do you solve it? I have this problem and with adapter ZBDongle-P.

BertoGT, 2025-02-16:
> Hello, do you solve it? I have this problem and with adapter ZBDongle-P.

No :-(

dinukro, 2025-02-21:
Hi, I want to add share my experience with LN4570CWI.
So if you try to pairing the model that is part of the kit (NC5400C+LN4570CWI) you never cad pair with other gateway of coordinator.
If you buy ONLY LN4570CWI (single) you can pair with any gateway or coordinator and is reconised like Legrand 064873

Regards
C.

BertoGT, 2025-02-21:
> Hi, I want to add share my experience with LN4570CWI. So if you try to pairing the model that is part of the kit (NC5400C+LN4570CWI) you never can pair with other gateway or coordinator. If you buy ONLY LN4570CWI (single) you can pair with any gateway or coordinator and is recognized like Legrand 064873
> 
> Regards C.

Thank you for sharing this information! It seems that the white version of the single command is not available :-(

closed 2025-05-01 by github-actions[bot]