Koenkk/zigbee2mqtt #26152: all devices are lost after ZBOSS adapter power reset

Morteza-24, 2025-02-03:
### What happened?

I'm using an ESP32H2 as my coordinator with the firmware provided [here](https://github.com/andryblack/esp-coordinator). Everything works fine except for one big problem: when I restart the coordinator, all the devices are gone/removed. Please note that if I only restart Z2M, there is no problem, the problem only happens when the ESP32H2 is restarted (e.g. after a power loss or resetting the ESP32H2 manually). I have tested the dev branch too.

### What did you expect to happen?

I expected that the devices remain connected and be controllable, without requiring re-pairing after coordinator's power reset.

### How to reproduce it (minimal and precise)

First build the code provided [here](https://github.com/andryblack/esp-coordinator) and flash it into an ESP32H2. Then connect it to a system  running Z2M and include the following in `configuration.yaml`:

```
serial:
    adapter: zboss
```

Then run Z2M and add some devices. Then reboot your system (power off, then power on), restarting both Z2M and coordinator. Make sure the power to your coordinator is disconnected for at least a second. Then the devices you added will be gone.

### Zigbee2MQTT version

2.1.0-dev commit: 75c3c5e5

### Adapter firmware version

 0.0.1.0

### Adapter

ZBOSS (ESP32H2)

### Setup

Plain Z2M on Orange Pi Zero

### Debug log

Before power reset, starting Z2M and adding a device:

[log.log](https://github.com/user-attachments/files/18643246/log.log)

After power reset, the added device is gone:

[log.log](https://github.com/user-attachments/files/18643248/log.log)

kirovilya, 2025-02-03:
I'm looking at your logs. The following happens:
- when the zboss adapter starts, the current state of the zigbee chip and its connection to the network is checked
- the network parameters and the current adapter parameters are compared
- if the network parameters do not match, a new network is formed, and the list of devices is cleared

This is exactly your case - when adapter start after turning off the power, for some reason, the network settings are reset: panid = 65535, empty extendedPanID [0,0,0,0,0,0,0,0], channel = 255.
so the network is restarted.

```
[2025-02-03 17:26:09] debug: 	zh:zboss:driv: Current network parameters: {"joined":false,"nodeType":3,"ieeeAddr":"0xa20d60feffbd4d74","network":{"panID":65535,"extendedPanID":{"type":"Buffer","data":[0,0,0,0,0,0,0,0]},"channel":255}}
[2025-02-03 17:26:09] info: 	zh:zboss:driv: Form network
```

In the case of the NRF chip and the zboss adapter this does not happen.

You can try to contact the author of the zboss firmware for esp32-h2, maybe there is an error there.

Morteza-24, 2025-02-03:
Thank you for taking the time and reading the log files. I didn't notice that `extendedPanID` and `channel` change after restart. Maybe the esp coordinator does not save network settings into NVRAM? I'll try to look more into it tomorrow. 

> You can try to contact the author of the zboss firmware for esp32-h2, maybe there is an error there.

Unfortunately the author of the ZBOSS firmware for ESP32H2/C6 doesn't respond to issues; there is already an issue about this problem on that repository.

Morteza-24, 2025-02-04:
Update: ESP32-C6 has this problem too.

Morteza-24, 2025-02-04:
The problem is now fixed in the following pull request:

https://github.com/andryblack/esp-coordinator/pull/6

closed 2025-02-04 by Morteza-24