Koenkk/zigbee2mqtt #26157: [Tuya ZWT198/ZWT100-BH] Thermostat's local temperature calibration out of range every time passed by schedule period

majkers, 2025-02-03:
### What happened?

I have local temperature calibration value set to -0.1 and each time my thermostat passes through one of schedule periods (set to 06:00/20.0°C 07:00/19.2°C 11:30/19.2°C 15:30/19.6°C 17:00/20.0°C 22:30/19.2°C)  this value changes by itself to 6553.5 and I get an error in logs:

`Invalid value for number.termostat_local_temperature_calibration: 6553.5 (range -9.9 - 9.9)`

### What did you expect to happen?

Value of calibration should stay as set

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

20230507

### Adapter

zStack3x0 Sonoff dongle P

### Setup

Hass OS on PI 4

### Debug log

Invalid value for number.termostat_local_temperature_calibration: 6553.5 (range -9.9 - 9.9)

majkers, 2025-02-03:
Looks like I am no the only one having this issue https://github.com/Koenkk/zigbee2mqtt/issues/24943 and https://github.com/Koenkk/zigbee2mqtt/issues/24758. I open new one cause old ones are stale or closed...

majkers, 2025-02-05:
Just like it was mentioned in https://github.com/Koenkk/zigbee2mqtt/issues/24758#issuecomment-2477275989
wrong code is in https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts and https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/tuya.ts

alexsotoaguilera, 2025-02-05:
> [@alexsotoaguilera](https://github.com/alexsotoaguilera) can You check it out?

is the change?

`   localTempCalibration3: {
        from: (v: number) => {
            if (v > 0x7fffffff) v -= 0x100000000;
            return v / 10;
        },
        to: (v: number) => {
            if (v > 0) return v * 10;
            if (v < 0) return v * 10 + 0x100000000; 
            return v;
        },
    },`

for this?

`
   localTempCalibration3: {
        return v * 10;
    },
`

....

in order not to break anything, you could create a new converter that returns ‘return v * 10;’ with a new index ‘localTempCalibration4’.

majkers, 2025-02-06:
Like I said. IMHP there is one:  tuya.valueConverter.divideBy10

majkers, 2025-02-06:
UPDATE. I think I know how those converters should work....more or less like that I guess:

`output = from(to(input))
`
and for input -0.1 output is -0.1 in my app written is typeScript and angular. TO be more precise. to generates value 4294967295 and from(4294967295) is -0.1
So I don't know what the problem is :(

majkers, 2025-02-06:
Reading throug past issues maybe there is something wrong that these negative values are being send as strings not as numbers?

OpaApo, 2025-02-19:
I have exactly the same problem. The value is first set correctly but after a random time it transforms to this big invalid number.

agak79, 2025-02-21:
Same problem here

agak79, 2025-03-18:
Is there anyway to solve this issue?

Palec, 2025-08-05:
Model
[ZWT198/ZWT100-BH]
(https://www.zigbee2mqtt.io/devices/ZWT198_ZWT100-BH.html#tuya-zwt198%252Fzwt100-bh)

Logger: homeassistant.components.mqtt.number
Source: components/mqtt/number.py:184
Integration: MQTT (documentation, issues)
First seen: 1 Aug 2025 at 20:32:26 (34155 occurrences)
Last logged in: 00:27:55

Invalid value for number.termostat_local_temperature_calibration: 6553.2 (range -9.9 - 9.9)

alexsotoaguilera, 2025-10-29:
> @chay150 <https://github.com/chay150> can You open PR with this change?
>
> Sorry, I have no knowledge on git,
> I don't know how to do it.
> I modified the code directly on my running zigbee2mqtt in two tuya.js
> files.

You add localTempCalibration4, but in the original file there are 1,2 and
3, so it makes no sense for the device to use 4.

Are you sure you haven't overwritten an existing one?

{code}
    selfTestResult: valueConverterBasic.lookup({checking: 0, success: 1,
failure: 2, others: 3}),
    lockUnlock: valueConverterBasic.lookup({LOCK: true, UNLOCK: false}),
    localTempCalibration1: {
        from: (v: number) => {
            // biome-ignore lint/style/noParameterAssign: ignored using
`--suppress`
            if (v > 55) v -= 0x100000000;
            return v / 10;
        },
        to: (v: number) => {
            if (v > 0) return v * 10;
            if (v < 0) return v * 10 + 0x100000000;
            return v;
        },
    },
    localTempCalibration2: {
        from: (v: number) => v,
        to: (v: number) => {
            if (v < 0) return v + 0x100000000;
            return v;
        },
    },
    localTempCalibration3: {
        from: (v: number) => {
            // biome-ignore lint/style/noParameterAssign: ignored using
`--suppress`
            if (v > 0x7fffffff) v -= 0x100000000;
            return v / 10;
        },
        to: (v: number) => {
            if (v > 0) return v * 10;
            if (v < 0) return v * 10 + 0x100000000;
            return v;
        },
    },
{code}

closed 2026-01-05 by github-actions[bot]