Koenkk/zigbee2mqtt #26158: [New device support]: 50139 SH ZB Paulmann Home Universal-Schaltaktor Weiß

ochstobi, 2025-02-03:
### Link

https://www.conrad.de/de/p/50139-sh-zb-paulmann-home-universal-schaltaktor-weiss-3345231.html

### Database entry

{"id":30,"type":"Router","ieeeAddr":"0xa4c1381b33ed1195","nwkAddr":9942,"manufId":4417,"manufName":"_TZ3210_hjxqqofs\u0000","powerSource":"Mains (single phase)","modelId":"TS000F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,57345,57344,61184],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"TS000F","manufacturerName":"_TZ3210_hjxqqofs\u0000","powerSource":1,"zclVersion":3,"appVersion":81,"stackVersion":1,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":81,"stackVersion":1,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1736458119350}

### Zigbee2MQTT version

2.1.0

### Comments

I tried to support this device here: https://github.com/Koenkk/zigbee-herdsman-converters/pull/8582
Should be added in 2.1.0 but it shows still unsupported in z2m frontend.

### External definition

```shell
const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS000F'],
    model: 'TS000F',
    vendor: '_TZ3210_hjxqqofs
    description: 'Automatically generated definition',
    extend: [m.onOff({"powerOnBehavior":false})],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

I can only see the linkquality

Koenkk, 2025-02-04:
This device is supported in z2m 2.1.0

ochstobi, 2025-02-04:
I know it should, I opened the PR. When I try to pair it's not working.

```
info 2025-02-04 21:26:42zh:controller: Interview for '0xa4c1381b33ed1195' started
info 2025-02-04 21:26:42z2m: Device '0xa4c1381b33ed1195' joined
info 2025-02-04 21:26:42z2m: Starting interview of '0xa4c1381b33ed1195'
info 2025-02-04 21:26:42z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b33ed1195","ieee_address":"0xa4c1381b33ed1195"},"type":"device_joined"}'
info 2025-02-04 21:26:42z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b33ed1195","ieee_address":"0xa4c1381b33ed1195","status":"started"},"type":"device_interview"}'
info 2025-02-04 21:26:42z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b33ed1195","ieee_address":"0xa4c1381b33ed1195"},"type":"device_announce"}'
error 2025-02-04 21:27:03zh:controller: Interview failed for '0xa4c1381b33ed1195 with error 'Error: Interview failed because can not get active endpoints ('0xa4c1381b33ed1195')'
error 2025-02-04 21:27:03z2m: Failed to interview '0xa4c1381b33ed1195', device has not successfully been paired
info 2025-02-04 21:27:03z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b33ed1195","ieee_address":"0xa4c1381b33ed1195","status":"failed"},"type":"device_interview"}'
```