Koenkk/zigbee2mqtt #26161: Not able to pair any device after update

gvi70000, 2025-02-03:
Hello,
after updating zigbee2mqtt I am not able to pair my devices no matter what i try.
OS: Ubuntu server 24.04
mosquitto version 2.0.20
node --version v20.18.2
pnpm --version 9.15.5
Coordinator: Zigstar V4 running CC1352P2_CC2652P_launchpad_coordinator_20240710.hex

I have programmed the coordinator and erased NVRAM several times.

My configuration file
`version: 4
permit_join: true
homeassistant:
  enabled: false
frontend:
  enabled: true
  port: 8088
mqtt:
  base_topic: zigbee2mqtt
  server: '!secret.yaml server'
  user: '!secret.yaml user'
  password: '!secret.yaml password'
serial:
  port: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
  disable_led: false
  rtscts: false
  adapter: zstack
advanced:
  transmit_power: 20
  channel: 11
  log_level: debug
  network_key:
    - 128
    - 93
    - 208
    - 198
    - 121
    - 106
    - 96
    - 115
    - 247
    - 177
    - 234
    - 153
    - 142
    - 229
    - 103
    - 222
  pan_id: 58498
  ext_pan_id:
    - 153
    - 75
    - 239
    - 24
    - 203
    - 234
    - 51
    - 195
My log file
[log.log](https://github.com/user-attachments/files/18646246/log.log)

`
I have turned off all routers and powered devices, turn off WiFi Only the battery powered sensors are "online" and none of them are joining the network.

Any tips/advices on what to do?

Nerivec, 2025-02-03:
```yaml
permit_join: true
```

This is no longer a valid setting in 2.0.0 as mentioned in [breaking changes](https://github.com/Koenkk/zigbee2mqtt/discussions/24198).
I do no see any occurrence of join being permitted in your log file. You must use the Permit Join button in frontend or via MQTT to allow devices to join the network.

gvi70000, 2025-02-04:
Thank you for reply Nerivec.
That solved my issue.

closed 2025-02-04 by gvi70000