Koenkk/zigbee2mqtt #26162: [New device support]: TS0601 Manufacturer _TZE284_5m4nchbm

zvAndrey, 2025-02-03:
### Link

https://aliexpress.ru/item/1005007842975372.html?spm=a2g2w.orderdetail.0.0.31ba4aa6WU6sH6&sku_id=12000042464470583

### Database entry

{"id":32,"type":"Router","ieeeAddr":"0xf84477fffef84813","nwkAddr":23355,"manufId":4098,"manufName":"_TZE284_5m4nchbm","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"W�3/iX�3/iY�3/iZ�3/i","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE284_5m4nchbm","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1738605363089}

### Zigbee2MQTT version

2.1.0

### Comments

Good evening,
This device is successfully interviewed and not supported

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_5m4nchbm',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

not supported

zvAndrey, 2025-02-08:
I was able to run the device with the following converter.
I am not sure that all fault flags are correct, as I was only able to check the current leakage.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy = require('zigbee-herdsman-converters/lib/legacy');

const fzLocal = {
    value: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const lookup = {0: 'OFF', 1: 'ON'};
            payload = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp;
                switch (dp) {
                    case 48:
                    {
                        payload['leakage_breaker'] = lookup[legacy.getDataValue(dpValue)[1]];
                        payload['leakage_threshold'] = legacy.getDataValue(dpValue)[2]*256+legacy.getDataValue(dpValue)[3];
                        payload['high_temperature_breaker'] = lookup[legacy.getDataValue(dpValue)[5]];
                        payload['high_temperature_threshold'] = legacy.getDataValue(dpValue)[6]*256+legacy.getDataValue(dpValue)[7];
                        payload['high_power_breaker'] = lookup[legacy.getDataValue(dpValue)[9]];
                        payload['high_power_threshold'] = legacy.getDataValue(dpValue)[10]*256+legacy.getDataValue(dpValue)[11];
                        return payload;
                    };
                    case 49:
                    {
                        payload['over_current_breaker'] = lookup[legacy.getDataValue(dpValue)[1]];
                        payload['over_current_threshold'] = legacy.getDataValue(dpValue)[2]*256+legacy.getDataValue(dpValue)[3];
                        payload['over_voltage_breaker'] = lookup[legacy.getDataValue(dpValue)[5]];
                        payload['over_voltage_threshold'] = legacy.getDataValue(dpValue)[6]*256+legacy.getDataValue(dpValue)[7];
                        payload['under_voltage_breaker'] = lookup[legacy.getDataValue(dpValue)[9]];
                        payload['under_voltage_threshold'] = legacy.getDataValue(dpValue)[10]*256+legacy.getDataValue(dpValue)[11];
                        return payload;
                    };
                };
            };
        },
    },
};

const tzLocal = {
    key: [
          'leakage_breaker',
          'leakage_threshold',
          'high_temperature_breaker',
          'high_temperature_threshold',
          'high_power_breaker',
          'high_power_threshold',
          'over_current_breaker',
          'over_current_threshold',
          'over_voltage_breaker',
          'over_voltage_threshold',
          'under_voltage_breaker',
          'under_voltage_threshold'
         ],

    convertSet: async (entity, key, value, meta) => {
        const lookup = {'OFF': 0, 'ON': 1}

        const value_send1 = meta.state['alarm1'];
[... 210 lines cut ...]
                    overcurrent_alarm: 1024,
                    outage_alarm: 2048,
                    magnetism_alarm: 4096,
                    credit_alarm: 8192
                }),
            ],

            [38, 'power_outage_memory', tuya.valueConverterBasic.lookup({off: tuya.enum(0), on: tuya.enum(1), restore: tuya.enum(2)})],
            [41, 'child_lock', tuya.valueConverter.lockUnlock],
            [45, 'leakage_test', tuya.valueConverter.onOff],
            [47, 'temperature', tuya.valueConverter.raw],

            [48, 'alarm1', tuya.valueConverter.raw],
            [49, 'alarm2', tuya.valueConverter.raw],

            // Ignored for now; don't know what the values mean
            [40, null, null],
            [46, null, null],
            [53, null, null],
        ],
    }
};

module.exports = definition;
```

zvAndrey, 2025-02-08:
Corrected. Leakage data point change to 53 

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy = require('zigbee-herdsman-converters/lib/legacy');

const fzLocal = {
    value: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const lookup = {0: 'OFF', 1: 'ON'};
            payload = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp;
                switch (dp) {
                    case 48:
                    {
                        payload['leakage_breaker'] = lookup[legacy.getDataValue(dpValue)[1]];
                        payload['leakage_threshold'] = legacy.getDataValue(dpValue)[2]*256+legacy.getDataValue(dpValue)[3];
                        payload['high_temperature_breaker'] = lookup[legacy.getDataValue(dpValue)[5]];
                        payload['high_temperature_threshold'] = legacy.getDataValue(dpValue)[6]*256+legacy.getDataValue(dpValue)[7];
                        payload['high_power_breaker'] = lookup[legacy.getDataValue(dpValue)[9]];
                        payload['high_power_threshold'] = legacy.getDataValue(dpValue)[10]*256+legacy.getDataValue(dpValue)[11];
                        return payload;
                    };
                    case 49:
                    {
                        payload['over_current_breaker'] = lookup[legacy.getDataValue(dpValue)[1]];
                        payload['over_current_threshold'] = legacy.getDataValue(dpValue)[2]*256+legacy.getDataValue(dpValue)[3];
                        payload['over_voltage_breaker'] = lookup[legacy.getDataValue(dpValue)[5]];
                        payload['over_voltage_threshold'] = legacy.getDataValue(dpValue)[6]*256+legacy.getDataValue(dpValue)[7];
                        payload['under_voltage_breaker'] = lookup[legacy.getDataValue(dpValue)[9]];
                        payload['under_voltage_threshold'] = legacy.getDataValue(dpValue)[10]*256+legacy.getDataValue(dpValue)[11];
                        return payload;
                    };
                };
            };
        },
    },
};

const tzLocal = {
    key: [
          'leakage_breaker',
          'leakage_threshold',
          'high_temperature_breaker',
          'high_temperature_threshold',
          'high_power_breaker',
          'high_power_threshold',
          'over_current_breaker',
          'over_current_threshold',
          'over_voltage_breaker',
          'over_voltage_threshold',
          'under_voltage_breaker',
          'under_voltage_threshold'
         ],

    convertSet: async (entity, key, value, meta) => {
        const lookup = {'OFF': 0, 'ON': 1}

        const value_send1 = meta.state['alarm1'];
        const value_send2 = meta.state['alarm2'];
[... 211 lines cut ...]
                    credit_alarm: 8192
                }),
            ],

            [38, 'power_outage_memory', tuya.valueConverterBasic.lookup({off: tuya.enum(0), on: tuya.enum(1), restore: tuya.enum(2)})],
            [41, 'child_lock', tuya.valueConverter.lockUnlock],
            [45, 'leakage_test', tuya.valueConverter.onOff],
            [47, 'temperature', tuya.valueConverter.raw],

            [48, 'alarm1', tuya.valueConverter.raw],
            [49, 'alarm2', tuya.valueConverter.raw],

            [53, 'leakage', tuya.valueConverter.raw],

            // Ignored for now; don't know what the values mean
            [9, null, null],
            [40, null, null],
            [46, null, null]
        ],
    }
};

module.exports = definition;

```

robot468, 2025-02-28:
Thank you, its worked for me!
But total consumed energy seems not working.

zvAndrey, 2025-02-28:
This device returns only the delta that needs to be added to the Home Assistant statistics.
And the consumed power can be viewed in HA.

My data almost matched what was displayed on the screen.
There may be discrepancies due to HA restarts

robot468, 2025-03-01:
Strange, it seems like instant power is basically the same thing.
By the way, I don't have voltage and current values in Home Assistant. Do you know if this is a problem with my HA or this converter?

Windserfer, 2025-03-20:
Hi,
When I added external converter, - I'm getting next error:
error: 	z2m: Exception while calling fromZigbee converter: legacy.getDataValue is not a function}

could you please assist?

zvAndrey, 2025-03-20:
I've updated to the latest Z2M, and faced the same issue.

This PR https://github.com/Koenkk/zigbee-herdsman-converters/pull/8833 broke this converter

I replaced the function call with a direct access to the raw array

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const fzLocal = {
    value: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const lookup = {0: 'OFF', 1: 'ON'};
            payload = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp;
                switch (dp) {
                    case 48:
                    {
                        payload['leakage_breaker'] = lookup[dpValue.data[1]];
                        payload['leakage_threshold'] = dpValue.data[2]*256+dpValue.data[3];
                        payload['high_temperature_breaker'] = lookup[dpValue.data[5]];
                        payload['high_temperature_threshold'] = dpValue.data[6]*256+dpValue.data[7];
                        payload['high_power_breaker'] = lookup[dpValue.data[9]];
                        payload['high_power_threshold'] = dpValue.data[10]*256+dpValue.data[11];
                        return payload;
                    };
                    case 49:
                    {
                        payload['over_current_breaker'] = lookup[dpValue.data[1]];
                        payload['over_current_threshold'] = dpValue.data[2]*256+dpValue.data[3];
                        payload['over_voltage_breaker'] = lookup[dpValue.data[5]];
                        payload['over_voltage_threshold'] = dpValue.data[6]*256+dpValue.data[7];
                        payload['under_voltage_breaker'] = lookup[dpValue.data[9]];
                        payload['under_voltage_threshold'] = dpValue.data[10]*256+dpValue.data[11];
                        return payload;
                    };
                };
            };
        },
    },
};

const tzLocal = {
    key: [
          'leakage_breaker',
          'leakage_threshold',
          'high_temperature_breaker',
          'high_temperature_threshold',
          'high_power_breaker',
          'high_power_threshold',
          'over_current_breaker',
          'over_current_threshold',
          'over_voltage_breaker',
          'over_voltage_threshold',
          'under_voltage_breaker',
          'under_voltage_threshold'
         ],

    convertSet: async (entity, key, value, meta) => {
        const lookup = {'OFF': 0, 'ON': 1}
[... 213 lines cut ...]
                    magnetism_alarm: 4096,
                    credit_alarm: 8192
                }),
            ],

            [38, 'power_outage_memory', tuya.valueConverterBasic.lookup({off: tuya.enum(0), on: tuya.enum(1), restore: tuya.enum(2)})],
            [41, 'child_lock', tuya.valueConverter.lockUnlock],
            [45, 'leakage_test', tuya.valueConverter.onOff],
            [47, 'temperature', tuya.valueConverter.raw],

            [48, 'alarm1', tuya.valueConverter.raw],
            [49, 'alarm2', tuya.valueConverter.raw],

            [53, 'leakage', tuya.valueConverter.raw],

            // Ignored for now; don't know what the values mean
            [9, null, null],
            [40, null, null],
            [46, null, null]
        ],
    }
};

module.exports = definition;
```

Windserfer, 2025-03-20:
thanks a lot - that has been fix the problem

Windserfer, 2025-03-24:
> This device returns only the delta that needs to be added to the Home Assistant statistics. And the consumed power can be viewed in HA.
> 
> My data almost matched what was displayed on the screen. There may be discrepancies due to HA restarts

Could you please help how to calculate Total consumed energy in HA?

zvAndrey, 2025-03-27:
I'm not sure if this is a bug in the converter. 
A similar problem is described https://github.com/Koenkk/zigbee2mqtt/issues/21037
I'll try to figure out what can be done here.
So far, the only solution is workaround in issues 21037: remove void line in configuration file...

dengi76, 2025-04-04:
Hello. Today I bought this relay and used this converter. Thank you very much for the work done! The relay works perfectly with it! And the total consumption meter can be easily made in the Home Assistant itself. Once again - thank you!!!

MegaChel174, 2025-04-11:
Большое спасибо за конвертор, он очень сильно помог

mujdatdinc, 2025-04-12:
Reported energy delta is not correct as its precious is only 2.  It is not matching integral power.

robot468, 2025-04-17:
Maybe someone knows how to read the information about the reason for the relay reclosing? I have removed all thresholds and still once a day it switches off for some unknown reason.

zvAndrey, 2025-04-17:
> Maybe someone knows how to read the information about the reason for the relay reclosing? I have removed all thresholds and still once a day it switches off for some unknown reason.

I have had this happen a few times too, on two different devices. It turns off the load and turns it on after a few seconds. At the same time, the zigbee network does not send any events. Even the state does not change. Most likely, it is something hardware, not the best device, unfortunately.

robot468, 2025-04-17:
> I have had this happen a few times too, on two different devices. It turns off the load and turns it on after a few seconds. At the same time, the zigbee network does not send any events. Even the state does not change. Most likely, it is something hardware, not the best device, unfortunately.

It seems that some external causes are still influencing, but I can't figure out what they are. Every time before triggering, I see some changes in the voltage graph.

[image]

ISanteiI, 2025-04-17:
Thank you, it works.

francisc1968, 2025-04-18:
Thank you very much. It is working well.

francisc1968, 2025-04-18:
> Could you please help how to calculate Total consumed energy in HA?

You can create a Helper- Utility meter (select Delta values).

cizetta, 2025-05-05:
Thank you for the convertor @zvAndrey . It is working great.
However I have an issue with this breaker. Once it was mounted and connected on the panel it stops and starts every 15-20 mins without any reason - all protections are stopped. This behavior was reported in the comments above already. Can somebody confirm that his breaker is working well without interruptions and just some buggy ones do this?

Thank you!

dengi76, 2025-05-05:
> Спасибо за конвертор[@zvAndrey](https://github.com/zvAndrey). Он отлично работает. Однако у меня возникла проблема с этим выключателем. После того, как он был установлен и подключен к панели, он останавливается и запускается каждые 15-20 минут без какой-либо причины - все защиты останавливаются. Такое поведение уже было описано в комментариях выше. Может ли кто-нибудь подтвердить, что его выключатель работает хорошо, без перебоев, и это делают только некоторые глючные?
> 
> Спасибо!

I can confirm that I don't have any phantom shutdowns!

francisc1968, 2025-05-05:
I confirm that my relay is working well.

robot468, 2025-05-05:
> Thank you for the convertor [@zvAndrey](https://github.com/zvAndrey) . It is working great. However I have an issue with this breaker. Once it was mounted and connected on the panel it stops and [...]

Я понял причину своих "отключений". Это очень кратковременные (1-2с) падения напряжения на фазе. Но переключатель "Under voltage breaker", а возможно и остальные, не работают если значение минимального напряжения не выкручено в минимум. У меня сначала был исходный конвертер из этой ветки, где минимальное значение напряжение было 100в и реле рвало соединение нисмотря ни на что (порог срабатывания через сам экранчик реле выставлен в 5с). Потом я поставил последнюю версию из этой ветки, там минимальное напряжение уже 80в. И вот когда я выкрутил его в 80в - обрывы прекратились.

cizetta, 2025-05-05:
Thank you for the information, @robot468 . I had mine working for more than 4 hours today with no issues and suddenly it stopped and started again. I just changed the undervoltage to 85V (the minimum from the converter) and I'll monitor it.

wwwodsman, 2025-05-14:
Hello!
Zigbee2MQTT version 2.3.0
How can I add this external converters to my zigbee2mqtt ?
I created a file with the code from this topic /data/external_converters/convertor.js
restarted zigbee2mqtt and nothing.
I tried to add to the file /config/zigbee2mqtt/configuration.yaml
external_converters:
- convertor.js

After restarting zigbee2mqtt this line disappears.
logs
`[2025-05-14 12:10:52] debug:    z2m: No converter available for 'TS0601' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,8,24],"type":"Buffer"},"datatype":2,"dp":20}],"seq":27648}'`

wwwodsman, 2025-05-14:
Created file external_converters/convertor.js
restarted zigbee2mqtt
file convertor.js renamed to convertor.js.invalid

log file

```
[2025-05-14 13:40:29] error:    z2m: Invalid external converter 'convertor.js' was ignored and renamed to prevent interference with Zigbee2MQTT.
[2025-05-14 13:40:29] debug:    z2m: /app/dist/external_converters/convertor.js:1
t fz = require('zigbee-herdsman-converters/converters/fromZigbee');
  ^^

SyntaxError: Unexpected identifier 'fz'
    at wrapSafe (node:internal/modules/cjs/loader:1512:18)
    at Module._compile (node:internal/modules/cjs/loader:1534:20)
    at Object..js (node:internal/modules/cjs/loader:1699:10)
    at Module.load (node:internal/modules/cjs/loader:1313:32)
    at Function._load (node:internal/modules/cjs/loader:1123:12)
    at TracingChannel.traceSync (node:diagnostics_channel:322:14)
    at wrapModuleLoad (node:internal/modules/cjs/loader:217:24)
    at cjsLoader (node:internal/modules/esm/translators:263:5)
    at ModuleWrap.<anonymous> (node:internal/modules/esm/translators:196:7)
    at ModuleJob.run (node:internal/modules/esm/module_job:271:25)
```

What am I doing wrong?
Thanks!

EN500, 2025-06-02:
Thank you for .js file. It works fine

Zerarulez, 2025-06-03:
> Helper- Utility

Никак не могу разобраться со счетчиком киловатт. делаю счетчик ком услуг, выбираю энергию устройства, ставлю флажок на дельту. и все равно по нулям. Можно как то по подробнее?

AntonStrygin, 2025-06-06:
@zvAndrey Спасибо за конвертер!
А есть возможность интегрировать это устройство если Способ установки Home Assistant OS?
В данном случае zigbee2mqtt ставится как дополнение и там не поддерживаются external converters. Только расширения в виде class MyExampleExtension1749206981755 {....
Кто-нибудь уже сталкивался? Что можно сделать в данном случае?

yshrik, 2025-07-06:
> > Вспомогательная утилита
> 
> Никак не могу разобраться со счётчиком киловатт. Создаю счётчик коммунальных услуг, выбираю энергопотребление устройства, ставлю флажок на дельту. И всё равно ноль. Можно как-то подробнее?

У меня тоже не считает энергопотребление

bagamka, 2025-07-07:
Также сталкиваюсь с периодическими спонтанным отключениями. У меня их четыре штуки. Два не отключаются (может фаза другая), а два периодически выключают нагрузку на 1 минуту. Сейчас работают они через zigbee координатора, но пробую  перейти на z2m.  
В программе smartlife есть меню setting/log/switch_log, так вот там как раз и видны эти спонтанные отключения. Причем это не отключения по срабатыванию защиты а просто как будто кто то нажал кнопку выкл а потом вкл. Закономерности отключения не понял, иногда происходят ночью когда нагрузки нет а также может быть неделя без отключений. Как то так.
А как сделать чтобы через z2m можно было смотреть эти логи? Также хотелось бы  иметь доступ к переключателю Light Mode который выключает подсветку. Подсветка постоянно включается при пропадании сети и последующем включении. Неудобно.

yshrik, 2025-07-08:
У меня тоже выключается на небольшой период времени (зелёные стрелки) после того как перестают изменяться показания (отмечено красными стрелками). В результате чего так происходит, не понятно. А за конвертор огромное спасибо.

[image]

bagamka, 2025-07-14:
Who knows which DP is responsible for Light Mode?

manny199, 2025-08-29:
Hi everyone, I apologize for my ignorance. 
I don't understand where I should insert the lines of code from vaga9938 for the external converter definition.
I created a file called _TZE284_5m4nchbm.mjs in the zigbee2mqtt/data/external_converters/ folder, but the device isn't supported yet. My zigbee2mqtt version is 2.5.1.

Thanks so much for your help.

francisc1968, 2025-08-29:
I save it as a js file here:
/homeassistant/zigbee2mqtt/external_converters

francisc1968, 2025-08-29:
_TZE284_5m4nchbm.js

Zerarulez, 2025-09-02:
Всё сломался конвертер не работает 😢 после последнего обновления зигби до версии 2.6.1-1 пишет что конвертер инвалид. А я блин обрадовался обновлению и не сделал резервную копию 🤦

francisc1968, 2025-09-02:
The converter is working again. Thank you.

Stark0008, 2025-09-21:
@zvAndrey  Thanks for the converter! I spent two days trying to make it with neural networks (grok, chatgpt), but none of them were even partially successful.

BytEvil, 2025-10-02:
Hi. z2m 2.6.1
`z2m: Invalid external converter '_TZE284_5m4nchbm.mjs' was ignored and renamed to prevent interference with Zigbee2MQTT.  (require is not defined in ES module scope, you can use import instead)`

wwwodsman, 2025-12-25:
> Most probably it's related to incorrect initial settings of your device and
> some of them are the result of an unexpected shutdown. Try to switch off
> all checks/limits and see how it works after that. If shutdowns are gone,
> then try to switch them back one by one and as a result you'll find the
> reason.
> 
> пн, 15 дек. 2025 г. в 15:41, wwwodsman ***@***.***>:
> […](#)

When I bought it, it started malfunctioning right away.
I disabled all checks on it and, just in case, raised all the values ​​to the maximum possible.
But the behavior remained the same.

closed 2026-03-04 by github-actions[bot]