Koenkk/zigbee2mqtt #26164: DJT11LM "is not a Number" issue after 2.1.0

clowncracker, 2025-02-04:
### What happened?

The following errors are showing up in my logs:

```Logger: homeassistant.components.mqtt.number
Source: components/mqtt/number.py:178
integration: MQTT (documentation, issues)
First occurred: 9:11:29 AM (18 occurrences)
Last logged: 3:38:28 PM

Payload '{"angle":6,"angle_x":5,"angle_x_absolute":85,"angle_y":6,"angle_y_absolute":84,"angle_z":-83,"battery":77,"device_temperature":30,"linkquality":80,"power_outage_count":189,"sensitivity":"medium","strength":26,"vibration":false,"voltage":2965,"x_axis":59,"y_axis":70,"z_axis":-711}' is not a Number
Payload '{"angle":6,"angle_x":5,"angle_x_absolute":85,"angle_y":6,"angle_y_absolute":84,"angle_z":-83,"battery":77,"device_temperature":30,"linkquality":105,"power_outage_count":189,"sensitivity":"medium","strength":16,"vibration":false,"voltage":2965,"x_axis":59,"y_axis":70,"z_axis":-711}' is not a Number
Payload '{"angle":6,"angle_x":5,"angle_x_absolute":85,"angle_y":6,"angle_y_absolute":84,"angle_z":-83,"battery":77,"device_temperature":30,"linkquality":109,"power_outage_count":189,"sensitivity":"medium","strength":16,"vibration":false,"voltage":2965,"x_axis":59,"y_axis":70,"z_axis":-711}' is not a Number
Payload '{"angle":6,"angle_x":5,"angle_x_absolute":85,"angle_y":6,"angle_y_absolute":84,"angle_z":-83,"battery":77,"device_temperature":30,"linkquality":105,"power_outage_count":189,"sensitivity":"medium","strength":7,"vibration":false,"voltage":2965,"x_axis":59,"y_axis":70,"z_axis":-711}' is not a Number
Payload '{"angle":6,"angle_x":5,"angle_x_absolute":85,"angle_y":6,"angle_y_absolute":84,"angle_z":-83,"battery":77,"device_temperature":31,"linkquality":109,"power_outage_count":189,"sensitivity":"medium","strength":7,"vibration":false,"voltage":2965,"x_axis":59,"y_axis":70,"z_axis":-711}' is not a Number`

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus Gateway

### Setup

Add-on

LeoI5, 2025-02-04:
Same error. Rollback to 2.0.0 version

Maggie70, 2025-02-04:
Same error here

Koenkk, 2025-02-04:
Could you provide the z2m debug log of when this happens?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

clowncracker, 2025-02-04:
> Could you provide the z2m debug log of when this happens?
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

```[2025-02-04 12:19:56] debug: 	z2m:mqtt: Received MQTT message on 'homeassistant/sensor/0x00158d0007e06b0b/battery/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":3,"identifiers":["zigbee2mqtt_0x00158d0007e06b0b"],"manufacturer":"Aqara","model":"Vibration sensor","model_id":"DJT11LM","name":"Kitchen.Sensor.Counter-Chair-Left","sw_version":"3000-0001","via_device":"zigbee2mqtt_bridge_0x00124b0024c0b827"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","object_id":"kitchen.sensor.counter-chair-left_battery","origin":{"name":"Zigbee2MQTT","sw":"2.1.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Kitchen.Sensor.Counter-Chair-Left","unique_id":"0x00158d0007e06b0b_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
[2025-02-04 12:19:56] debug: 	z2m:mqtt: Received MQTT message on 'homeassistant/sensor/0x00158d0007e06b0b/device_temperature/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":3,"identifiers":["zigbee2mqtt_0x00158d0007e06b0b"],"manufacturer":"Aqara","model":"Vibration sensor","model_id":"DJT11LM","name":"Kitchen.Sensor.Counter-Chair-Left","sw_version":"3000-0001","via_device":"zigbee2mqtt_bridge_0x00124b0024c0b827"},"device_class":"temperature","enabled_by_default":true,"entity_category":"diagnostic","object_id":"kitchen.sensor.counter-chair-left_device_temperature","origin":{"name":"Zigbee2MQTT","sw":"2.1.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Kitchen.Sensor.Counter-Chair-Left","unique_id":"0x00158d0007e06b0b_device_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.device_temperature }}"}'
[2025-02-04 12:19:56] debug: 	z2m:mqtt: Received MQTT message on 'homeassistant/sensor/0x00158d0007e06b0b/strength/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":3,"identifiers":["zigbee2mqtt_0x00158d0007e06b0b"],"manufacturer":"Aqara","model":"Vibration sensor","model_id":"DJT11LM","name":"Kitchen.Sensor.Counter-Chair-Left","sw_version":"3000-0001","via_device":"zigbee2mqtt_bridge_0x00124b0024c0b827"},"enabled_by_default":true,"name":"Strength","object_id":"kitchen.sensor.counter-chair-left_strength","origin":{"name":"Zigbee2MQTT","sw":"2.1.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Kitchen.Sensor.Counter-Chair-Left","unique_id":"0x00158d0007e06b0b_strength_zigbee2mqtt","value_template":"{{ value_json.strength }}"}'
[2025-02-04 12:19:56] debug: 	z2m:mqtt: Received MQTT message on 'homeassistant/sensor/0x00158d0007e06b0b/angle_x/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":3,"identifiers":["zigbee2mqtt_0x00158d0007e06b0b"],"manufacturer":"Aqara","model":"Vibration sensor","model_id":"DJT11LM","name":"Kitchen.Sensor.Counter-Chair-Left","sw_version":"3000-0001","via_device":"zigbee2mqtt_bridge_0x00124b0024c0b827"},"enabled_by_default":true,"name":"Angle x","object_id":"kitchen.sensor.counter-chair-left_angle_x","origin":{"name":"Zigbee2MQTT","sw":"2.1.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Kitchen.Sensor.Counter-Chair-Left","unique_id":"0x00158d0007e06b0b_angle_x_zigbee2mqtt","unit_of_measurement":"°","value_template":"{{ value_json.angle_x }}"}'
[2025-02-04 12:19:56] debug: 	z2m:mqtt: Received MQTT message on 'homeassistant/sensor/0x00158d0007e06b0b/angle_y/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":3,"identifiers":["zigbee2mqtt_0x00158d0007e06b0b"],"manufacturer":"Aqara","model":"Vibration sensor","model_id":"DJT11LM","name":"Kitchen.Sensor.Counter-Chair-Left","sw_version":"3000-0001","via_device":"zigbee2mqtt_bridge_0x00124b0024c0b827"},"enabled_by_default":true,"name":"Angle y","object_id":"kitchen.sensor.counter-chair-left_angle_y","origin":{"name":"Zigbee2MQTT","sw":"2.1.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Kitchen.Sensor.Counter-Chair-Left","unique_id":"0x00158d0007e06b0b_angle_y_zigbee2mqtt","unit_of_measurement":"°","value_template":"{{ value_json.angle_y }}"}'
[2025-02-04 12:19:56] debug: 	z2m:mqtt: Received MQTT message on 'homeassistant/sensor/0x00158d0007e06b0b/angle_z/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":3,"identifiers":["zigbee2mqtt_0x00158d0007e06b0b"],"manufacturer":"Aqara","model":"Vibration sensor","model_id":"DJT11LM","name":"Kitchen.Sensor.Counter-Chair-Left","sw_version":"3000-0001","via_device":"zigbee2mqtt_bridge_0x00124b0024c0b827"},"enabled_by_default":true,"name":"Angle z","object_id":"kitchen.sensor.counter-chair-left_angle_z","origin":{"name":"Zigbee2MQTT","sw":"2.1.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Kitchen.Sensor.Counter-Chair-Left","unique_id":"0x00158d0007e06b0b_angle_z_zigbee2mqtt","unit_of_measurement":"°","value_template":"{{ value_json.a [... cut ...] gbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
```

```
[2025-02-04 12:19:56] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen.Sensor.Counter-Chair-Left', payload '{"angle":6,"angle_x":5,"angle_x_absolute":85,"angle_y":6,"angle_y_absolute":84,"angle_z":-83,"battery":77,"device_temperature":30,"linkquality":116,"power_outage_count":190,"sensitivity":"medium","strength":8,"vibration":false,"voltage":2965,"x_axis":59,"y_axis":70,"z_axis":-711}'
```

LeoI5, 2025-02-04:
homeassistant.components.mqtt.number
Payload '{"angle":52,"angle_x":6,"angle_x_absolute":84,"angle_y":3,"angle_y_absolute":87,"angle_z":-83,"battery":100,"device":{"applicationVersion":8,"friendlyName":"0x00158d0002c709d8","ieeeAddr":"0x00158d0002c709d8","manufacturerID":4151,"manufacturerName":"LUMI","model":"DJT11LM","networkAddress":21449,"powerSource":"Battery","type":"EndDevice"},"device_temperature":27,"last_seen":"2025-02-03T06:51:38.207Z","linkquality":127,"power_outage_count":286,"sensitivity":"high","strength":6,"vibration":false,"voltage":3025,"x_axis":79,"y_axis":36,"z_axis":-738}' is not a Number
components/mqtt/number.py
I've returned to 2.0.0 version. Above my HA log record. So error was in mqtt.
For z2m log I need some time to install z2m 2.0.1 and wait for errors.

Koenkk, 2025-02-05:
Could you try to change the sensitivity via the z2m frontend once and see if that fixes the issue?

Koenkk, 2025-02-06:
Make sure to wake-up the device right before sending the command.

TimofeyKovalev, 2025-02-06:
Hi! Same error.

I woke up the device, and a couple of seconds after that tried to change its sensitivity. If I'm reading logs right, at 23:13:09 the device was online, and at 23:13:13 I got errors in z2m logs:

```
[2025-02-06 23:13:09] debug: 	z2m: Received Zigbee message from 'snsr.vibration.xiaomi.1stFl.cat-door-RFID', type 'attributeReport', cluster 'closuresDoorLock', data '{"1285":1310720}' from endpoint 1 with groupID 0
[2025-02-06 23:13:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/snsr.vibration.xiaomi.1stFl.cat-door-RFID', payload '{"angle":5,"angle_x":1,"angle_x_absolute":89,"angle_y":77,"angle_y_absolute":13,"angle_z":13,"battery":37,"device_temperature":13,"last_seen":"2025-02-06T23:13:09+03:00","linkquality":176,"power_outage_count":521,"sensitivity":"high","strength":20,"vibration":false,"voltage":2905,"x_axis":20,"y_axis":1051,"z_axis":239}'
[2025-02-06 23:13:10] debug: 	z2m: Received Zigbee message from 'snsr.vibration.xiaomi.1stFl.cat-door-RFID', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":2905,"10":46041,"152":55,"153":572,"154":1017976193044,"3":13,"4":17320,"5":522,"6":5,"8":776}}' from endpoint 1 with groupID 0
[2025-02-06 23:13:10] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/snsr.vibration.xiaomi.1stFl.cat-door-RFID', payload '{"angle":5,"angle_x":1,"angle_x_absolute":89,"angle_y":77,"angle_y_absolute":13,"angle_z":13,"battery":37,"device_temperature":13,"last_seen":"2025-02-06T23:13:10+03:00","linkquality":176,"power_outage_count":521,"sensitivity":"high","strength":20,"vibration":false,"voltage":2905,"x_axis":20,"y_axis":1051,"z_axis":239}'
[2025-02-06 23:13:11] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/snsr.vibration.xiaomi.1stFl.cat-door-RFID', payload '{"angle":5,"angle_x":1,"angle_x_absolute":89,"angle_y":77,"angle_y_absolute":13,"angle_z":13,"battery":37,"device_temperature":13,"last_seen":"2025-02-06T23:13:11+03:00","linkquality":176,"power_outage_count":521,"sensitivity":"high","strength":20,"vibration":false,"voltage":2905,"x_axis":20,"y_axis":1051,"z_axis":239}'
[2025-02-06 23:13:11] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/snsr.vibration.xiaomi.1stFl.cat-door-RFID', payload '{"angle":5,"angle_x":1,"angle_x_absolute":89,"angle_y":77,"angle_y_absolute":13,"angle_z":13,"battery":37,"device_temperature":13,"last_seen":"2025-02-06T23:13:11+03:00","linkquality":176,"power_outage_count":521,"sensitivity":8,"strength":20,"vibration":false,"voltage":2905,"x_axis":20,"y_axis":1051,"z_axis":239}'
[2025-02-06 23:13:13] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/snsr.vibration.xiaomi.1stFl.cat-door-RFID/set' with data '{"sensitivity":14}'
[2025-02-06 23:13:13] debug: 	z2m: Publishing 'set' 'sensitivity' to 'snsr.vibration.xiaomi.1stFl.cat-door-RFID'
[2025-02-06 23:13:13] error: 	z2m: Publish 'set' 'sensitivity' to 'snsr.vibration.xiaomi.1stFl.cat-door-RFID' failed: 'Error: ZCL command 0x00158d0002adc7fb/1 genBasic.write({"65293":{"value":10,"type":32}}, {"timeout":35000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":4447,"writeUndiv":false}) failed (Timeout - 7922 - 1 - 233 - 0 - 4 after 35000ms)'
[2025-02-06 23:13:13] error: 	z2m: Publish 'set' 'sensitivity' to 'snsr.vibration.xiaomi.1stFl.cat-door-RFID' failed: 'Error: ZCL command 0x00158d0002adc7fb/1 genBasic.write({"65293":{"value":9,"type":32}}, {"timeout":35000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":4447,"writeUndiv":false}) failed (Timeout - 7922 - 1 - 234 - 0 - 4 after 35000ms)'
```

And that's what was happening in HA logs at that moment:
```
2025-02-06 23:02:39.214 WARNING (MainThread) [homeassistant.components.mqtt.number] Payload '{"angle":5,"angle_x":1,"angle_x_absolute":89,"angle_y":77,"angle_y_absolute":13,"angle_z":13,"battery":37,"device_temperature":13,"last_seen":"2025-02-06T23:02:24+03:00","linkquality":176,"power_outage_count":521,"sensitivity":"high","strength":17,"vibration":false,"voltage":2905,"x_axis":20,"y_axis":1051,"z_axis":239}' is not a Number
2025-02-06 23:06:21.075 WARNING (MainThread) [homeassistant.components.mqtt.number] Payload '{"angle":5,"angle_x":1,"angle_x_absolute":89,"angle_y":77,"angle_y_absolute":13,"angle_z":13,"battery":37,"device_temperature":13,"last_seen":"2025-02-06T23:06:21+03:00","linkquality":176,"power_outage_count":521,"sensitivity":"high","strength":19,"vibration":false,"voltage":2905,"x_axis":20,"y_axis":1051,"z_axis":239}' is not a Number
2025-02-06 23:09:15.047 WARNING (MainThread) [homeassistant.components.mqtt.number] Payload '{"action":"vibration","angle":5,"angle_x":1,"angle_x_absolute":89,"angle_y":77,"angle_y_absolute":13,"angle_z":13,"battery":37,"device_temperature":13,"last_seen":"2025-02-06T23:09:15+03:00","linkquality":176,"power_outage_count":521,"sensitivity":"high","strength":19,"vibration":true,"voltage":2905,"x_axis":20,"y_axis":1051,"z_axis":239}' is not a Number
2025-02-06 23:09:30.046 WARNING (MainThread) [homeassistant.components.mqtt.number] Payload '{"angle":5,"angle_x":1,"angle_x_absolute":89,"angle_y":77,"angle_y_absolute":13,"angle_z":13,"battery":37,"device_temperature":13,"last_seen":"2025-02-06T23:09:15+03:00","linkquality":176,"power_outage_count":521,"sensitivity":"high","strength":19,"vibration":false,"voltage":2905,"x_axis":20,"y_axis":1051,"z_axis":239}' is not a Number
[... cut ...]
2025-02-06 23:13:11.226 WARNING (MainThread) [homeassistant.components.mqtt.number] Payload '{"angle":5,"angle_x":1,"angle_x_absolute":89,"angle_y":77,"angle_y_absolute":13,"angle_z":13,"battery":37,"device_temperature":13,"last_seen":"2025-02-06T23:13:11+03:00","linkquality":176,"power_outage_count":521,"sensitivity":"high","strength":20,"vibration":false,"voltage":2905,"x_axis":20,"y_axis":1051,"z_axis":239}' is not a Number
```

Proxmox VM, HAOS 14.2, HA 2025.1.4
Zigbee2MQTT 2.1.0
SONOFF Zigbee 3.0 USB Dongle Plus v. 6.10.3.0 build 297

fenty17, 2025-02-07:
I have this issue too. I tried manually changing the sensitivity level after waking the device but it doesn’t stick. Seems the old setting of “medium” is persisting and now not recognised as a number.

jeremysherriff, 2025-02-10:
After several attempts I seem to have a working around the error.  I cannot tell you exactly what I did, but it involved a combination of;

- Pressing the button on the device (blue light flashes 2-3 times)
- Attempting to set the sensitivity via the Z2M web gui (not via the HA integration!)
- Initiating a Rediscovery of the device via Z2M web gui (yellow button: [image] )

It took several attempts but the last error was logged in HA about 4 hours ago.

McGiverGim, 2025-02-12:
I enabled the join to zigbee, pressed the button for several seconds, and after that modified the sensitiviy from Home Assistant. It seems it worked for me.

zslibra, 2025-02-14:
Same error here

Nevynnyi, 2025-02-17:
the same: Sensitivity not found in: [low, medium, high] and no Angle, axis
Manual publish topic to Sensitivity as Low, Medium and High change numeric value and work.

jocamane, 2025-02-27:
#26528

jocamane, 2025-02-27:
The error disappeared. I changed the sensitivity, pressed the device button and moved it to change the values. Something like that made the error go away.

closed 2025-05-06 by github-actions[bot]