Koenkk/zigbee2mqtt #26168: TS0601_dimmer_1_gang_1 don't work as expected on Zigbee2MQTT 2.1.0-1

pjagielski0, 2025-02-04:
### What happened?

After updating to version "2.1.0-1", the dimmer does not work from the device level.
After turning it on or off, it does not provide any information to the system.
If I turn on the light/switch from the MQTT level, the device works as it should. From the MQTT level, you can also dim/brighten. However, these actions do not work the other way around, i.e. from the device side.
I noticed that in "MQTT discovery data" there is no "Topic: homeassistant/light/0x3425b4fffe387ea1/light/config".

### What did you expect to happen?

I should physically be able to turn it on and off, and be able to use the dimmer function.

### How to reproduce it (minimal and precise)

Just update Zigbee2MQTT to version "2.1.0-1"

### Zigbee2MQTT version

2.1.0-1

### Adapter firmware version

 20240710

### Adapter

XZG UZG-01

### Setup

add-on

losip, 2025-02-04:
I can confirm this problem. But in my case, if I switch ON the light from the physical dimmer, its state IS reported to z2m but if I turn it OFF from the dimmer, there is no message seen in mqtt explorer. I use the switch state to trigger other automations regardless of whether the state is changed from Home Assistant or locally on the dimmer.

Mikescotland, 2025-02-04:
Same here. Moes dimmers.
If switched on / off by hand, the status does not get updated in z2m.

LeonCame, 2025-02-04:
I allow myself to add a related problem: in this reference ts0601_dimmer_1_gang_1: you included the Moes Zigbee Smart Touch Light Dimmer. The light_type ** always rocks in LED mode **, whatever the choice: LED / Incandescent / Halogen. This mode works properly via a Tuya coordinator with the same device.

MauiR, 2025-02-04:
Since updating to Zigbee2MQTT version 2.1.0, I also have been experiencing issues with my dimmer.
For me Turning the light on and off via Zigbee2MQTT works without any problems. However, there is an other strange behavior:
When the light is turned off via an automation in Home Assistant, it physically turns off, but its status in Zigbee2MQTT does not update.

In Zigbee2MQTT, the light still appears as on, even though it is actually off.
As a result, Home Assistant also incorrectly shows the light as on.

losip, 2025-02-05:
I've just re-installed the latest Edge version and it seems to be fixed. See https://github.com/Koenkk/zigbee-herdsman-converters/pull/8723. I shall now stay on the Edge version until 1st March

Koenkk, 2025-02-05:
Could you see if this is fixed in the latest dev branch?

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Koenkk, 2025-02-06:
Fixed in 2.1.1

closed 2025-02-06 by Koenkk

LeonCame, 2025-02-07:
Version Zigbee2MQTT [2.1.1]

For information this problem still exists:

in this reference ts0601_dimmer_1_gang_1: you included the Moes Zigbee Smart Touch Light Dimmer, I'm talking about the one who looks exactly like this: https://www.zigbee2mqtt.io/devices/EDM-1ZBB-EU.html

Problem: the `light_type` always rocks in `LED mode`, whatever the choice: LED / Incandescent / Halogen.

This mode works properly via a Tuya coordinator (and Tuya App) with the same device.

I specify that the LED / Incandescent / Halogen mode is well memorized and postponed in Z2M but it has **no effect on hardware**. I have LEDs but I prefer halogen mode because it allows you to descend the much lower LEDs brightness.

I just postpone the problem, I help out by configuring the light_type via the Tuya app, then I once again repatriate the Device under Z2M. You must then no longer touch the light_type, otherwise it returns to LED mode whatever the choice.