Koenkk/zigbee2mqtt #26169: Zemismart ZM16EL-03/33 no longer reporting position after updating to 2.1.0

sh-nguyen, 2025-02-04:
### What happened?

Choosing the open and close states causes the blind to move to the respective endstop but the position is no longer updated to reflect 0 or 100%.

### What did you expect to happen?

Position to change to 0 or 100% when the blind is completely opened or closed.

### How to reproduce it (minimal and precise)

Updated Zigbee2Mqtt to 2.1.0
Pressed Open, blind physically moved, no change to position,  
Pressed Close, blind physically moved, no change to position 

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

7.4.4 [GA]

### Adapter

SONOFF ZBDongle-E

### Setup

Docker Container

### Debug log

info 2025-02-04 17:36:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living Room Blind', payload '{"battery":100,"border":"down","click_control":"up","linkquality":196,"motor_fault":true,"position":0,"reverse_direction":"forward","state":"OPEN"}'
info 2025-02-04 17:36:50z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living Room Blind', payload '{"battery":100,"border":"down","click_control":"up","linkquality":200,"motor_fault":true,"position":0,"reverse_direction":"forward","state":"OPEN"}'
info 2025-02-04 17:36:50z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living Room Blind', payload '{"battery":100,"border":"down","click_control":"up","linkquality":204,"motor_fault":true,"position":0,"reverse_direction":"forward","state":"OPEN"}'
info 2025-02-04 17:36:50z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living Room Blind', payload '{"battery":100,"border":"down","click_control":"up","linkquality":204,"motor_fault":true,"position":100,"reverse_direction":"forward","state":"OPEN"}'
info 2025-02-04 17:37:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living Room Blind', payload '{"battery":100,"border":"down","click_control":"up","linkquality":192,"motor_fault":true,"position":100,"reverse_direction":"forward","state":"OPEN"}'
info 2025-02-04 17:37:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living Room Blind', payload '{"battery":100,"border":"down","click_control":"up","linkquality":192,"motor_fault":true,"position":100,"reverse_direction":"forward","state":"OPEN"}'
info 2025-02-04 17:37:21z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living Room Blind', payload '{"battery":100,"border":"down","click_control":"up","linkquality":192,"motor_fault":true,"position":100,"reverse_direction":"forward","state":"CLOSE"}'
info 2025-02-04 17:38:06z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living Room Blind', payload '{"battery":100,"border":"down","click_control":"up","linkquality":192,"motor_fault":true,"position":100,"reverse_direction":"forward","state":"STOP"}'
info 2025-02-04 17:38:10z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living Room Blind', payload '{"battery":100,"border":"down","click_control":"up","linkquality":196,"motor_fault":true,"position":100,"reverse_direction":"forward","state":"OPEN"}'

Globetrotter27, 2025-02-12:
I have a similar problem.

I'm using 2 of this motors for about a year and the problem showed only on one of them. Second one works same as before the update. 

Thought the motor/battery died (according to "motor_fault":true flag and battery showing 100%/0% charge without any other intermediate values). Motor react sporadically, or doesn't react at all, not only to ZB commands but to RF remote also. So I thought it's hardware related issue.

Globetrotter27, 2025-02-15:
No, problem exists on 2.1.1 currently running (skipped upgrade to 2.1.0). I think it's motor hardware related problem, because I can't exactly link the moment of Z2M upgrade to the motor start being faulty. And again, the second (exactly the same) motor I have still works with no issues at all.

sh-nguyen, 2025-02-15:
I've only observed the issue with one of my blinds. I have contacted the manufacturer and they have advised me to replace it as they believe it is a motor fault, so it seems that Z2M is behaving correctly.

closed 2025-02-15 by sh-nguyen