Koenkk/zigbee2mqtt #26171: Automation not triggered after update

AnsusC, 2025-02-04:
### What happened?

After updating several devices are not working anymore as expected. Button presses are not fired in Automations. 
Some lamps ( not all ) are not marked as unknown - automations broke.

### What did you expect to happen?

The device is recognized in Zigbee2mqtt itself and I can see the actions if I for example press a button. The events itself are recognized.

But the automation that will use the device are not triggered.

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

20210708

### Adapter

zStack3x0

### Setup

Intel NUC

J1tata, 2025-02-04:
Same here - noted with Mueller Licht Tint Remote 404011. 
I enabled `legacy_action_sensor: true` as the [blueprint](https://gist.github.com/SeLLeRoNe/c3494463240e100fdfdb9abeed968113) is not updated to use the "new" recommended way as of now. 

When checking under State of the remote in Z2M, I can see that clicking the buttons will update the state like to "on" or "off" and immediately changing back to "". I can see also the status change with MQTT Explorer.  

Running:
Z2M: 2.1.0-1
HA Core: 2025.1.4
HA Supervisor: 2024.12.3
HA OS: 14.2

closed 2025-02-04 by AnsusC

J1tata, 2025-02-04:
@AnsusC: Did you find a solution?