Koenkk/zigbee2mqtt #26176: After updating to 2.1.0 - curtain motor TS0601 (_TZE204_r0jdjrvi) does not show the position correctly. (The position does not change)

autosvet22, 2025-02-04:
### What happened?

 After updating to version 2.1.0, the positions of the "TS0601_lilistore (_TZE204_r0jdjrvi) curtain motors are displayed incorrectly. 

### What did you expect to happen?

Maybe this can be fixed? I have two such devices. Both behave the same. I can't use the latest version of the addon because of this problem.

### How to reproduce it (minimal and precise)

The motor receives commands, executes them, but does not transmit data about its actual position. (the position status does not change)

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

20230507

### Adapter

 zStack3x0

### Setup

HA addon x86-64

Koenkk, 2025-02-04:
Should be fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/8723

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

mnowok, 2025-02-05:
I have the same problem with similar blinds TS0601 (_TZE200_icka1clh). I've just created an issue #26193.

autosvet22, 2025-02-10:
Fixed in update 2.1.1. Thanks for the prompt solution.