Koenkk/zigbee2mqtt #26177: Z2M not starting anymore -reason unclear

Tempel12, 2025-02-04:
### What happened?

Suddenly my Z2M Home Assistant Add on is not starting. I specified my adapter under configuration. Till yesterday everything was running fine. Today the add on is not starting. LOG:

[14:34:38] INFO: Preparing to start...
[14:34:38] INFO: Socat not enabled
[14:34:38] INFO: Starting Zigbee2MQTT...
node:internal/modules/cjs/loader:1252
  throw err;
  ^
Error: Cannot find module 'glob'
Require stack:
- /app/node_modules/.pnpm/rimraf@6.0.1/node_modules/rimraf/dist/commonjs/index.js
- /app/index.js
    at Function._resolveFilename (node:internal/modules/cjs/loader:1249:15)
    at Function._load (node:internal/modules/cjs/loader:1075:27)
    at TracingChannel.traceSync (node:diagnostics_channel:315:14)
    at wrapModuleLoad (node:internal/modules/cjs/loader:218:24)
    at Module.require (node:internal/modules/cjs/loader:1340:12)
    at require (node:internal/modules/helpers:141:16)
    at Object.<anonymous> (/app/node_modules/.pnpm/rimraf@6.0.1/node_modules/rimraf/dist/commonjs/index.js:7:16)
    at Module._compile (node:internal/modules/cjs/loader:1546:14)
    at Object..js (node:internal/modules/cjs/loader:1689:10)
    at Module.load (node:internal/modules/cjs/loader:1318:32) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/app/node_modules/.pnpm/rimraf@6.0.1/node_modules/rimraf/dist/commonjs/index.js',
    '/app/index.js'
  ]
}
Node.js v22.11.0

### Zigbee2MQTT version

2.1.0-1

### Adapter firmware version

frontend not reachable

### Adapter

zstack

### Setup

Add on Home Assistant VM

Cjkeenan, 2025-02-05:
What did you do to fix this?

Tempel12, 2025-02-05:
My configuration.yaml was wrong. It was my own fault, so I closed the issue. There was also an issue with the breaking changes since Version 2 of Zigbee2MQTT. I added this

advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false

To my yaml.