Koenkk/zigbee2mqtt #26179: IKEA E2013 PARASOLL door sensor reports wrong battery percentage

Plasticas, 2025-02-04:
### What happened?

Hi,

IKEA PARASOLL E2013 door sensor always reports 100% battery until its dead. Today measured real voltage and got 0.68V left, last report was 100%.

Firmware build date
20230516
Firmware version
1.0.19

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

7.4.4 [GA]

### Adapter

EmberZNet

### Setup

RPi4, HAOS

pontuslandin89, 2025-02-11:
Same here, you seem to get the battery value that you had during pairing of device, no update since then. I tried MQTT zigbee2mqtt/FRIENDLY_NAME/get with payload {"battery": ""} but it only reports whatever is in Zigbee2MQTT, its not fetching a fresh value from device.

bitbike, 2025-02-15:
I think this is related to the open issue #22579 

I have the same problem with different Hardware (Odroid N2+ & Home Bridge).

x4N70pHyLL, 2025-02-17:
Same here, they just drop dead at some point and with that you figure out that you have to change the battery.

evorgevol, 2025-03-18:
Same issue. I have had multiple die on me while all still reporting 100%. When I've taken the batteries out, the battery is definitely dead. 

Not sure it's related to the linked open issue, as that seems to be the device just randomly going offline, but mine is due to it reporting an incorrect battery level.

x4N70pHyLL, 2025-05-19:
Still an issue, please remove the label. It is probably related with https://github.com/Koenkk/zigbee2mqtt/issues/22579
but another issue. Some of my Parasoll are reporting battery (value changes/decreases over time), some only pretend to.

x4N70pHyLL, 2025-07-27:
Still an issue!