Koenkk/zigbee2mqtt #26180: After updating to 2.1.0 - roller blind motor TS0601 (_TZE200_zah67ekd) does not update position and state

reaper7, 2025-02-04:
### What happened?

After updating from 2.0.0 to 2.1.0 - roller blind motor TS0601 (_TZE200_zah67ekd) does not update position (0..100) and state (OPEN CLOSE STOP)

Additionally I don't understand the intention of the state (from many versions back for this motor):
 - state OPEN appears only for the duration of the roller shutter movement when opening and when it's fully open state changes to STOP
 - state CLOSE appears only for the duration of the roller shutter movement when closes and when it is completely closed state changes to STOP
so without physical observation, or without checking the POSITION value, you can't tell conclusively whether the shutter is open or closed

### What did you expect to happen?

Previously, the position was updated while the roller shutter was moving and when the operation ended.
The same for the current state (OPEN CLOSE STOP), this value in version 2.1.0 does not change.

### How to reproduce it (minimal and precise)

nothing special - install version 2.1.0

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

20240710

### Adapter

Electrolama

### Setup

x86-64

Serios, 2025-02-04:
I can confirm the issue, have same devices with same behavior on v2.1

@reaper7 Regarding the state, as I remember there was some problem for z2m to properly handle the states of the device combined with the buttons for issuing the Open/Close commands... don't remember quite well. However you are right, while closing the state should be Closing not Close, the same is true for opening/open. Another thing is that these blinds are not quite supported, I mean the operational part is fine, however the Deconz guys somehow managed to get working battery status and lux value from the solar panel

Koenkk, 2025-02-05:
This has been fixed in the dev branch already

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-02-05 by Koenkk

Matt2k34, 2025-02-06:
> This has been fixed in the dev branch already
> 
> > Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Considering how impactful this issue is, and the fact (I can't see where) you'd be able to perform the edit in https://github.com/Koenkk/zigbee-herdsman-converters/pull/8723 which would persist, any plans to push an out of band release?? 

As things like the haasio addon are intended for a more "plug and play" experience, which are currently broken.

For me - my TRV's were still reporting - and could be seen in the logs reporting correctly, but no update made it to HA - meaning from a user POV it's difficult to see what's "wrong" - for me it killed my heating.

Ontop of HA essentially removing auto-backup on upgrade, (which has been "kinda" reverted here -> https://www.home-assistant.io/blog/2025/02/05/release-20252/#backup-on-updating-add-ons) - it makes this messy / open to risk for less technical users, as they can't just roll-back to the previous version, and unless I'm missing something, documentation to move from stable to edge within the addon store isn't particularly user friendly, either.

Thank you for the swift fix (and the product in general!), I just think it would be "better" from a user POV if this was pushed out as its quite a breaking change.

Koenkk, 2025-02-07:
> I just think it would be "better" from a user POV if this was pushed out

Fix is available in the 2.1.1 release