Koenkk/zigbee2mqtt #26181: Wrong CO2 value exposed with Release V 2.1.0

k3067e3, 2025-02-04:
### What happened?

I made today a fresh install of Zigbee2Mqtt on a new SD Card.
Raspberry 3b, followed installation guide for new install. 
For configuration.yaml I copied my old file, coming from version 1.41.0 to data directory, everything was migrated, and all functions are working fine so far.

The only issue I found is the following. 
I am using https://www.zigbee2mqtt.io/devices/DIYRuZ_AirSense.html
With 1.41.0 CO2 Value in Zigbee2MQTT Logfile is the following:

 2025-02-04 17:57:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/CO2_Sensor/co2', payload '460'

When I switch the SD Card with 1.41.0 in my raspberry to the SD Card with the new version, I get from the same sensor, within only a few minutes the following result:

2025-02-04 18:00:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/CO2_Sensor/co2', payload '2164'

So same sensor, no additional CO2 in this few minutes, only differenc ist the Zigbee2MQTT Version

### What did you expect to happen?

Getting the same value with old and new Zigbee2MQTT Version

### How to reproduce it (minimal and precise)

Change SD Card with version 1.41.0 vs 2.1.0

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle

### Setup

Raspberry PI 3b, plain installation, followed Linux installation guide

Zerkon77, 2025-02-05:
Same sensor. Same problerm after update to 2.1.0. Seem like baseline of sensor about 2900-3000 and value of CO2 subtracted from it. When I'm in room value decreasing to 1800, when I ventilate room - increasing to 2500.

Koenkk, 2025-02-05:
Should be fixed now! let me know if it works.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-02-05 by Koenkk

k3067e3, 2025-02-06:
With https://github.com/Koenkk/zigbee-herdsman-converters/releases/tag/v21.29.0 values look fine.

Switching to dev branch was a little bit tricky. 
Needed https://github.com/Koenkk/zigbee2mqtt/discussions/17938#discussioncomment-8016525 additional to 
https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html but then everything was ok.