Koenkk/zigbee2mqtt #26182: [New device support]: Aqara LED Bulb T2

ai, 2025-02-04:
### Link

https://eu.aqara.com/en-eu/products/aqara-led-bulb-t2-e27

### Database entry

{"id":40,"type":"Router","ieeeAddr":"0x54ef4410010fcfc5","nwkAddr":34789,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.light.agl003","epList":[1,21],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[5,4,3,0,768,8,6,2820,1794,64704],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.light.agl003","manufacturerName":"Aqara","powerSource":4,"zclVersion":3,"appVersion":27,"stackVersion":27,"hwVersion":1,"dateCode":"20250113"}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":0,"inClusterList":[12],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":27,"stackVersion":27,"hwVersion":1,"dateCode":"20250113","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1738692849080}

### Zigbee2MQTT version

2.1.0

### Comments

It supports both Zigbee and Thread. User will need Aqara Home app to switch lamp mode to Zigbee.

### External definition

```shell
Autogenerated:

const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['lumi.light.agl003'],
    model: 'lumi.light.agl003',
    vendor: 'Aqara',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"21":21}}), m.light({"colorTemp":{"range":[111,500]},"color":true}), m.electricityMeter()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Works: I can turn off/on, change color and brightness

Don't work: can’t change `Power-on behavior` and Power/Voltage/etc shows `zero`.

coldridj, 2025-02-10:
I have some of these bulbs as well - there are [several variants](https://www.aqara.com/wp-content/uploads/2025/01/1464-600-10.png) depending on which market you are in (I am EU) and whether they support CCT or CCT + RGB

Mine report as `lumi.light.agl004` rather than `agl003` as yours do.

I had a go at extending the converter using some of the available functions from the lumi library and have the power on behaviour working and device temperature reporting in addition to state, brightness and cct. 

Power measurement did not work with either of the existing metering functions from this library, and I am most interested in exposing the power on/off fade timeout. I can see relevant looking nodes in the database so will persist with testing these more thoroughly.

```js
const { lumiModernExtend } = require('zigbee-herdsman-converters/lib/lumi');
const { presets } = require('zigbee-herdsman-converters/lib/exposes');
const { metering } = require('zigbee-herdsman-converters/converters/fromZigbee');

const definition = {
    zigbeeModel: ['lumi.light.agl004'],
    model: 'T2 E27 CCT',
    vendor: 'Aqara',
    description: 'LED Bulb T2 CCT',
    extend: [
        lumiModernExtend.lumiLight({colorTemp: true, powerOutageMemory: false}),
        lumiModernExtend.lumiZigbeeOTA(),
        lumiModernExtend.lumiPowerOnBehavior(),
};

module.exports = definition;
```
 
My database entry appears to contain substantially more than yours:

```json
{"id":64,"type":"Router","ieeeAddr":"0x54ef441000efefcf","nwkAddr":18013,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.light.agl004","epList":[1,21],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[5,4,3,0,768,8,6,2820,1794,64704],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"61441":0,"modelId":"lumi.light.agl004","manufacturerName":"Aqara","powerSource":4,"zclVersion":3,"appVersion":27,"stackVersion":27,"hwVersion":1,"dateCode":"20250113"}},"lightingColorCtrl":{"attributes":{"245":4611445,"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370,"colorTemperature":290,"colorMode":2}},"haElectricalMeasurement":{"attributes":{"acPowerDivisor":10,"acPowerMultiplier":1}},"manuSpecificLumi":{"attributes":{"22":"56DD9AEEF565CEBA1B8788CBD72C5F1EEC93C749B04FD827BF0E6B8FFC3A7E1D","223":{"type":"Buffer","data":[19,33,0,0,20,33,0,0,22,33,0,0,23,33,0,0,12,33,0,0,24,33,115,4]},"229":{"type":"Buffer","data":[23,15,8,149,192,37,89,67,193,37,0,127,204,37,50,193,221,37,84,82,235,37,40,173,240,37,88,201,244,37,102,167,249,37,86]},"230":0,"247":{"type":"Buffer","data":[3,40,33,5,33,2,0,8,33,27,1,9,33,0,23,10,33,0,0,11,32,0,12,32,10,160,32,1,154,32,0,100,32,1,101,32,154,102,33,34,1,106,35,140,10,100,25,105,32,0,113,32,1]},"1301":1,"1302":100,"1303":1,"1320":0.800000011920929,"1324":0,"65522":{"type":"Buffer","data":[170,117,19,132,201,129,7,65,0,15,12,35,1,4,0,23,10,35,8,0,12,100,11,32,4]}}},"genLevelCtrl":{"attributes":{"245":4611453,"onTransitionTime":15,"offTransitionTime":15,"currentLevel":154}},"genOnOff":{"attributes":{"245":4611522,"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":0,"inClusterList":[12],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":0}}},"binds":[{"cluster":12,"type":"endpoint","deviceIeeeAddress":"0x00124b00215fff90","endpointID":1}],"configuredReportings":[{"cluster":12,"attrId":85,"minRepIntval":10,"maxRepIntval":3600,"repChange":5}],"meta":{}}},"appVersion":27,"stackVersion":27,"hwVersion":1,"dateCode":"20250113","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1739184043604}
```

z2m version 2.1.1

coldridj, 2025-02-15:
I obtained an Aqara hub to start exercising the available controls and have a mostly working prototype.

I don't believe the power metering functionality is implemented, or if it is it may be exposed through some of the buffers in the `manuSpecificLumi` cluster rather than on `seMetering` and `haElectricalMeasurement`. I havent yet attempted to decode these and have focused on the controls available in the Aqara Home App.

The UI controls need more work:

- It is possible to cross over the min and max dimmer range values, these could also be exposed in a composite control similar to transition time. 
- Transition time is currently configured in 1/10ths of a second, converting the unit would provide a better experience as the granulartiy appers to be 0.5s.
-  I would like to enhance the CCT control to report the more familiar kelvin instead of the mired value.

I intend to obtain an RGB/CCT bulb and will add additional controls then.

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;

const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const { lumiModernExtend, fromZigbee, manufacturerCode } = require('zigbee-herdsman-converters/lib/lumi');

const lumiTransitionCurve = () => 
    modernExtend.numeric({
        name: 'transition_curve_curvature',
        label: 'Transition Curve Curvature',
        cluster: 'manuSpecificLumi',
        attribute: {ID: 0x0528, type: 0x39},
        description: 'Range 0.2>x>1: first fast then slow, 1: linear, 1>x>6: first slow then fast',
        entityCategory: 'config',
        zigbeeCommandOptions: {manufacturerCode},
        valueMin: 0.2,
        valueMax: 6,
        valueStep: 0.01,
    });

const lumiTransitionBrightness = () => 
    modernExtend.numeric({
        name: 'transition_initial_brightness',
        label: 'Initial Brightness',
        cluster: 'manuSpecificLumi',
        attribute: {ID: 0x052c, type: 0x28},
        description: 'After turning on the light, the light will brighten from the preset brightness',
        entityCategory: 'config',
        zigbeeCommandOptions: {manufacturerCode},
        unit: '%',
        valueMin: 0,
        valueMax: 50,
        valueStep: 1,
    });

const lumiLevelConfig = () => {
    let levelConfig = new exposes.Composite('level_config', 'level_config', exposes.access.ALL);
    levelConfig = levelConfig.withFeature(
        new exposes.Numeric('on_transition_time', exposes.access.ALL)
            .withLabel('ON transition time')
            .withDescription(
                'Represents the time taken to move the current level from the minimum level to the maximum level when an On command is received',
            )
            .withUnit('1/10th second')
            .withValueMax(600)
            .withValueStep(5),
    );
    levelConfig = levelConfig.withFeature(
        new exposes.Numeric('off_transition_time', exposes.access.ALL)
            .withLabel('OFF transition time')
            .withDescription(
                'Represents the time taken to move the current level from the maximum level to the minimum level when an Off command is received',
            )
            .withUnit('1/10th second')
[... 73 lines cut ...]
    const transitionBrightness = lumiTransitionBrightness();
    lightT2.toZigbee.push(...transitionBrightness.toZigbee);
    lightT2.exposes.push(...transitionBrightness.exposes);

    const dimmingMin = lumiDimmingRangeMinimum();
    lightT2.toZigbee.push(...dimmingMin.toZigbee);
    lightT2.exposes.push(...dimmingMin.exposes);

    const dimmingMax = lumiDimmingRangeMaximum();
    lightT2.toZigbee.push(...dimmingMax.toZigbee);
    lightT2.exposes.push(...dimmingMax.exposes);

    return lightT2;
}

const definition = {
    zigbeeModel: ['lumi.light.agl004'],
    model: 'T2 E27 CCT',
    vendor: 'Aqara',
    description: 'LED Bulb T2 CCT',
    extend: [ lumiLightT2() ],
};

module.exports = definition;
```

hoppel118, 2025-02-15:
I really appreciate what you guys are doing here. Is this for both types of T2 bulbs (GU10, E27)?

coldridj, 2025-02-15:
> I really appreciate what you guys are doing here. Is this for both types of T2 bulbs (GU10, E27)?

I would assume the functions are the same as for E27 which is what I have been testing, they may report a different model id which would need adding to the converter.

hoppel118, 2025-02-15:
I have the following bulbs:

- LED Bulb T2 (GU10, RGB CCT) (6149)
- LED Bulb T2 (E27, RGB CCT) (6147)

I'm really new to Zigbee2MQTT and don't really understand what I need to do to make them compatible. Right now they are paired with Home Assistant via Matter over Thread.

But let me check the model IDs next week. I'll report them back here then.

Is there any other information you are interested in that I can easily find out?

coldridj, 2025-02-17:
> I'm really new to Zigbee2MQTT and don't really understand what I need to do to make them compatible. Right now they are paired with Home Assistant via Matter over Thread.

I dont use HA, matter or thread. In order to pair them with z2m and get the id's you'll need to put them into pairing mode, and start the process of adding an accessory to via the Aqara home app on your phone. The first screen you see after discovering the device will allow you to switch protocol to Zigbee - which will go through the firware upload process and then the device will enter Zigbee pairing mode at which time you can enable joining on z2m and it should be discovered. 

From there you can go to the device card in z2m and from the dev console tab click generate_external_definition - this will show you the model id. You can also access the databases.db file in your z2m data directory and copy the corresponding entry (likely the last).

If you want to go back to matter protocol, repeat the above process in aqara home after entering pairing mode.

> Is there any other information you are interested in that I can easily find out?

Database entry for the GU10 RGB in particular as I dont have or need any of this model.

SalGnt, 2025-02-17:
I got the same definition as the OP with the EU (E27, RGB CCT) variant. I can help with tests if needed.

In the attached archive, you can find the E27 images.

[T2_E27.zip](https://github.com/user-attachments/files/18829375/T2_E27.zip)

MarcelMrcn, 2025-02-19:
Great to see people working on this. Thank you to everyone!
The GU10 (RGB CCT) model reports as lumi.light.agl005
The generated external defintion is the same as OP's, apart from the model number.

coldridj, 2025-02-19:
I've obtained an E27 RGB CCT and can confirm this is reporting as `agl003` - I've updated the gist and added rgb control functionality for the two models we have confirmed.

I will try and obtain a GU10 CCT to get the model id, my assumption is this will be `agl006` as the numbers appear to crrespond to their position on the variants table.

hoppel118, 2025-02-20:
@coldridj Thanks for the explanation.

Here are the generate_external_definition outputs:

**Aqara T2 GU10 (RGB CCT)**

```
const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['lumi.light.agl005'],
    model: 'lumi.light.agl005',
    vendor: 'Aqara',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"21":21}}), m.light({"colorTemp":{"range":[111,500]},"color":true}), m.electricityMeter()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

**Aqara T2 E27 (RGB CCT)**

```
const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['lumi.light.agl003'],
    model: 'lumi.light.agl003',
    vendor: 'Aqara',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"21":21}}), m.light({"colorTemp":{"range":[111,500]},"color":true}), m.electricityMeter()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

So nothing new. All we need now is the model of the GU10 CCT bulb.

myrwon, 2025-02-23:
nice to see some trackion on this. I'm very new to this. I'm not even sure where to place all this information.... 
I have the aqara T2 E26 CCT (US)...am to toggle on/off, white to yellow, and dim but getting Z2M errors regarding payload, temp and such

`const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['lumi.light.agl002'],
    model: 'lumi.light.agl002',
    vendor: 'Aqara',
    description: 'Automatically generated definition T2 Aqara',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"21":21}}), m.light({"colorTemp":{"range":[153,370]}}), m.electricityMeter()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;`

coldridj, 2025-02-25:
I think @myrwon confirms the theory that the `alg00x` corresponds to the row in the [variant table](https://www.aqara.com/wp-content/uploads/2025/01/1464-600-10.png) - I have speculatively added the four NA region bulbs and the missing EU GU10 CCt to the [converter gist](https://gist.github.com/coldridj/527b05575f8b94197a0ef805904b7893)

- For those following this thread and unsure what to do with this file see [how to add an external converter](https://www.zigbee2mqtt.io/advanced/more/external_converters.html)

I have been experimenting with the RGB bulb and noticed that it reports a larger colour temperature range than the CCT (likely mixing in RGB) and have updated this range for those bulbs as well.

I am not in a rush to get a PR open on the converters repository as there are a number of features I want to explore further before doing so, and all of the low level zigbee protocol details are new to me:

- Explore how to expose and send the level and cct/color control commands (these can be executed via the dev console)
  -  see [zigbee cluster spec 3.10.2.3, 5.2.2.3 etc](https://github.com/Koenkk/zigbee-herdsman/blob/master/docs/07-5123-08-Zigbee-Cluster-Library.pdf)
- Explore how to save and recall the RGB custom scenes - this will likely involve some zigbee packet sniffing with my CC2531 [bookmark](https://github.com/Koenkk/zigbee2mqtt/discussions/13976)
  - note: I suspect these arent stored/exposed in manuSpecificLumi in a way that is currently exposed in the z2m database
- Explore exposing RGB control in 8bit rgb web colour notation - I think this is a lot more user friendly than CIE 1931
 - see: [related stack overflow discussion](https://stackoverflow.com/questions/70612781/how-to-convert-ikea-light-bulb-color-xy-cie-1931-colorspace-to-rgb) - there must already be code in the converter utilities to do this as the colour picker control has to use web colours
- Above mentioned improvements to z2m device card controls:
  - on/off transition time unit transaltion to native 1/10th second
  - CCT exposed in Kelvin
  - validate/restrict dimmer range crossover point

I believe the converter in its current form covers the majority of the functionality of these bulbs and should hopefully get these useable for the majority of situations in the meantime.

Anyone with time/experience/enthusiasm to contribute to hte above or improve my hacky JS is welcome to open a dicsussion or start comment thread on the gist.

Flumeded, 2025-02-25:
Really appriciate everyone for contrubuting!
added converter to test with T2 (E27, CCT) [EU], power on control is a god send!

myrwon, 2025-02-25:
also can confirm that using the [converter gist](https://gist.github.com/coldridj/527b05575f8b94197a0ef805904b7893), and placing the JS file to /zigbee2mqtt/external_converters. Z2M is now showing supported

coldridj, 2025-02-26:
[converter revision 4](https://gist.github.com/coldridj/527b05575f8b94197a0ef805904b7893) fixes incorrect data type on the initial brightness and dimming range max fields.

myrwon, 2025-03-03:
yes ....if you haven't, remember to change the protocol on the bulbs from Matter to Zigbee

bradley-rose, 2025-03-03:
I operate Zigbee2MQTT within Home Assistant as an add-on, so I wonder if I'm doing something wrong here...

I navigated to the root `zigbee2mqtt` directory, created a directory called `external_converters`, and placed the `aqara_tr_cct.js` file in there. I then rebooted the virtual machine and re-interviewed the bulb, but it doesn't look like Zigbee2MQTT did anything differently. Still showing as incompatible. 

I had a brief look at [Z2M's documentation for external converters](https://www.zigbee2mqtt.io/advanced/more/external_converters.html), and these instructions look a little different to the above.

I can just wait for April's Zigbee2MQTT release where these bulbs will likely be supported in, but if anyone here has exact details for how to make this Zigbee bulb compatible ahead of time, I'd appreciate that.

myrwon, 2025-03-04:
are you able to control the bulbs at all in Zigbee2MQTT? the error message should say "unsupported" 
I'm running HAos 2025.2.5 with a SLZB-06P7 in Zigbee coordinator mode.
what do your Logs show? 
here's my screenshots...the directory and Z2M status

[image]

jonny190, 2025-03-28:
I cant even get my T2 in to Zigbee mode

Flumeded, 2025-03-29:
@jonny190 
> I cant even get my T2 in to Zigbee mode

You need to download the app -> add device -> Select lamp, and you can choose "This device also supports the Zigbee protocol...", click on it, app will install Zigbee firmware

KennoPa, 2025-04-02:
> > > I'm really new to Zigbee2MQTT and don't really understand what I need to do to make them compatible. Right now they are paired with Home Assistant via Matter over Thread.
> > 
> > 
> > I dont use HA, matter or thread. In order to pair them with z2m and get the id's you'll need to put them into pairing mode, and start the process of adding an accessory in the Aqara home app on your phone. The first screen you see after discovering the device will allow you to switch protocol to Zigbee - which will go through the firware upload process and then the device will enter Zigbee pairing mode at which time you can enable joining on z2m and it should be discovered.
> > From there you can go to the device card in z2m and from the dev console tab click generate_external_definition - this will show you the model id. You can also access the databases.db file in your z2m data directory and copy the corresponding entry (likely the last).
> > If you want to go back to matter protocol, repeat the above process in aqara home after entering pairing mode.
> > > Is there any other information you are interested in that I can easily find out?
> > 
> > 
> > Database entry for the GU10 RGB in particular as I dont have or need any of this model.
> 
> How can I add this device to my aqara app without any of their hubs? i want to use this bulb with z2m but i think i'm stuck at matter mode. i can add it to ha matter server

Aqara app uses Bluetooth to connect to the bulb. You may need to reset the bulb for it to connect. The app add wizard should guide through the process. I do not have aqara hub and have switched my T2 s several times between Matter and Zigbee, ended up using Z2M with custom converter.

vascozorrinho, 2025-04-02:
> Aqara app uses Bluetooth to connect to the bulb. You may need to reset the bulb for it to connect. The app add wizard should guide through the process. I do not have aqara hub and have switched my T2 s several times between Matter and Zigbee, ended up using Z2M with custom converter.

i managed to find that but it was super unintuitive. the official manual does not refer to the bluetooth pairing at all and after i found online that was a think i tried to manually pair the bulb with the magic bluetooth paring mode but the app couldn't find the bulb. only after a few tryes and me leaving the app opened doing nothing it magically found the bulb and i was able to select zigbee

KennoPa, 2025-04-02:
> i managed to find that but it was super unintuitive. the official manual does not refer to the bluetooth pairing at all and after i found online that was a think i tried to manually pair the bulb [...]

Yes it was having that mind of its own for me to :)

HASKADOG, 2025-05-05:
> ### TL;DR UPDATE
> 
> please give the [prototype external converter](https://gist.github.com/coldridj/527b05575f8b94197a0ef805904b7893) a go and report any issues or missing functionality
> 
>     * if unsure what to do with this file see [how to add an external converter](https://www.zigbee2mqtt.io/advanced/more/external_converters.html)
> 
> 
> --
> 
> I have some of these bulbs as well - there are [several variants](https://www.aqara.com/wp-content/uploads/2025/01/1464-600-10.png) depending on which market you are in (I am EU) and whether they support CCT or CCT + RGB
> 
> Mine report as `lumi.light.agl004` rather than `agl003` as yours do.
> 
> I had a go at extending the converter using some of the available functions from the lumi library and have the power on behaviour working and device temperature reporting in addition to state, brightness and cct.
> 
> Power measurement did not work with either of the existing metering functions from this library, and I am most interested in exposing the power on/off fade timeout. I can see relevant looking nodes in the database so will investigate these more thoroughly.
> 
> const { lumiModernExtend } = require('zigbee-herdsman-converters/lib/lumi');
> 
> const definition = {
>     zigbeeModel: ['lumi.light.agl004'],
>     model: 'T2 E27 CCT',
>     vendor: 'Aqara',
>     description: 'LED Bulb T2 CCT',
>     extend: [
>         lumiModernExtend.lumiLight({colorTemp: true, powerOutageMemory: false}),
>         lumiModernExtend.lumiZigbeeOTA(),
>         lumiModernExtend.lumiPowerOnBehavior(),
> };
> 
> module.exports = definition;
> 
> My database entry appears to contain substantially more than yours:
> 
> {"id":64,"type":"Router","ieeeAddr":"0x54ef441000efefcf","nwkAddr":18013,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.light.agl004","epList":[1,21],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[5,4,3,0,768,8,6,2820,1794,64704],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"61441":0,"modelId":"lumi.light.agl004","manufacturerName":"Aqara","powerSource":4,"zclVersion":3,"appVersion":27,"stackVersion":27,"hwVersion":1,"dateCode":"20250113"}},"lightingColorCtrl":{"attributes":{"245":4611445,"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370,"colorTemperature":290,"colorMode":2}},"haElectricalMeasurement":{"attributes":{"acPowerDivisor":10,"acPowerMultiplier":1}},"manuSpecificLumi":{"attributes":{"22":"56DD9AEEF565CEBA1B8788CBD72C5F1EEC93C749B04FD827BF0E6B8FFC3A7E1D","223":{"type":"Buffer","data":[19,33,0,0,20,33,0,0,22,33,0,0,23,33,0,0,12,33,0,0,24,33,115,4]},"229":{"type":"Buffer","data":[23,15,8,149,192,37,89,67,193,37,0,127,204,37,50,193,221,37,84,82,235,37,40,173,240,37,88,201,244,37,102,167,249,37,86]},"230":0,"247":{"type":"Buffer","data":[3,40,33,5,33,2,0,8,33,27,1,9,33,0,23,10,33,0,0,11,32,0,12,32,10,160,32,1,154,32,0,100,32,1,101,32,154,102,33,34,1,106,35,140,10,100,25,105,32,0,113,32,1]},"1301":1,"1302":100,"1303":1,"1320":0.800000011920929,"1324":0,"65522":{"type":"Buffer","data":[170,117,19,132,201,129,7,65,0,15,12,35,1,4,0,23,10,35,8,0,12,100,11,32,4]}}},"genLevelCtrl":{"attributes":{"245":4611453,"onTransitionTime":15,"offTransitionTime":15,"currentLevel":154}},"genOnOff":{"attributes":{"245":4611522,"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":0,"inClusterList":[12],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":0}}},"binds":[{"cluster":12,"type":"endpoint","deviceIeeeAddress":"0x00124b00215fff90","endpointID":1}],"configuredReportings":[{"cluster":12,"attrId":85,"minRepIntval":10,"maxRepIntval":3600,"repChange":5}],"meta":{}}},"appVersion":27,"stackVersion":27,"hwVersion":1,"dateCode":"20250113","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1739184043604}
> 
> z2m version 2.1.1

i tried it, but rgb control is not working

KennoPa, 2025-05-06:
Having the latest converter, my T2 cct rgb eu bulbs do not react on power on behaviour and will stay act as previous state when power off and on does not matter what value I choose

bgolinvaux, 2025-05-10:
I have just bought the EU version (` lumi.light.agl004` → `LED Bulb T2 (E27, CCT) [EU]`) and, thanks to [the gist](https://gist.github.com/coldridj/527b05575f8b94197a0ef805904b7893) and the instructions provided, I managed to control the temperature and the power on/off state.

Thank you very much!

I noticed @bradley-rose writing : "_I navigated to the root zigbee2mqtt directory, created a directory called `external_converters`, and placed the aqara_tr_cct.js file in there._" and then explaining it does not work.  Please note that the `external_converters` folder needs to be created in the **data** directory. I don't know where it is for your particular install, but in a Docker compose setup, the data folder is `/app/data` and that's where the `external_converters` folder needs to be.   This folder needs to sit next to the `configuration.yaml` file.

epf, 2025-05-12:
Disclaimer: I am fairly new to Home Assistant and have never had to do anything out of the ordinary with Z2M prior to tonight.

I'm running the Z2M `2.3.0-1` add-on in HA `2025.5.1` on HAOS `15.2`.

I have a `lumi.light.agl001` that I am attempting to use with Z2M. To get this far, I've done the following:

1. Installed the [Advanced SSH & Web Terminal](https://github.com/hassio-addons/addon-ssh) add-on, protection mode: disabled.
2. SSH'd into HA and run:
    ```bash
    mkdir /config/zigbee2mqtt/external_extensions
    curl --output-dir /config/zigbee2mqtt/external_extensions -O https://gist.githubusercontent.com/coldridj/527b05575f8b94197a0ef805904b7893/raw/4f68ce274dc7a1082c351554774588fce25ad939/aqara_t2_cct.js
    docker restart $(docker ps | grep zigbee2mqtt | cut -d" " -f 1)
    ```
The external converter fails to load and all I get in the logs are:
```
[2025-05-11 22:35:33] error:    z2m: Failed to load external extension 'aqara_t2_cct.js'. Check the code for syntax error and make sure it is up to date with the current Zigbee2MQTT version.
[2025-05-11 22:35:33] error:    z2m: Invalid external extension 'aqara_t2_cct.js' was ignored and renamed to prevent interference with Zigbee2MQTT.
```
Given the lack of useful information in those log lines, I'm not sure where to even look next. Any pointers would be appreciated!

bgolinvaux, 2025-05-12:
@epf   Maybe you could check that the contents of the `.js` file is the same at what you can see in your browser while visiting [the same link](https://gist.githubusercontent.com/coldridj/527b05575f8b94197a0ef805904b7893/raw/4f68ce274dc7a1082c351554774588fce25ad939/aqara_t2_cct.js)

Perhaps you'll find out that the actual file in your install instead contains "HTTP error blah when retrieving blah" or sth along those lines?  (it happened to me more than a few times, so...)

Apologies if this is obvious, but you should SSH and type:
```
cat /config/zigbee2mqtt/external_extensions/aqara_t2_cct.js
```

epf, 2025-05-12:
@bgolinvaux good thinking, but it's the same file as here in the gist. (Yes, I was surprised I didn't need to add `-L` to `curl` to get it to work!)

1. The raw file doesn't end in `\n`—I fixed that, same error.
2. I've copy-pasted the file into vim instead of fetching it with curl—same error.

<details>
<summary>Supporting evidence (long)</summary>

```
➜  ~ curl --output-dir /config/zigbee2mqtt/external_extensions -O https://gist.githubusercontent.com/coldridj/527b05575f8b94197a0ef805904b7893/raw/4f68ce274dc7a1082c351554774588fce25ad939/aqara_t2_cct.js
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  7300  100  7300    0     0  12394      0 --:--:-- --:--:-- --:--:-- 12393
➜  ~ echo >> /config/zigbee2mqtt/external_extensions/aqara_t2_cct.js
➜  ~ cat /config/zigbee2mqtt/external_extensions/aqara_t2_cct.js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;

const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const { lumiModernExtend, fromZigbee, manufacturerCode } = require('zigbee-herdsman-converters/lib/lumi');

const { DefinitionWithExtend } = require('zigbee-herdsman-converters/lib/types');

const lumiTransitionCurve = () =>
    modernExtend.numeric({
        name: 'transition_curve_curvature',
        label: 'Transition Curve Curvature',
        cluster: 'manuSpecificLumi',
        attribute: {ID: 0x0528, type: 0x39},
        description: 'Range 0.2>x>1: first fast then slow, 1: linear, 1>x>6: first slow then fast',
        entityCategory: 'config',
        zigbeeCommandOptions: {manufacturerCode},
        valueMin: 0.2,
        valueMax: 6,
        valueStep: 0.01,
    });

const lumiTransitionBrightness = () =>
    modernExtend.numeric({
        name: 'transition_initial_brightness',
        label: 'Initial Brightness',
        cluster: 'manuSpecificLumi',
        attribute: {ID: 0x052c, type: 0x20},
        description: 'After turning on the light, the light will brighten from the preset brightness',
        entityCategory: 'config',
        zigbeeCommandOptions: {manufacturerCode},
        unit: '%',
        valueMin: 0,
        valueMax: 50,
        valueStep: 1,
    });

const lumiLevelConfig = () => {
    let levelConfig = new exposes.Composite('level_config', 'level_config', exposes.access.ALL);
    levelConfig = levelConfig.withFeature(
        new exposes.Numeric('on_transition_time', exposes.access.ALL)
            .withLabel('ON transition time')
            .withDescription(
                'Represents the time taken to move the current level from the minimum level to the maximum level when an On command is received',
            )
            .withUnit('1/10th second')
            .withValueMax(600)
            .withValueStep(5),
    );
    levelConfig = levelConfig.withFeature(
        new exposes.Numeric('off_transition_time', exposes.access.ALL)
            .withLabel('OFF transition time')
[... 151 lines cut ...]
        model: 'LED Bulb T2 (GU10, CCT) [NA]',
        vendor: 'Aqara',
        description: 'LED Bulb with CCT adjustment',
        extend: [ lumiLightT2(false) ],
    },
];

module.exports = definitions;
```
</details>

And still no dice:

```
➜  ~ docker restart $(docker ps | grep zigbee2mqtt | cut -d" " -f 1)
➜  ~ docker logs $(docker ps | grep zigbee2mqtt | cut -d" " -f 1) | grep aqara_t2_cct
[07:43:34] INFO: Preparing to start...
[07:43:34] INFO: Socat not enabled
[07:43:34] INFO: Starting Zigbee2MQTT...
[2025-05-12 07:43:37] error: 	z2m: Failed to load external extension 'aqara_t2_cct.js'. Check the code for syntax error and make sure it is up to date with the current Zigbee2MQTT version.
[2025-05-12 07:43:37] error: 	z2m: Invalid external extension 'aqara_t2_cct.js' was ignored and renamed to prevent interference with Zigbee2MQTT.
```
If we logged a syntax error then I would have somewhere to look.

Are you running the same version of Z2M that I am?

epf, 2025-05-12:
There were changes to the extension class made for z2m 2.3.0. I had ChatGPT (o3) update @coldridj's prototype to fit the new class structure. It can be found [here](https://gist.github.com/epf/5d9ad479e667f32f0d855706ba395ada). It's working for me so far.

bmurphy1976, 2025-05-16:
I have a model that claims to be `lumi.light.agl001`.  I tried @epf's gist and it's still refusing to recognize the bulb.  I can see the z2m successfully loaded the converter:

```
[2025-05-16 18:48:03] info:     z2m: Loaded external converter 'aqara_t2_bulb_family.js'.
```

And I'm currently running z2m v2.3.0.  I don't see anything in the logs that hints at the root problem.

bmurphy1976, 2025-05-16:
I switched to the converter posted by @coldridj and that's working for me.  🤷

Eld0h, 2025-05-31:
> i managed to find that but it was super unintuitive. the official manual does not refer to the bluetooth pairing at all and after i found online that was a think i tried to manually pair the bulb [...]

Can you eleborate? I have exact samen issue. I have no aqara hub and I am unable to pair the bulb. How exactly did you manage to do the pairing via Bluetooth? Do you need a thread border router to make the bulbs switch to zigbee?

vascozorrinho, 2025-05-31:
> Can you eleborate? I have exact samen issue. I have no aqara hub and I am unable to pair the bulb. How exactly did you manage to do the pairing via Bluetooth? Do you need a thread border router to [...]

Put the bulb in pairing mode.
On your aqara app press add button but don't choose any model. Wait for the bulb to be detected and the icon will appear on the top left corner.
Select it and you'll see a little link saying something like "this device can also operate in zigbee mode"
After that follow the wizard

Eld0h, 2025-06-01:
> Put the bulb in pairing mode. On your aqara app press add button but don't choose any model. Wait for the bulb to be detected and the icon will appear on the top left corner. Select it and you'll [...]

I tried but not succeeded. I waited 15minutes when bulb is in pairing mode but the icon never popups up. Its frustrating, i even tried the 5-10 on off switching to fore a reset and still not. I am sure the bulb is in pairing mode because lamp dimms 2 times when I set it on. I only have the option to scan the qr code of to manually select but I dont have an aqara hub

Eld0h, 2025-06-01:
> > I tried but not succeeded. I waited 15minutes when bulb is in pairing mode but the icon never popups up. Its frustrating, i even tried the 5-10 on off switching to fore a reset and still not. I am sure the bulb is in pairing mode because lamp dimms 2 times when I set it on. I only have the option to scan the qr code of to manually select but I dont have an aqara hub
> 
> [@Eld0h](https://github.com/Eld0h) I don't quite follow what you are saying What icon are you waiting for and do you have the Aqara app on your phone? I had to install the app, connect to the bulb via Bluetooth with the phone app, switch it from thread to Zigbee, and then I was able to connect the bulb to my Zigbee network. No hub required.

Yes I use Aqara home app on iOS. Well they advised me not to use qr code or manually select the bulb from the list but wait until the device pops up in the app when I am in the ‘add accessorry’ screen. I waited 15 minutes but nothing. The bulb is in pairing mode. I have no idea how to force the Bluetooth connection. I have the RGB GU10 T2 bulb

vascozorrinho, 2025-06-17:
Any idea when these devices will be officially supported?

carldanley, 2025-06-26:
I've tested the gist (v4) and things seem to be working pretty well with my `
lumi.light.agl001`. However, when I use the `Exposes` tab to select a color temperature and then select a color, I receive the attached error.

[image]

I *think* everything else seems to be working correctly. The icon obviously needs to be updated. I'm new to this code base but if there's anything I can do to push this along, I'm happy to help provide feedback.

I just purchased 15 bulbs so I'm invested in getting this patched =P

vascozorrinho, 2025-06-26:
I have this error on boot with the RGB version

> [2025-06-26 09:00:36] error: 	z2m: Failed to configure 'Light/Bedroom', attempt 1 (Error: ZCL command 0x54ef4410012e9621/1 haElectricalMeasurement.read(["acCurrentDivisor","acCurrentMultiplier"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@4.1.0/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:363:28)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@4.1.0/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:987:26)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Endpoint.read (/app/node_modules/.pnpm/zigbee-herdsman@4.1.0/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:459:29)
    at result.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.53.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:2102:37)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.53.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:21)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
[2025-06-26 09:00:49] info: 	z2m: Configuring 'Light/Bedroom'

Apart from that works without any problem

leonlegant, 2025-07-04:
Hello,

I've spent all morning trying to get the external converter to work. I am new to this, but I have gone over everything written above and got ChatGPT to try to help, all with no success. 

All bulbs continue to show as unsupported no matter what I do.

Can someone please help me troubleshoot and figure out where I'm going wrong?

The bulbs I'm using are:

- Aqara Bulb T2 (E27, RG  CCT) model version lumi.light.agl003
- Aqara Bulb T2 (GU10, RG  CCT) modle version lumi.light.agl005

I have tried both current versions of Zigbee2mqtt and Zigbee2mqtt Edge as suggested by ChatGPT.

I have tried many many adjustments based on the above and ChatGPT and I seem very close, but cannot get the bulb to show as supported or to get the additional features to display. 

I have added the following to zigbee2mqtt/configuration.yaml

```
advanced:
   external_converters:
    - external/aqara_t2_cct.js

``` 

I have added uploaded the following file to zigbee2mqtt/external/aqara_t2_cct.js

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;

const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const { lumiModernExtend, fromZigbee, manufacturerCode } = require('zigbee-herdsman-converters/lib/lumi');

const { DefinitionWithExtend } = require('zigbee-herdsman-converters/lib/types');

const lumiTransitionCurve = () => 
    modernExtend.numeric({
        name: 'transition_curve_curvature',
        label: 'Transition Curve Curvature',
        cluster: 'manuSpecificLumi',
        attribute: {ID: 0x0528, type: 0x39},
        description: 'Range 0.2>x>1: first fast then slow, 1: linear, 1>x>6: first slow then fast',
        entityCategory: 'config',
        zigbeeCommandOptions: {manufacturerCode},
        valueMin: 0.2,
        valueMax: 6,
        valueStep: 0.01,
    });

const lumiTransitionBrightness = () => 
    modernExtend.numeric({
        name: 'transition_initial_brightness',
        label: 'Initial Brightness',
        cluster: 'manuSpecificLumi',
        attribute: {ID: 0x052c, type: 0x20},
        description: 'After turning on the light, the light will brighten from the preset brightness',
        entityCategory: 'config',
        zigbeeCommandOptions: {manufacturerCode},
        unit: '%',
        valueMin: 0,
        valueMax: 50,
        valueStep: 1,
    });

const lumiLevelConfig = () => {
[... 149 lines cut ...]
    {
        zigbeeModel: ['lumi.light.agl006'],
        model: 'LED Bulb T2 (GU10, CCT) [EU]',
        vendor: 'Aqara',
        description: 'LED Bulb with and CCT adjustment',
        extend: [ lumiLightT2(false) ],
    },
    {
        zigbeeModel: ['lumi.light.agl007'],
        model: 'LED Bulb T2 (GU10, CCT RGB) [NA]',
        vendor: 'Aqara',
        description: 'LED Bulb with RGB and CCT adjustment',
        extend: [ lumiLightT2(true) ],
    },
    {
        zigbeeModel: ['lumi.light.agl008'],
        model: 'LED Bulb T2 (GU10, CCT) [NA]',
        vendor: 'Aqara',
        description: 'LED Bulb with CCT adjustment',
        extend: [ lumiLightT2(false) ],
    },
];

module.exports = definitions;
```

leonlegant, 2025-07-04:
I got it figured out :)

I removed the external_converters reference from the zigbee2mqtt/configuration.yaml

I renamed the folder "external" to "external_converters" and uploaded the aqara_t2_cct.js there.

After restart and repair, everything seems to work now.

SalGnt, 2025-07-04:
Is there something we can do to move forward with this integration?

K-SDaniel, 2025-07-24:
> Rájöttem :)
> 
> Eltávolítottam az external_converters hivatkozást a zigbee2mqtt/configuration.yaml fájlból.
> 
> Átneveztem az "external" mappát "external_converters"-re, és feltöltöttem oda az aqara_t2_cct.js fájlt.
> 
> Újraindítás és javítás után úgy tűnik, most már minden működik.

I did everything the same way, but it doesn't work for me :( What could I be doing wrong?

jayceegee87, 2025-08-05:
> > > I got it figured out :)
> > > I removed the external_converters reference from the zigbee2mqtt/configuration.yaml
> > > I renamed the folder "external" to "external_converters" and uploaded the aqara_t2_cct.js there.
> > > After restart and repair, everything seems to work now.
> > 
> > 
> > So to be clear the only thing you did was add that file and reboot and everything worked? Is there nothing else you have to do?
> 
> I cannot remember exactly what happened.
> 
> I had read that I needed to add something to the zigbee2mqtt/configuration.yaml, but that did not work.
> 
> I simply added the file to the "external_converters" folder and restarted. Then it worked for me once /I removed the entry I added to the zigbee2mqtt/configuration.yaml

I similarly was unable to get Z2M to recognize the converter file when I attempted to do so 6-8 weeks ago and decided to give this a try again. I just replicated his steps and the devices were recognized immediately after restarting Z2M. 

I don't know if this was just my error or if there is an error in the instructions, but I inadvertently uploaded the aqara_t2_cct.js file to zigbee2mqtt/_data_/external_converters when it should be **zigbee2mqtt/external_converters**. Hopefully this helps folks figure it out.

SeveranExp, 2025-08-06:
Curiosity from my part - are you guys also seeing some weirdness in the UI buttons for the gu10 T2?

jayceegee87, 2025-08-08:
> Curiosity from my part - are you guys also seeing some weirdness in the UI buttons for the gu10 T2?

Not to be rude, but could you kindly provide more detail regarding the weirdness you've identified? We don't know if the weird UI is simply a device configuration you're unfamiliar with, an issue with the converter that folks have been testing, or simply because z2m is defaulting to the generic light definition since it doesn't currently have a supported herdsman definition. Is it related to the color temperature issue you wrote about several months ago? 

It appears that the most recent version of the converter file is working for most people across the various iterations of the T2 line, so the more detail you provide the faster we can wrap all of this up and submit for review/approval.

To answer your question from back in June: You should provide relevant logs/screenshots, a description of the issue based on your observations, a description of the expected behavior, a simple set of instructions for how to reproduce said issue, the version of z2m you are running as well as the model and version of your coordinator, etc. 

I find it helpful to start (but not submit/open) a new bug report in the "known issues" tab so I can use the fields/structure as a guide for what one should ideally provide when raising a potential issue. It may be a good place for you to start.

SeveranExp, 2025-08-08:
@jayceegee87 i didn’t find you rude at all, but the lack of acknowledgment to my question had to raise the topic again. There are no errors in the logs, all I have is the external component in the folder, and overall the ui does work, except for that temperature thing. I’ve updated zigbee2mqtt at least twice since then.

Im not sharing more because I don’t know what to share, and that’s why im here :). And in June I shared screenshots regarding what I’m seeing:
https://github.com/Koenkk/zigbee2mqtt/issues/26182#issuecomment-2927413144

I’m also replaced the external component at least once since June, and I see the same thing :/

zeratul021, 2025-08-12:
@Koenkk what is typically needed to merge the new device to upstream? 
Can we help with some extra testing/logs?

Koenkk, 2025-08-13:
I added initial support.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `T2.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

zeratul021, 2025-08-21:
@Koenkk I don't think so. I know my model is **agl005**, and based on this table I think the E27 socket is **agl003**:
https://opendoc.aqara.com/en/docs/SDKDevelopment/AndroidDevelopmentGuide/Equipmentcontrol/DeviceControlSDKSupportedDeviceList.html

vascozorrinho, 2025-08-22:
I have 2 bulbs:

1. E27 CCT reports as lumi.light.agl004
2. E27 RGB reports as lumi.light.agl003

when i boot the Z2M addon the RGB gives me an error:

`[2025-08-22 09:09:00] error: 	z2m: Failed to configure 'Light/Bedroom', attempt 1 (Error: ZCL command 0x54ef4410012e9621/1 haElectricalMeasurement.read(["acCurrentDivisor","acCurrentMultiplier"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:372:28)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:1005:26)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Endpoint.read (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:481:29)
    at result.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:2065:37)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:21)
    at Configure.configure (/app/lib/extension/configure.ts:123:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:17))`

but it works fine after that

SeveranExp, 2025-08-24:
Oh I just noticed that I may have a different model.
Zigbee Model
lumi.light.agl006
Model
Model
[LED Bulb T2 (GU10, CCT) [EU]]

ds-sebastian, 2025-08-25:
Tested the latest dev build with the North-American T2 E26 RGB + CCT bulb.
- Zigbee model reported: lumi.light.agl001
- Zigbee2MQTT / herdsman versions: 2.6.0-dev / 25.8.0 / 6.0.1

On dev the bulb still pairs as “unsupported”:
```
Device '...' with Zigbee model 'lumi.light.agl001' and manufacturer name 'Aqara' is NOT supported...
```

Using the external converter gist above everything seems to work great. It seems the latest dev only has support for the `lumi.light.agl003`?

TacticalApples, 2025-08-28:
Has anyone gotten the below T2 bulb model to show as supported? I believe that I'm on the latest dev build and its still showing "unsupported".

- Model: North-American T2 E26 CCT Bulb 
- Zigbee model reported: lumi.light.agl002
- Z2M Version: 2.6.0-dev / 25.10.0 / 6.01

Thanks!

vascozorrinho, 2025-09-02:
the changelog for 2.6.1 only specifies lumi.light.agl002. why?

absent42, 2025-09-02:
> The ones supported in 2.6.1 are: `zigbeeModel: ["lumi.light.agl003", "lumi.light.agl005", "lumi.light.agl006", "lumi.light.agl001", "lumi.light.agl002"],`
> 
> [@Cave-Johnson](https://github.com/Cave-Johnson) agl004 is this an E26 or GU10?

My agl003 (RGB) and agl004 (CCT) are both E26.

absent42, 2025-09-02:
> So the `agl004` doesn't support changing colours?

Just tunable whites, from the specs: Color Temperature	2700K to 6500K

Cave-Johnson, 2025-09-02:
> > [@Cave-Johnson](https://github.com/Cave-Johnson) agl004 is this an E26 or GU10?

Mine were advertised as T2 E27

`Aqara Thread/Zigbee Smart LED Light Bulb T2 E27, Tunable White (2700K-6500K), 75W Equivalent`

Not sure if useful:
```
export default {
    zigbeeModel: ['lumi.light.agl004'],
    model: 'lumi.light.agl004',
    vendor: 'Aqara',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"21":21}}), m.light({"colorTemp":{"range":[153,370]}}), m.electricityMeter()],
    meta: {"multiEndpoint":true},
};
```

vascozorrinho, 2025-09-03:
my e27 CCT non RGB reports as lumi.light.agl004 and it still reports as unsupported

neggles, 2025-10-26:
So these don't actually use the power outage memory setting the other Aqara bulbs use, it seems. if I reconfigure the GU10 bulb with this converter:

```js
import { lumiModernExtend } from "zigbee-herdsman-converters/lib/lumi";

const { lumiLight, lumiZigbeeOTA } = lumiModernExtend;

export default {
    zigbeeModel: ["lumi.light.agl005"],
    model: "T2_GU10",
    vendor: "Aqara",
    description: "GU10 led bulb",
    extend: [
        lumiLight({
            colorTemp: true,
            color: true,
            colorTempRange: [111, 500],
            levelConfig: { features: ["on_level"] },
        }),
        lumiZigbeeOTA(),
    ],
};

```
I get the ability to set the on_level to a specific value or to `previous` which seems to make it remember the last turned-on value, so that's effectively power outage memory.

Weirdly the unit reports correctly for the other level config features - `on_off_transition_time` / `on_transition_time` / `off_transition_time` - but they don't seem to actually impact the behaviour of the thing. And setting the on_level to "previous" doesn't seem to stick, though it does seem to work as intended - just z2m then reports the currently-set brightness as the `on_level`. Weird.

It also sends payloads like these every 13 seconds:
```
{"65522":{"data":[170,117,19,132,24,50,7,65,0,15,12,35,1,2,0,5,10,35,8,0,12,100,11,32,5],"type":"Buffer"}},
{"65522":{"data":[170,117,19,132,25,49,7,65,0,15,12,35,1,2,0,5,10,35,8,0,12,100,11,32,5],"type":"Buffer"}},
{"65522":{"data":[170,117,19,132,26,48,7,65,0,15,12,35,1,2,0,5,10,35,8,0,12,100,11,32,5],"type":"Buffer"}},
{"65522":{"data":[170,117,19,132,27,47,7,65,0,15,12,35,1,2,0,5,10,35,8,0,12,100,11,32,5],"type":"Buffer"}},
{"65522":{"data":[170,117,19,132,28,46,7,65,0,15,12,35,1,2,0,5,10,35,8,0,12,100,11,32,5],"type":"Buffer"}},
{"65522":{"data":[170,117,19,132,29,45,7,65,0,15,12,35,1,2,0,5,10,35,8,0,12,100,11,32,5],"type":"Buffer"}},
{"65522":{"data":[170,117,19,132,30,44,7,65,0,15,12,35,1,2,0,5,10,35,8,0,12,100,11,32,5],"type":"Buffer"}},
{"65522":{"data":[170,117,19,132,31,43,7,65,0,15,12,35,1,2,0,5,10,35,8,0,12,100,11,32,5],"type":"Buffer"}},
{"65522":{"data":[170,117,19,132,32,42,7,65,0,15,12,35,1,2,0,5,10,35,8,0,12,100,11,32,5],"type":"Buffer"}},
{"65522":{"data":[170,117,19,132,33,41,7,65,0,15,12,35,1,2,0,5,10,35,8,0,12,100,11,32,5],"type":"Buffer"}},
{"65522":{"data":[170,117,19,132,34,40,7,65,0,15,12,35,1,2,0,5,10,35,8,0,12,100,11,32,5],"type":"Buffer"}},
{"65522":{"data":[170,117,19,132,35,39,7,65,0,15,12,35,1,2,0,5,10,35,8,0,12,100,11,32,5],"type":"Buffer"}},
{"65522":{"data":[170,117,19,132,36,38,7,65,0,15,12,35,1,2,0,5,10,35,8,0,12,100,11,32,5],"type":"Buffer"}},
{"65522":{"data":[170,117,19,132,37,37,7,65,0,15,12,35,1,2,0,5,10,35,8,0,12,100,11,32,5],"type":"Buffer"}},
```
and these every 3 minutes:
```
{"65522":{"data":[170,117,41,132,23,29,7,65,0,37,1,35,0,0,0,0,2,33,0,0,3,35,0,0,0,0,4,35,0,0,0,0,5,35,0,3,54,205,10,35,8,0,12,100,11,32,5],"type":"Buffer"}},
{"65522":{"data":[170,117,41,132,38,14,7,65,0,37,1,35,0,0,0,0,2,33,0,0,3,35,0,0,0,0,4,35,0,0,0,0,5,35,0,3,54,205,10,35,8,0,12,100,11,32,5],"type":"Buffer"}},
```
which i can't seem to work out the meaning of / don't seem to match up with anything else existing in the code. But it's a little hard to tell because I'm not good with TS/JS and there aren't much in the way of docs/comments. There should be an energy meter in this thing, it exposes it when in Matter mode, but these repeated messages don't seem to be that, so idk what it's doing really. Might be to do with effects modes?

@Koenkk apologies for the ping, but if there's anything I can grab for you to dig into this a little and work out what it's doing, please let me know - just having correct behaviour for the power on brightness/memory would be nice.

Bridge state dump: [state.json.zip](https://github.com/user-attachments/files/23148930/state.json.zip)

[9 more replies]