Koenkk/zigbee2mqtt #26185: RB-SRAIN01 not expose entity after upgrade Zigbee2MQTT

snclab, 2025-02-04:
### What happened?

RB-SRAIN01 not expose entity after upgrade Zigbee2MQTT

https://www.zigbee2mqtt.io/devices/RB-SRAIN01.html

### What did you expect to happen?

Hi,

after update Z to version 2.1.0-1 RB-SRAIN01 not expose entity
expose only battery % and rain but alwais in true value

I tried rebooting, removing the device and adding it again

https://www.zigbee2mqtt.io/devices/RB-SRAIN01.html

Thanks

### Zigbee2MQTT version

2.1.0-1

### Adapter firmware version

7.4.3 [GA]

### Adapter

EmberZNet

### Setup

HomeAssistant add-on

crazyserver, 2025-02-04:
Same here, I have battery and the binary sensor but all the other sensors: Illuminance, intensity, unknown

crazyserver, 2025-02-04:
BTW: Duplicated from https://github.com/Koenkk/zigbee2mqtt/issues/26124