Koenkk/zigbee2mqtt #26186: Device shows not supported

ochstobi, 2025-02-04:
### What happened?

I try to connect 50139 SH ZB Paulmann Home Universal-Schaltaktor Weiß.
It got supported in v2.1.0 but I still can't pair it. It seems that the device itself does not know that it is a Paulmann device, because the shown `Manufacturer name` is: `_TZ3210_hjxqqofs�`
The logs:

```
info 2025-02-04 21:32:24zh:controller: Interview for '0xa4c1381b33ed1195' started
info 2025-02-04 21:32:24z2m: Starting interview of '0xa4c1381b33ed1195'
info 2025-02-04 21:32:24z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b33ed1195","ieee_address":"0xa4c1381b33ed1195","status":"started"},"type":"device_interview"}'
info 2025-02-04 21:32:24z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b33ed1195","ieee_address":"0xa4c1381b33ed1195"},"type":"device_announce"}'
info 2025-02-04 21:32:26zh:controller: Succesfully interviewed '0xa4c1381b33ed1195'
info 2025-02-04 21:32:26z2m: Successfully interviewed '0xa4c1381b33ed1195', device has successfully been paired
warning 2025-02-04 21:32:26z2m: Device '0xa4c1381b33ed1195' with Zigbee model 'TS000F' and manufacturer name '_TZ3210_hjxqqofs�' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info 2025-02-04 21:32:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b33ed1195","ieee_address":"0xa4c1381b33ed1195","status":"successful","supported":false},"type":"device_interview"}'
```

### What did you expect to happen?

The device gets paired successfully.

### How to reproduce it (minimal and precise)

Try to pair the device.

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

20200925

### Adapter

zzh!

### Setup

Plain on Docker Container

### Debug log

[log.log](https://github.com/user-attachments/files/18663523/log.log)

chris-1243, 2025-02-05:
Your device is well supported 

https://github.com/Koenkk/zigbee-herdsman-converters/blob/b91e0a8e60be1850f9d6bf10cbf47c23f10e0dcc/src/devices/paulmann.ts#L294

As far as I can see in your log, there is a special character which should not be present. I would try to `force remove` the device, restart Z2M and try to pair your device again. Please provide the debug log

ochstobi, 2025-02-05:
There is a fix now: https://github.com/Koenkk/zigbee2mqtt/issues/26158#issuecomment-2637922242

Thanks for your help

ochstobi, 2025-02-06:
I tried to pair now with the latest dev branch. The device itself get recognized now. But somehow it says there are no active endpoints.

```
info 2025-02-06 09:25:41zh:controller: Interview for '0xa4c1381b33ed1195' started
info 2025-02-06 09:25:41z2m: Device '0xa4c1381b33ed1195' joined
info 2025-02-06 09:25:41z2m: Starting interview of '0xa4c1381b33ed1195'
info 2025-02-06 09:25:41z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b33ed1195","ieee_address":"0xa4c1381b33ed1195"},"type":"device_joined"}'
info 2025-02-06 09:25:41z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b33ed1195","ieee_address":"0xa4c1381b33ed1195","status":"started"},"type":"device_interview"}'
info 2025-02-06 09:25:41z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b33ed1195","ieee_address":"0xa4c1381b33ed1195"},"type":"device_announce"}'
error 2025-02-06 09:26:01zh:controller: Interview failed for '0xa4c1381b33ed1195 with error 'Error: Interview failed because can not get active endpoints ('0xa4c1381b33ed1195')'
error 2025-02-06 09:26:01z2m: Failed to interview '0xa4c1381b33ed1195', device has not successfully been paired
info 2025-02-06 09:26:01z2m:mqtt: MQTT publish: topic 'homeassistant/switch/0xa4c1381b33ed1195/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0xa4c1381b33ed1195/set","device":{"hw_version":1,"identifiers":["zigbee2mqtt_0xa4c1381b33ed1195"],"manufacturer":"Paulmann","model":"Universal-switch white","model_id":"501.39","name":"0xa4c1381b33ed1195","via_device":"zigbee2mqtt_bridge_0x00124b0021600458"},"name":null,"object_id":"0xa4c1381b33ed1195","origin":{"name":"Zigbee2MQTT","sw":"2.1.0-dev","url":"https://www.zigbee2mqtt.io"},"payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0xa4c1381b33ed1195","unique_id":"0xa4c1381b33ed1195_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
info 2025-02-06 09:26:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1381b33ed1195","ieee_address":"0xa4c1381b33ed1195","status":"failed"},"type":"device_interview"}'
info 2025-02-06 09:26:01z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c1381b33ed1195/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":1,"identifiers":["zigbee2mqtt_0xa4c1381b33ed1195"],"manufacturer":"Paulmann","model":"Universal-switch white","model_id":"501.39","name":"0xa4c1381b33ed1195","via_device":"zigbee2mqtt_bridge_0x00124b0021600458"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"0xa4c1381b33ed1195_linkquality","origin":{"name":"Zigbee2MQTT","sw":"2.1.0-dev","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c1381b33ed1195","unique_id":"0xa4c1381b33ed1195_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
```

[log.log](https://github.com/user-attachments/files/18686526/log.log)

ochstobi, 2025-02-06:
A new interview through the frontend was successful and its working now.

closed 2025-02-06 by ochstobi