Koenkk/zigbee2mqtt #26190: Regularly losing adapter after update to 2.1.0

Ten0, 2025-02-05:
### What happened?

Since update to 2.1.0 adapter regularly crashes. It is necessary to unplug it or reboot the PI (first time unplug was enough, second time I had to reboot).

Logs:
```
[2025-01-28 20:20:56] error: 	z2m: Publish 'set' 'color_temp' to 'Couloir entree Lamp' failed: 'Error: ZCL command 0xa4c13818716dce3c/1 lightingColorCtrl.moveToColorTemp({"colortemp":454,"transtime":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":38063,"apsFrame":{"profileId":260,"clusterId":768,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":226},"zclSequence":175,"commandIdentifier":11} timed out after 10000ms)'
[2025-01-28 23:23:36] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2025-01-28 23:23:36] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2025-01-28 23:23:36] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2025-01-28 23:23:36] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2025-01-28 23:23:37] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
```

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

7.4.1 [GA]

### Adapter

SLZB-07

### Setup

Ha addon

Tribalwarrior, 2025-02-05:
Same [issue](https://github.com/Koenkk/zigbee2mqtt/issues/26147). Could you please send device info from your adapter?

chris-1243, 2025-02-05:
I would try to flash firmware 7.4.5 or 8.0.2 (8.0.0 and 8.0.1 are not working correctly). It might solve your issue

Tribalwarrior, 2025-02-05:
> I would try to flash firmware 7.4.5 or 8.0.2 (8.0.0 and 8.0.1 are not working correctly). It might solve your issue

In my case it doesn't help

Ten0, 2025-02-05:
> Could you please send device info from your adapter?

I'm not sure what you mean. As specified above it's an SMLIGHT SLZB-07 adapter. Is there other information I should extract besides that and "Adapter firmware version"?

Tribalwarrior, 2025-02-05:
> I'm not sure what you mean.

Go to Settings>System>Hardware>All hardware 
Scroll down to your dongle, and paste "Attributes:" here

Ten0, 2025-02-06:
Device info:
```
hardware
Search hardware
3-1
autofs
card0
card1
cec0
controlC0
controlC1
cpu_dma_latency
device-mapper
ecryptfs
event0
full
fuse
gpiochip0
gpiochip1
gpiochip2
gpiochip3
gpiochip4
gpiochip5
hw_random
i2c-0
i2c-1
i2c-3
i2c-4
iio:device0
kmsg
kvm
loop-control
media0
media1
mem
mice
mmcblk0
/dev/disk/by-id/mmc-D9D16G_0x093a59a0
mmcblk0boot0
mmcblk0boot1
mmcblk0p1
/dev/disk/by-id/mmc-D9D16G_0x093a59a0-part1
mmcblk0p2
/dev/disk/by-id/mmc-D9D16G_0x093a59a0-part2
mmcblk0p3
/dev/disk/by-id/mmc-D9D16G_0x093a59a0-part3
mmcblk0p4
/dev/disk/by-id/mmc-D9D16G_0x093a59a0-part4
mmcblk0p5
/dev/disk/by-id/mmc-D9D16G_0x093a59a0-part5
mmcblk0p6
/dev/disk/by-id/mmc-D9D16G_0x093a59a0-part6
mmcblk0p7
/dev/disk/by-id/mmc-D9D16G_0x093a59a0-part7
mmcblk0p8
/dev/disk/by-id/mmc-D9D16G_0x093a59a0-part8
mmcblk0rpmb
null
pcmC0D0c
pcmC0D0p
pcmC1D0c
pcmC1D0p
port
psaux
random
renderD128
reserved
rfkill
rtc0
snapshot
system
timer
ttyS0
ttyS1
ttyS2
ttyS3
ttyS4
ttyS5
ttyS6
ttyS7
ttyUSB0
/dev/serial/by-id/usb-SMLIGHT_SMLIGHT_SLZB-07_32f3dc09a177ed119ff1f6fafdf7b791-if00-port0
Subsystem:
tty
Device path:
/dev/ttyUSB0
ID:
/dev/serial/by-id/usb-SMLIGHT_SMLIGHT_SLZB-07_32f3dc09a177ed119ff1f6fafdf7b791-if00-port0
Attributes:
DEVLINKS: >-
  /dev/serial/by-id/usb-SMLIGHT_SMLIGHT_SLZB-07_32f3dc09a177ed119ff1f6fafdf7b791-if00-port0
  /dev/serial/by-path/platform-xhci-hcd.1.auto-usb-0:1:1.0-port0
  /dev/serial/by-path/platform-xhci-hcd.1.auto-usbv2-0:1:1.0-port0
DEVNAME: /dev/ttyUSB0
DEVPATH: >-
  /devices/platform/usb@fe900000/fe900000.usb/xhci-hcd.1.auto/usb3/3-1/3-1:1.0/ttyUSB0/tty/ttyUSB0
ID_BUS: usb
ID_MODEL: SMLIGHT_SLZB-07
ID_MODEL_ENC: SMLIGHT\x20SLZB-07
ID_MODEL_ID: ea60
ID_PATH: platform-xhci-hcd.1.auto-usb-0:1:1.0
ID_PATH_TAG: platform-xhci-hcd_1_auto-usb-0_1_1_0
ID_PATH_WITH_USB_REVISION: platform-xhci-hcd.1.auto-usbv2-0:1:1.0
ID_REVISION: '0100'
ID_SERIAL: SMLIGHT_SMLIGHT_SLZB-07_32f3dc09a177ed119ff1f6fafdf7b791
ID_SERIAL_SHORT: 32f3dc09a177ed119ff1f6fafdf7b791
ID_TYPE: generic
ID_USB_DRIVER: cp210x
ID_USB_INTERFACES: ':ff0000:'
ID_USB_INTERFACE_NUM: '00'
ID_USB_MODEL: SMLIGHT_SLZB-07
ID_USB_MODEL_ENC: SMLIGHT\x20SLZB-07
ID_USB_MODEL_ID: ea60
ID_USB_REVISION: '0100'
ID_USB_SERIAL: SMLIGHT_SMLIGHT_SLZB-07_32f3dc09a177ed119ff1f6fafdf7b791
ID_USB_SERIAL_SHORT: 32f3dc09a177ed119ff1f6fafdf7b791
ID_USB_TYPE: generic
ID_USB_VENDOR: SMLIGHT
ID_USB_VENDOR_ENC: SMLIGHT
ID_USB_VENDOR_ID: 10c4
ID_VENDOR: SMLIGHT
ID_VENDOR_ENC: SMLIGHT
ID_VENDOR_ID: 10c4
MAJOR: '188'
MINOR: '0'
SUBSYSTEM: tty
TAGS: ':systemd:'
USEC_INITIALIZED: '8854442'
```

Tribalwarrior, 2025-02-06:
ID_USB_DRIVER: cp210x

It looks like the problem is in this driver. I ask the developers to pay attention to this. @Koenkk

Ten0, 2025-02-06:
Aah update, I realize now that the first occurrence of this issue was just before I applied 2.1.0 update, so it may in fact be unrelated to the update.
Sorry for the misleading information :(

gtalusan, 2025-03-27:
I experienced this today when upgrading my zigbee2mqtt docker image to latest.  I had to rollback to tag 2.0.0 and the crashing disappeared.

I'm running latest coordinator firmware on an slzb-06m.

MacDefender, 2025-04-25:
Have the same problems here. A few weeks ago I noticed  that I get a lot of these errors:
zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
zh:ember: Adapter fatal error: HOST_FATAL_ERROR
z2m: Adapter disconnected, stopping

As I just added new devices I thought my Somoff-E dongle might be too old and bought a new SLZB-06Mg24 with FW updated to the latest version ( zigbee 20250212 and core v2.8.5.dev7 ) but still get the same errors. 

Every time I try OTA or often when I activate pairing the adapter (or better the ember driver in z2m) stops working as the SLZB kan be reached normal using its webinterface and show no error other than z2m disconnected.

closed 2025-07-03 by github-actions[bot]