Koenkk/zigbee2mqtt #26192: Z2M latency and lag issues after 2.1.0 update

niw87, 2025-02-05:
### What happened?

Hi there, not sure if this has any thing to do with the recent update but it seems thst my z2m often has issue of loading up the z2m dashboard / devices. Tried to generate the map and often failed and it took considerably much much longer time than previous experience.

Not sure what's happening since I didn't add any new devices nor setup recently save and except the few recent HA updates and z2m 2.1.0 updates.

Re-restarting z2m or not? I have been restarting z2m or even rebooting the entire system but the issues seem like not going away.

My devices are facing latency issues in response. Often delay for few secs and the User Experience just getting worst.

I notice that all infos at the "about" page in z2m shows "unknown" when I scrolled through today. Restarted z2m and it shows up my coodinator's info again. 

I have no idea what to share but My hardware are as follows:-
HA Green
Core2025.1.4
Supervisor2024.12.3
Operating System14.2
Frontend20250109.2

Zigbee2MQTT version2.1.0 commit: unknown
Coordinator typezStack3x0
Coordinator revision20240710
Coordinator IEEE Address0x00124b002f89a0bc
Frontend version0.9.4
zigbee-herdsman-converters version21.27.1
zigbee-herdsman version3.2.5

### What did you expect to happen?

Latency minimise? Less buggy or better response?

### How to reproduce it (minimal and precise)

Not sure how to reproduce but it is happening now and then.

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

Coordinator revision20240710

### Adapter

Coordinator typezStack3x0

### Setup

Home Assistant Green

Gecko3600, 2025-02-06:
Seems like that youve got the same issue as me: #25945

niw87, 2025-02-07:
> Seems like that youve got the same issue as me: [#25945](https://github.com/Koenkk/zigbee2mqtt/issues/25945)

Looking forward to get update fix. The delay etc kinda frustrating

closed 2025-04-16 by github-actions[bot]