Koenkk/zigbee2mqtt #26194: After update to 2.1.0 Develco MOSZB-140 (frient motion sensor pro) do not update temperature and illuminance

diefraschw, 2025-02-05:
### What happened?

After the update of zigbee2mqtt to version 2.1.0 the values of illuminance and temperature show fixed fantasy values and do not update for 3 of 5 Develco MOSZB-140 (frient motion sensor pro) devices (2 devices work properly). The other values are updated correctly.

### What did you expect to happen?

all provided values of all devices are updated to the correct values and do not show fixed wrong values

### How to reproduce it (minimal and precise)

stable situation for some hours already

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

0x26690700

### Adapter

ConBee2/RaspBee2

### Setup

Z2M docker container on Pi 4b running Ubuntu

RobLeighton67, 2025-02-06:
This seems to be similar to the problem that has happened with other Frient devices where the configuration of attribute reporting fails with NW_TABLE_FULL and some attributes are then not reported - see problems with SMSZB-120 on here.

diefraschw, 2025-02-12:
I think I manged to solve the problem: I deleted the 3 not properly working devices completely from z2m, rebound them and gave them the same name as before. This helped. Now all values are reported again and the automations in Home Assistant work again.
Thus the problem can be closed.

closed 2025-02-12 by diefraschw