Koenkk/zigbee2mqtt #26199: external converter DOES NOT take precedence of standard converter

SSG1111, 2025-02-05:
### What happened?

external converter doesn't work anymore after the device support was added

### What did you expect to happen?

I expect that external converter will remain functional, as it says in documentation: "Note that external converters take precedence of standard converters"

https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#network-map

### How to reproduce it (minimal and precise)

Zigbee2MQTT >> Dev console >> generate_external_definition
Copy the code and save as external converter ( \external_coverters\name.js); add corresponding line to the configuration file.

Observe that device is still working with standard definition.

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

20220219

### Adapter

usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0

### Setup

rpi4-64, add-on

### Debug log

[log.log](https://github.com/user-attachments/files/18677112/log.log)

external converter \external_coverters\moesss.js:

```
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_xalsoe3m',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;

```

configuration.yaml:
```

homeassistant:
  enabled: true
  legacy_action_sensor: false
  experimental_event_entities: true
  base_topic: zigbee2mqtt
  force_disable_retain: false
  include_device_information: false
  keepalive: 60
  maximum_packet_size: 1048576
  reject_unauthorized: true
  server: mqtt://core-mosquitto:1883
  user: mqtt-user
  version: 4
mqtt:
  server: mqtt://core-mosquitto:1883
  user: mqtt-user
  password: SSG1mqtt%%
serial:
  port: >-
    /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0
  adapter: zstack
frontend:
  enabled: true
  port: 8099
devices:
  '0x0c4314fffef0f4a3':
    friendly_name: MH2_thermostat
    debounce: 5
    homeassistant: {}
    optimistic: true
  '0x00158d00089782ef':
    friendly_name: K_TCL
    temperature_precision: 1
    pressure_precision: 0
    humidity_precision: 0
  '0x50325ffffe28b580':
    friendly_name: OH_thermostat
    debounce: 5
    homeassistant: {}
    optimistic: true
  '0x00158d0008cafea0':
    friendly_name: MH1_TCL
    temperature_precision: 1
    pressure_precision: 0
    humidity_precision: 0
  '0x0c4314fffee0e1c7':
    friendly_name: MH1_thermostat
    debounce: 5
    homeassistant: {}
    optimistic: true
  '0x0017880109a60531':
    friendly_name: Remote Hue2
  '0x00124b0024c19739':
    friendly_name: Switch WC1
  '0xa4c1382bbe297613':
    friendly_name: OH_TE_40
  '0xa4c13819a198585a':
    friendly_name: TK_TCL
  '0x00124b0026b79a57':
    friendly_name: Socket sonoff
  '0x50325ffffe401d12':
    friendly_name: KHH_thermostat
    debounce: 5
    homeassistant: {}
    optimistic: true
  '0x50325ffffe5485ab':
    friendly_name: MH3_thermostat
    debounce: 5
    homeassistant: {}
    optimistic: true
  '0x0017880108f63057':
    friendly_name: Remote Hue1
  '0x9035eafffe792b8e':
    friendly_name: MH1_bulb_2
    homeassistant: {}
    optimistic: true
    disabled: true
  '0x94deb8fffebaf8dc':
    friendly_name: MH1_bulb_1
    homeassistant: {}
    optimistic: true
    disabled: true
  '0x60a423fffeff5d7a':
    friendly_name: Switch tuya2g
  '0x00178801093eee3f':
    friendly_name: MH2_bulb_3
  '0x00178801093ee293':
    friendly_name: MH2_bulb_2
  '0x001788010939e668':
    friendly_name: MH2_bulb_1
  '0x00178801097fffbd':
    friendly_name: MH3_CellingLamp
  '0x54ef441000442e0f':
    friendly_name: Temperature_d aqara
    disabled: false
    homeassistant: {}
    optimistic: true
  '0x00158d000705ded1':
    friendly_name: MH3_TCL
    temperature_precision: 1
    pressure_precision: 0
    humidity_precision: 0
  '0x54ef441000564c31':
    friendly_name: KHH_ventilation
  '0x00158d0008ab179c':
    friendly_name: KHH_TCL
    temperature_precision: 1
    pressure_precision: 0
    humidity_precision: 0
  '0xa4c13843a5361d31':
    friendly_name: VAR_tcl
  '0x00158d0007edf6c3':
    friendly_name: Water leakage1
    disabled: false
    homeassistant: {}
    optimistic: true
  '0x54ef441000688ba2':
    friendly_name: K_freezer
    disabled: false
    homeassistant: {}
    optimistic: true
  '0x54ef441000687f17':
    friendly_name: K_fridge
  '0x54ef441000564a17':
    friendly_name: plug_AC
  '0x804b50fffee8dd38':
    friendly_name: Remote ikea1
  '0x5c0272fffeacf034':
    friendly_name: TK_BT_01
  '0x00158d0007016a10':
    friendly_name: Door sensor
  '0x00158d0001fdf010':
    friendly_name: Remote tint
  '0xf0d1b800001d750f':
    friendly_name: Bulb FL
  '0xf0d1b800001d7def':
    friendly_name: Bulb FM
  '0xf0d1b800001d759a':
    friendly_name: Bulb FR
  '0x0017880104243107':
    friendly_name: Bulb BR
  '0x0017880104242f5f':
    friendly_name: Bulb BL
  '0x0017880108011784':
    friendly_name: Button hue1
  '0xa4c138c39ca2f8f7':
    friendly_name: K_VB_01
  '0x001788010917a654':
    friendly_name: motion_sensor_hue1
    temperature_precision: 1
  '0x0017880109adf21c':
    friendly_name: VAR_bulb_2
  '0x0017880109aed5c6':
    friendly_name: VAR_bulb_1
  '0xb4e3f9fffe25a502':
    friendly_name: VAR_MT_22
  '0x001788010cdbf471':
    friendly_name: OD_motion
  '0x84b4dbfffef09ea7':
    friendly_name: Bulb TK 1
  '0x0015bc001b022846':
    friendly_name: VAR_PM
  '0x60a423fffeff6f90':
    friendly_name: Temperature_d_moes1
    temperature_precision: 1
    temperature_calibration: ''
    homeassistant: {}
    optimistic: true
    disabled: true
  '0x54ef4410004ea974':
    friendly_name: plug_PK
  '0x943469fffe24fe9f':
    friendly_name: Bulb TK 2
  '0xa4c1380818bc88cf':
    friendly_name: car_heat1
    state_action: false
    disabled: false
    homeassistant: {}
    optimistic: true
  '0xa4c138a55b9a7a18':
    friendly_name: car_heat2
  '0x0017880109e5f8fe':
    friendly_name: Hue_go
    disabled: false
    homeassistant: {}
    optimistic: true
  '0x001788010c0405f9':
    friendly_name: '0x001788010c0405f9'
  '0x7cb03eaa00afd880':
    friendly_name: Plug_1
    disabled: false
    homeassistant: {}
    optimistic: true
  '0x00158d0006ed25c6':
    friendly_name: OH_TCL
  '0xf4b3b1fffe993da3':
    friendly_name: TK_MT_21
  '0x001788010802cf24':
    friendly_name: Button hue2
  '0x003c84fffe21454a':
    friendly_name: VAR_BT_01
  '0xdc8e95fffebbd3c3':
    friendly_name: TK_BT_02
  '0xd44867fffe563637':
    friendly_name: XX_MT_33
    identify_timeout: 15
  '0x00158d0008ab1be2':
    friendly_name: MH2_TCL
  '0xa4c13822d8a107b8':
    friendly_name: MH2_LED_Strip
  '0x8c65a3fffeb11787':
    friendly_name: switch_tuya4
  '0x943469fffe801a6b':
    friendly_name: switch_tuya1b
  '0xa4c138a7fe7a616b':
    friendly_name: K_thermostat
  '0xa4c1384672809268':
    friendly_name: TK_thermostat
  '0xa4c1385f363ad774':
    friendly_name: ET_thermostat
  '0x00158d0008ab17a0':
    friendly_name: WC_TCL
  '0xa4c138985bef3d2d':
    friendly_name: wc_thermostat
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
  cache_state: true
  cache_state_persistent: true
  cache_state_send_on_startup: true
  log_level: debug
external_converters:
  - moesss.js
device_options:
  legacy: false
availability:
  enabled: true
  active:
    timeout: 10
  passive:
    timeout: 1500
groups:
  '1':
    friendly_name: MH2_lights
  '2':
    friendly_name: MH1_lights
  '4':
    friendly_name: VAR_lights
  '5':
    friendly_name: MH3_lights
  '6':
    friendly_name: OD_lights_F
  '7':
    friendly_name: OD_lights_B
  '8':
    friendly_name: OD_lights_ALL
  '9':
    friendly_name: TK_lights
version: 4

```

chris-1243, 2025-02-05:
Since v2.0.0, you do not need to specify anything in your `configuration.yaml`.

So remove

```
external_converters:
  - moesss.js
```

In your `/data` folder, you have to create a subfolder `external_converters` and put your *.js inside this folder. Restart Z2M and it should be fine.

SSG1111, 2025-02-05:
@chris-1243  thanks, didn't notice that.
Removing line from config didn't help.

Log shows that it finds and loads converter, but nothing happen still.

[image]

chris-1243, 2025-02-05:
> ```

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_xalsoe3m',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;

```

The `extend` is empty... you need to fill there what the device is able to do (on/off,...)

SSG1111, 2025-02-06:
when I had unsupported device, the code above helped to get it to the list with empty exposes... just to see if that converter is working.

But yes, adding dummy extend definition didn't change anything, I still see exposes and status from recently added standard converter:

```
const {temperature, humidity, battery} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
  zigbeeModel: ['TS0601'],
  model: 'TS0601',
  vendor: '_TZE204_xalsoe3m',
  description: 'Automatically generated definition',
  extend: [temperature(), humidity(), battery()],
};

module.exports = definition;
```

Does this work work for you? If you will replace model/vendor fields to one of your devices, will it overwrite exposes?

Pannben, 2025-02-06:
I have the same issue with my HZC motion sensor since updating. It is listed as "unsupported" but I have up until this version been running it fine with this external definition (placed in the "external_converters" directory).

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['MotionSensor-ZB3.0'],
    model: 'MotionSensor-ZB3.0',
    vendor: 'Shyugj',
    description: 'Motion sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery, fz.illuminance],
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.occupancy(), e.battery_low(), e.battery(), e.illuminance(), e.tamper()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};
module.exports = definition;
```

SSG1111, 2025-02-06:
yes, looks like ones device become "supported", old external converters don't recognize those anymore. Maybe something shall be updated in "zigbeeModel", "model", "vendor" ?

I think I still have few unsupported devcies, it would be good to check if ext. converters will still work with those.

Pannben, 2025-02-06:
> yes, looks like ones device become "supported", old external converters don't recognize those anymore. Maybe something shall be updated in "zigbeeModel", "model", "vendor" ?
> 
> I think I still have few unsupported devcies, it would be good to check if ext. converters will still work with those.

My device was (and still is) unsupported, but it was working fine with my external definition before this update.

SSG1111, 2025-02-06:
@Pannben Okay! This means that the converters database update has nothing to do with this. Maybe some other breaking changes which were not taken into account or bug...

At least LOG says "_z2m: Loaded external converter 'xxxxx.js'" which means that it finds correct file.  Not sure how else this can be troubleshooted. Maybe just wait when more users will report similar issues.

You might be still able to rollback to the previous version via backup restore?

SSG1111, 2025-02-06:
yes it works again!! 👍 amazing...
I can catch it up from here and figure out why my old converter stopped to work. Thanks a lot @chris-1243 !!!

SSG1111, 2025-02-06:
> [@SSG1111](https://github.com/SSG1111) I just had a try on a test system and external converters are working perfectly well.
> 
> [image]
> 
> [image]
> 
> As you may see, I changed the name of the devices and the brands. I created two external converters to be sure with two different devices.
> 
> I just had a close look to your converter and also check in `zigbee-herdsman-converters` as your device is Tuya based... Would you mind try the lines below ?
> 
> ```
> const m = require('zigbee-herdsman-converters/lib/modernExtend');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> 
> const definition = {
>     fingerprint: tuya.fingerprint('TS0601', ['_TZE204_xalsoe3m']),
>     model: 'TS0601',
>     vendor: 'MOES',
>     description: 'Converter made by the community for a try',
>     extend: [
>         m.temperature(),
>         m.humidity(), 
>         m.battery(),
>     ],
> };
> 
> module.exports = definition;
> ```

can't find how to mark this as SOLUTION, sorry....

chris-1243, 2025-02-06:
No worries.... Just for my own understanding as I do not have any Tuya devices in my network, any values reported with the `extend` I provided in the converter?

chris-1243, 2025-02-06:
I would just change your converter like this and keep the rest if everything is working as you like

```
...
const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_xalsoe3m']),
    model: 'TS0601',
    vendor: 'MOES',
    description: 'MOES thermostat',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
...
``` 

If I am not wrong, `extend` is the new and easiest way to write a converter as everything relies in this library. At least what it is standard. I am not an expert at all.

As you have a Tuya device, I guess and according to what I saw in `tuya.ts` from `zigbee-herdsman-converters`, it is better to keep those datapoints. Tuya is not doing everything according to the spectification...