Koenkk/zigbee2mqtt #26205: General Instability: Restarting z2m or rebooting the system does not resolve the issues.

Allenspeed1, 2025-02-06:
### What happened?

After updating to Zigbee2MQTT version 2.1.0, I’ve been experiencing several issues:

Latency Problems: Devices experience delays in response, often taking several seconds to execute commands, which has significantly degraded the user experience.

Electrical Switch Going Offline: A plugged-in electrical switch (e.g., Ventilador 2) frequently goes offline and throws the following error:

z2m: Failed to read state of 'Ventilador 2' after reconnect (ZCL command 0x00XXXXXXXXXX/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 42268 - 1 - 3 - 6 - 1 after 10000ms))

### What did you expect to happen?

Improved stability and performance after updating to Zigbee2MQTT 2.1.0.

Minimal latency and faster response times for device commands.

Reliable connectivity for all devices, including the plugged-in electrical switch (Ventilador 2).

### How to reproduce it (minimal and precise)

Update to Zigbee2MQTT version 2.1.0.

Attempt to load the z2m dashboard or generate the network map.

Observe delays in device response times.

Monitor the plugged-in electrical switch (Ventilador 2) for frequent disconnections and timeout errors.

### Zigbee2MQTT version

Zigbee2MQTT Version: 2.1.0 (commit: 5e9d53f0aa94c947837dbb8252b1c70ff3bc08a2)

### Adapter firmware version

20220219

### Adapter

Sonoff Dongle Plus

### Setup

Docker Container for HA, Z2M and MQTT

Gecko3600, 2025-02-06:
Seems like that youve got the same issue as me: #25945

RobLeighton67, 2025-02-06:
I've moved from 20220219 Coordinator firmware to the much newer 20240710 - it seems logical to expect that this will be more reliable with newer builds ?
So far seems the case, definitely no worse - I run only about 10 devices per site though.

Allenspeed1, 2025-02-06:
I have about 20, plus i have another zigbee hub, but on diferrent channel, its a hue bridge, but on way different channels, hue is on 11 if i recall and z2m is on 20, my wifi is on 6, i was thinking on updating firmware, but i see no differences that could apply to this case

HenrikHills, 2025-02-07:
Having same issue, started when upgrading to 2.1.1 and  (maybe) triggering some OTA updates that popped up…

Error 2025-02-07 23:44:15z2m: Failed to read state of 'Kontoret Spotter' after reconnect (ZCL command 0x7cc6b6fffebc42ea/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))

Allenspeed1, 2025-02-19:
> I figured that the builds were probably tested with the latest firmware, Also possible that something in the firmware changes interacts with the build changes to (accidentally) solve an issue.

I just updated now its worst, after the forced reset, now devices wont pair

Allenspeed1, 2025-02-25:
Same result, I am having unstable devices now and then, suddenly a few drop from the mesh, and endpoints suddenly go offline for more than 24hrs, so far only 2 endpoints at the time, what its more nerve wrecking is that one of them is 3 feet away from the coordinator, so...

Koenkk, 2025-02-25:
Could you check if this fw is more stable? https://github.com/Koenkk/Z-Stack-firmware/tree/Z-Stack_3.x.0_coordinator_20221226/coordinator/Z-Stack_3.x.0/bin

Allenspeed1, 2025-02-25:
I will try to give it a go with that FW:

My current INFO:

Zigbee2MQTT version
[2.1.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.1.1) commit: [1accb8b66b95b99658c50e4c0209edcb946a5fe2](https://github.com/Koenkk/zigbee2mqtt/commit/1accb8b66b95b99658c50e4c0209edcb946a5fe2)
Coordinator type
zStack3x0
Coordinator revision
20240710
Coordinator IEEE Address
0x00
Frontend version
0.9.4
zigbee-herdsman-converters version
21.31.0
zigbee-herdsman version
3.2.5

Koenkk, 2025-03-01:
Let's continue in https://github.com/Koenkk/Z-Stack-firmware/issues/540, please try the 99250223 from https://github.com/Koenkk/Z-Stack-firmware/issues/540#issuecomment-2692420392

closed 2025-03-01 by Koenkk