Koenkk/zigbee2mqtt #26208: [New device support]: SMLIGHT SLZB-06p10 in Router mode

stephdust, 2025-02-06:
### Link

https://smlight.tech/product/slzb-06p10/

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0x00124b00336cc6a9","nwkAddr":36532,"manufId":0,"manufName":"SMLIGHT","powerSource":"Mains (single phase)","modelId":"SLZB-06P10","epList":[8,242],"end

### Zigbee2MQTT version

2.1.0

### Comments

Hello, I'm using the SMLIGHT SLZB-06P10 in 'Router' mode
could you add this model in the file src/devices/smlight.ts for the next release branch ?

see below the external definition file
thanks and regards
Stephane

### External definition

```shell
from file src/devices/smlight.ts :

import * as m from '../lib/modernExtend';
import {DefinitionWithExtend} from '../lib/types';

const definitions: DefinitionWithExtend[] = [
    {
        zigbeeModel: ['SLZB-06M', 'SLZB-06', 'SLZB-06p7', 'SLZB-06P10', 'SLZB-07', 'SLZB-0xp7', 'SLZB-07Mg24'],
        model: 'SLZB-06M',
        vendor: 'SMLIGHT',
        description: 'Router',
        whiteLabel: [
            {vendor: 'SMLIGHT', model: 'SLZB-06', description: 'Router', fingerprint: [{modelID: 'SLZB-06'}]},
            {vendor: 'SMLIGHT', model: 'SLZB-06p7', description: 'Router', fingerprint: [{modelID: 'SLZB-06p7'}]},
            {vendor: 'SMLIGHT', model: 'SLZB-06p10', description: 'Router', fingerprint: [{modelID: 'SLZB-06p10'}]},
            {vendor: 'SMLIGHT', model: 'SLZB-07', description: 'Router', fingerprint: [{modelID: 'SLZB-07'}]},
            {vendor: 'SMLIGHT', model: 'SLZB-0xp7', description: 'Router', fingerprint: [{modelID: 'SLZB-0xp7'}]},
            {vendor: 'SMLIGHT', model: 'SLZB-07Mg24', description: 'Router', fingerprint: [{modelID: 'SLZB-07Mg24'}]},
        ],
        extend: [m.linkQuality({reporting: true}), m.forcePowerSource({powerSource: 'Mains (single phase)'})],
    },
];

export default definitions;
module.exports = definitions;
```

### What does/doesn't work with the external definition?

just would like to modify the existing definition entry file src/devices/smlight.ts
and add SLZB-06p10, so it is recognized when used in router mode
thanks

Koenkk, 2025-02-07:
Added!

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `SLZB-06p10.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2025-02-07 by Koenkk

stephdust, 2025-02-08:
Hi,
I don't think a new picture is necessary
this definition uses the picture of the existing model 'SLZB-06M'

  model: 'SLZB-06M',

this picture is already in z2m
thanks
(I see a picture in mine)

a10kiloham, 2025-02-25:
this is still not working for me - i can't get the logs up however, it just shows ERROR in the front end. i thought these were generallly supported weren't they?
cheers