Koenkk/zigbee2mqtt #26215: Tuya TS0601_soil_3 battery state is 50% with brand new batteries

mafr95, 2025-02-06:
### What happened?

I use over 10 of the TS0601_soil_3, all with brand new batteries and each sensor shows me a maximum battery level of 50 percent.

I have seen this problem #24611 should have been solved by a fix in the zigbee-herdsman-converters with release 21.3.0. 

> https://github.com/Koenkk/zigbee-herdsman-converters/pull/8449#issue-2723595995

Unfortunately not for me. I am using the latest Zigbee2MQTT version 2.1.0 and the Zigbee Herdsman converter version 21.27.1.

I'm sorry to open the problem again, but I don't know how to help myself.

### What did you expect to happen?

Battery state should be 100% with brand new batteries

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

20230507

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Homeassistant 2025.2.0 on Rasberry PI 4

mrmaximas, 2025-02-07:
please post database.db entry for this device.

mafr95, 2025-02-07:
```json
{
    "id": 42,
    "type": "EndDevice",
    "ieeeAddr": "0xa4c1386f56958f64",
    "nwkAddr": 50827,
    "manufId": 4417,
    "manufName": "_TZE284_aao3yzhs",
    "powerSource": "Battery",
    "modelId": "TS0601",
    "epList": [
        1
    ],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 81,
            "inClusterList": [
                4,
                5,
                61184,
                0,
                60672
            ],
            "outClusterList": [
                25,
                10
            ],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "65487": 14400,
                        "65503": "0 -/i",
                        "65506": 56,
                        "65508": 0,
                        "65534": 0,
                        "modelId": "TS0601",
                        "manufacturerName": "_TZE284_aao3yzhs",
                        "powerSource": 3,
                        "zclVersion": 3,
                        "appVersion": 77,
                        "stackVersion": 0,
                        "hwVersion": 1,
                        "dateCode": ""
                    }
                }
            },
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        }
    },
    "appVersion": 77,
    "stackVersion": 0,
    "hwVersion": 1,
    "dateCode": "",
    "zclVersion": 3,
    "interviewCompleted": true,
    "meta": {
        "configured": 217189513
    },
    "lastSeen": 1738922435909
}
```

mrmaximas, 2025-02-07:
have you tried re-interviewing and/or reconfiguration?

mafr95, 2025-02-07:
Just did it with one sensor, no luck. 

And as I mentioned before, I have a total of 18 of these sensors. All of them show a maximum of 50 percent.

mafr95, 2025-02-07:
Does the battery report come with every sensor update? 

I put the sensor in a glass of water to force some update. The soil moisture value was then reported as 100 %. However, the battery value is still unchanged at 39 %

```json
{
    "battery": 39,
    "battery_state": "medium",
    "last_seen": "2025-02-07T11:40:08.704Z",
    "linkquality": 91,
    "soil_moisture": 100,
    "temperature": 21.2,
    "temperature_unit": "celsius"
}
```

RubenKelevra, 2025-02-07:
See https://github.com/Koenkk/zigbee2mqtt/issues/24611#issuecomment-2642740794

bo0tzz, 2025-04-09:
This isn't stale.

What does it take to get the correct data for this? I have a few of these devices and a multimeter, is that sufficient?

foddy-foo, 2025-04-16:
I also have a few of these sensors and can confirm the battery remaining percentage is not being shown correctly.
Zigbee2MQTT version 2.2.1-1 
Sonoff Zigbee USB Dongle Plus P
Soil sensor manufacturer: _TZE284_aao3yzhs
Soil sensor model: TS0601_soil_3
Home Assistant Core: 2025.4.2

hoppke, 2025-04-17:
TS0601 / _TZE284_aao3yzhs
Detected as TS0601_soil_3
Zigbee2mqtt 2.2.1-1 (via HAOS add-on)

I've 3 '_TZE284_aao3yzhs' units, and with fresh AAA batteries they're reporting battery % as 50 / 50 / 52.
At the same time 'battery state' reported as 'high'.
Maybe the firmware was originally meant for 2x the input voltage, maybe it needs an offset / correction curve;

Maybe someone can show a graph of the battery %, as it discharges from 'fresh' to 'offline', to shed more light on this.

bo0tzz, 2025-04-17:
I have three of these sensors, here is the graph of their reported battery values in my prometheus instance over time.

[image]

The peak at the start is with fresh rechargeable batteries and reads 35, the lowest value right now is 8. If necessary I can get voltage readings straight off the batteries to compare, though I don't have particularly accurate equipment.

Mistic92, 2025-04-21:
Same issue with [TS0601_soil_3](https://www.zigbee2mqtt.io/devices/TS0601_soil_3.html#tuya-ts0601-soil-3) both of them have ~50% with new batteries.

Koenkk, 2025-04-21:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/9abd163d8da73f9753a3866d5d9d8e49): 

- save this as file next to `configuration.yaml` in the `external_converters` directory (create if it doesn't exist yet) as `ext_converter.js` (so `external_converters/ext_converter.js`)
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- check if issue is fixed

bo0tzz, 2025-04-21:
I did this, but the reported power value hasn't changed. I will update if it changes with a new report from the device.

[image]

hoppke, 2025-04-22:
Enabling the external converter certainly did something for me:

[image]

I've a feeling the numbers are still too low (84, 80, 45% on 5 day old batteries?), but I'll need to measure the voltage on battery terminals in each unit to have more than a 'feeling'...

bo0tzz, 2025-04-22:
Looking at my graphs now, it actually ended up causing a sharp _drop_. So maybe the values being reported are more accurate now and my batteries are actually just empty :P I'll charge one sensor up and report back.

[image]

bo0tzz, 2025-04-22:
With freshly charged batteries it reported 62%.

pozdrowiony, 2025-05-26:
> Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/9abd163d8da73f9753a3866d5d9d8e49):
>
> * save this as file next to `configuration.yaml` in [...]

nothing changed to me

Daradaal, 2025-05-31:
Hi everyone,

I'm experiencing the same issues with the TS0601_soil_3 sensor (_TZE284_aao3yzhs):

- **Soil moisture readings are way too high:** For example, my trusted non-connected probe shows around 30% soil moisture, while the Tuya sensor reports 93% in the same spot.
- **Battery reporting is unreliable:** Even with brand new AAA batteries (Duracell, measured at 1.6V), the battery percentage shows between 35% and 50%. Sometimes, after a few days, it drops to 20-30% even though the voltage is still good.

I've seen that there's an option to set an offset for the moisture, but I don't really understand how it works, and I'm not sure if it makes the data more reliable or just shifts a broken scale.

Is anyone here actually able to get **reliable soil moisture readings** (with values that match a calibrated or analog probe) with this device?  
Any feedback about **battery reporting**, or workaround for this?  
I'm happy to help test if someone has a converter, or some new trick.

Thanks in advance!

zyi840449, 2025-06-22:
> Hi everyone,
>
> I'm experiencing the same issues with the TS0601_soil_3 sensor (_TZE284_aao3yzhs):
>
> * **Soil moisture readings are way too high:** For example, my trusted non-connected probe shows [...]

I'm having exactly the same problem as you.

RubenKelevra, 2025-06-25:
I'm using one `_TZE284_aao3yzhs` and cannot confirm the reports regarding soil moisture.

Please make sure that there's *no* metal close to the sensors, or other objects. Those sensors measure the conductivity, but the capacitance of the soil in the environment.

So if the soil sensor is placed at the edge of the pot or close to metal objects, like rods to keep the plant upright, the measurement will be skewed.

If this does not help, please create a dedicated issue for this problem, going off-topic here isn't really helpful, to get reports by other people if they see the same issue.

t21drl, 2025-08-17:
Can confirm that the battery never goes above 50% and the moisture seems to be mixed up and it will now read either 100% or 0% since a recent update

MatzeMuc86, 2025-08-17:
Is there a fix planned?

bo0tzz, 2025-08-20:
@Koenkk I set the log level to debug but didn't get any debug logs output. Here's everything I did get out (as well as a config dump to show the log level is right) https://gist.github.com/bo0tzz/fd015702d050d8103ee12f2a13ee48d9.

bo0tzz, 2025-08-20:
My bad, I hadn't realised that debug logs would only show up on stdout and not in the UI. https://gist.github.com/bo0tzz/ac08513f4a369994c0766a0786b71deb

Tschaegged, 2025-09-09:
I can confirm the same issue with 3 sensors

phyzical, 2025-10-31:
@Koenkk can confirm it started reporting what was represented in the payload unedited now 👍 

(if this is actually correct im not sure i dont have a tuya hub on hand to test)

[image]

alexsotoaguilera, 2025-11-04:
in my case, from reporting 6% with rechargeable batteries to reporting 0%

gycone, 2025-11-04:
> in my case, from reporting 6% with rechargeable batteries to reporting 0%

Same in my case

phyzical, 2025-11-04:
i just tested my other device just incase on the latest release, its funny i changed the battery twice first time spat out 87% second time 100% testing with the same set of batteries. 

[image]

makes me wonder if there is some sort of timing around when it detects the battery charge and sends cause i know it doe not send the battery updates often

they are probably just  designed in a way to reduce battery use ironically doesn't seem to reduce much 😆 these things chew battery

alexsotoaguilera, 2025-11-14:
"I replaced the batteries with rechargeable ones. After this change, it still reports 6% even with fully charged batteries."

Isusami, 2025-12-31:
I'm using ```SUPER ALKALINE BATTERY 1.5v AAA Q-Connect``` for all my devices, but still have the battery reading mismatches. I changed all the batteries on the same day.

[image]