Koenkk/zigbee2mqtt #26219: Philips 8718699693985 Hue Button OTA update not working

hansw1204, 2025-02-06:
### What happened?

When trying to update the Hue Button, the error message is:
 Update of 'XXX Button' failed (Device didn't respond to OTA request)

### What did you expect to happen?

OTA update to proceed

### How to reproduce it (minimal and precise)

Attempt to update Hue Button (Philips 8718699693985) from the Z2MQTT OTA menu

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

core: v2.7.1 / zigbee: 20221226

### Adapter

SLZB-06

### Setup

Add-ON on HA OS on rPi5 8GB

### Debug log

2025-02-06 15:25:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/XXX Button', payload '{"battery":100,"linkquality":112,"update":{"installed_version":33566472,"latest_version":33574183,"state":"available"}}'
Info 2025-02-06 15:25:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{},"error":"Update of 'XXX Button' failed (Device didn't respond to OTA request)","status":"error","transaction":"83zos-2"}'
Error 2025-02-06 15:25:49z2m: Update of 'XXX' failed (Device didn't respond to OTA request)

mundschenk-at, 2025-02-06:
Try to press it to wake up the button.

hansw1204, 2025-02-06:
This worked, thank you!
Sorry, too noob here.

hansw1204, 2025-02-06:
I also want to say

THANK YOU for such a great piece of software!